MarlinFirmware/Marlin #1601: ST7920 SCLK Timing

chris-bo, 2015-03-13:
I use:
-Sanguinololu with ATmega1284p @20Mhz
-Full Graphics Smart Controller Clone
-Marlin 1.0.2 compiled with Arduino 1.0.6

I noticed random pixelshifts (mostly missing inactive pixels, which caused a left shift of the remaining row).  
A quick fix was adding a second "WRITE(ST7920_CLK_PIN,1);" in ultralcd_st7920_u8glib_rrd.h :  

static void ST7920_SWSPI_SND_8BIT(uint8_t val)
{
  uint8_t i;
  for( i=0; i<8; i++ )
  {
    WRITE(ST7920_CLK_PIN,0);
    WRITE(ST7920_DAT_PIN,val&0x80); 
    val<<=1;
    WRITE(ST7920_CLK_PIN,1);
    WRITE(ST7920_CLK_PIN,1);
  }
}
I'm not shure if this affects other setups but it might be caused by exeeding the st7920 maximum f_sclk of 2.5MHz ( http://www.crystalfontz.com/controllers/ST7920.pdf Page 43 )

boelle, 2015-03-14:
i dont think it makes a difference but remember to compile with latest non-beta which right now is ide 1.6

chris-bo, 2015-03-14:
with arduino 1.6.1 (download for linux version 1.6.0 currently broken) its even more messed up:
 
[image]

thinkyhead, 2015-03-14:
Some of the LCD code inserts "nop" a few times to deal with timing issues. Perhaps this is a good place for some of those too.

boelle, 2015-03-14:
if ithe same display as http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller

i will be able to test also when i get the last board in my Ramps variant done.. but i bet you will get this one solved long before that :-D

nophead, 2015-03-14:
It needs to be conditional on the clock frequency because most people run
at 16MHz, not 20.

thinkyhead, 2015-03-15:
@nophead Is calling a microsecond delay function worth it? Or would the call and return take more than enough time?

avluis, 2015-03-15:
Related: https://github.com/MarlinFirmware/Marlin/issues/1611

avluis, 2015-03-15:
Issue fixed: https://github.com/MarlinFirmware/Marlin/pull/1615

boelle, 2015-03-17:
@avluis said solved but this one was reopened?

thinkyhead, 2015-03-19:
Still waiting to see how the original issue @chris-bo reported stands.

chris-bo, 2015-03-19:
For me everythink works fine now. Is there anyone to confim?

boelle, 2015-03-19:
could ask in testers needed issue down the list maybe?

AnHardt, 2015-03-19:
@boelle 
Will not make to much sense. 20 MHz boards are rare here. 
For me it works 16 MHz.
Please close.

boelle, 2015-03-19:
ok, i have a 20 mhz based 644 board so will try to remember this one even thou i close it

closed 2015-03-19 by boelle

thinkyhead, 2015-03-19:
Since we have a define for processor speed, we should be able to conditionally produce code with the right amount of delay. So I'll be keeping that in mind for Marlin as a whole also.