MarlinFirmware/Marlin #1604: What is DUAL_Y_CARRIAGE (stepper.cpp)?

thinkyhead, 2015-03-14:
I see that `DUAL_Y_CARRIAGE` is used in `stepper.cpp` in the `babystep` function, but it isn't defined anywhere, and it seems like maybe it should be using `Y_DUAL_STEPPER_DRIVERS` in the places where it's using this symbol. What do you think? Is `DUAL_Y_CARRIAGE` old and obsolete?

boelle, 2015-03-14:
not knowing much of the code but reading what you descriped i would say loose it... 

to me it could also look like a half finished job that was never completed... how long ago was this if you can see it?

thinkyhead, 2015-03-14:
Looks like a [very old bit of code](https://github.com/MarlinFirmware/Marlin/commit/d147a057ac75e52421c8d277655e2e48ca0376a3). It is probably as I assume. Such a rare use-case probably no one has ever encountered a problem. My new PR #1606 cleans that up and some other things besides.

Wurstnase, 2015-03-14:
Maybe the dual y was supposed to become a dual x carriage?

thinkyhead, 2015-03-15:
@Wurstnase Well, the `DUAL_X_CARRIAGE` code is all implemented. I'm sure the Y version is meant to be symmetrical, but it is only ever referenced in the `babystep` function (which stands alone and is called from the ISR in `temperature.cpp`). On the other hand, the `Y_DUAL_STEPPER_DRIVERS` and `Y_DUAL_STEPPER_DRIVERS` options are implemented, in clear parallel to the `DUAL_X_CARRIAGE` option.

The larger question is, in the `babystep()` method, should these options operate in the same manner as they do in the stepper code generally? For example, the X dual stepper option may be disabled depending on certain state conditions set in `Marlin_main.cpp`. Should the `babystep` function (which is only ever called from the LCD controller) take this flag into account, or always move both X motors? The existing code says: always move both, so I have preserved that in #1606.

Wurstnase, 2015-03-18:
I took a look to that old code. This looks like it was only copy'n'paste in that case.

thinkyhead, 2015-03-19:
Ok, switched to `Y_DUAL_STEPPER_DRIVERS` in that case, but kept the original behavior - no checking of flags for the X carriage, it always steps both carriages.

closed 2015-03-19 by thinkyhead