MarlinFirmware/Marlin #1608: Shouldn't configuration options be in Configuration.h ?

Roxy-3DPrintBoard, 2015-03-14:
I'm trying to move over to the current code base so I can contribute to the testing and help verify things are working right.    

In Marlin_Main.CPP this code exists in the G29 Auto Bed Level function:

  // Use one of these defines to specify the origin
  // for a topographical map to be printed for your bed.
  enum { OriginBackLeft, OriginFrontLeft, OriginBackRight, OriginFrontRight };
  #define TOPO_ORIGIN OriginBackRight

Shouldn't the #define for TOPO_ORIGIN be in Configuration.h ?

alexborro, 2015-03-14:
I think it can be removed at all.

The convention for 3D Printing is having the origin on the front left
corner.

We cannot support a complicated code because 0.01% of users wanna have a
swapped axis.

Cheers.

Alex.

NarimaanV, 2015-03-14:
Alex, just to elaborate, by "front left", do u mean the hot end sits at the closest left corner of the bed if you're looking at the the printer?

Roxy-3DPrintBoard, 2015-03-14:
Those options set where the origin is.   If you are looking at the front of the printer, where is the (0,0) coordinate?  It can be Front Left, Front Right, Back Left or Back Right.

The code is not complicated very much by allowing the user to select where the origin is.  And in fact, many commercially available printers (besides the home made printers) have the origins in different corners.    This is a very useful option for people to help them level their print beds.   Having the display correctly orientated makes it much easier and more intuitive to use!

NarimaanV, 2015-03-14:
Completely agree with Roxy! And a lot of the printers that have different endstop locations, for those that use the endstop home as 0,0,0 can't move the endstops because of the design of the printer and the necessary spacing needed for the switches (Makerfarm i3/i3v is a prime example, I own one and several others have mentioned owning one too, those have home be BackRight and you can't really move the endstops without potentially losing print space). Also in regards to the original topic, if it wouldn't cause any issues or be too much work, it is more intuitive to move the #define to Configuration.h at the end of the section where auto-bed leveling configuration takes place.

Roxy-3DPrintBoard, 2015-03-14:
> Also in regards to the original topic, if it wouldn't cause any issues or be too much work, it is more intuitive to move the #define to Configuration.h at the end of the section where auto-bed leveling configuration takes place.

I suspect it is a combination of things caused it to be where it is in the Marlin_main.cpp code.   When I originally did the code, I didn't want people to have to modify two files so I put it with the C code in Marlin_main.cpp.   But that doesn't make sense now.   And who ever moved the code over to the main code base was probably testing the various print outs to make sure they functioned correctly.   So it may be it was just easier (for the time being) to leave it where it was.

NarimaanV, 2015-03-14:
So until we can move the define, maybe we leave a commented note in Configuration.h to remind the user to make sure they have that setting set correctly if it's not BackRight in Marlin_Main.cpp if they plan on using the Bed Topography Map?

Roxy-3DPrintBoard, 2015-03-14:
I think I would argue we just move the #define to be with the rest of the Auto Bed Leveling stuff in Configuration.h     Mostly, I was asking the question because so much has changed with this code base with regard to the clean up and bug fixing.  It just seemed inconsistent with the rest of the code base.   So, mostly I was trying to understand the new coding style and rules.

Moving the #define is pretty trivial.  And in fact I'll do it, (and issue a Pull Request) if somebody doesn't beat me to it.   But right now I have to get the current code base up and running first.

NarimaanV, 2015-03-14:
Oh perfect then yah moving the define is the best thing to do if it's simple enough :+1:

alexborro, 2015-03-14:
@NarimaanV , you cannot mix Endstop position with Origin Position. You can have your endstops in any place and still have the origin on the Front Left corner. The Origin position is just an option. And I see no reason to not having it on the standard Front Left corner. Almost ALL professional printers (I'm talking about Stratasys and 3D Systems ones) follow this standard.

@Roxy-3DPrintBoard , The matter is not a complicated code.. Indeed it is not, but we have a LOT of stuffs like that in Marlin... and the whole code is becoming a huge mess!! Too much defines and configuration options. I'm trying to clean up the code; remove many legacy features that just one or two people in the world uses but makes the maintaining code task a really pain in the a... Also I usually have trouble implementing a new nice feature because there are a lot of other pre-historic things in the way...

Cheers.

Alex.

ntoff, 2015-03-14:
my origin is (looking at the printer from the front) the back right corner. My print bed is essentially on backwards. Technically though it's still the "front left" if you were looking at it how the model is oriented when it prints.

> Too much defines and configuration options. I'm trying to clean up the code; remove many legacy features that just one or two people in the world uses but makes the maintaining code task a really [...]

Maybe make your own version then? Isn't marlin pretty much just that? Bits and pieces of other firmwares glued together. What's wrong with starting up your own new version and get rid of all the old useless stuff.

Roxy-3DPrintBoard, 2015-03-15:
> @Roxy-3DPrintBoard , The matter is not a complicated code.. Indeed it is not, but we have a LOT of stuffs like that in Marlin... and the whole code is becoming a huge mess!! Too much defines and [...]

Well, I don't know what to tell you other than a lot of people really like and use the Topographical Map feature.  It is a big help when leveling the print bed.   It is also helpful when trying to identify imperfections and trying to figure out how to place objects on your print bed (to be printed).   I can state with confidence that more than one or two people are using it.

alexborro, 2015-03-15:
Roxy, do not get me wrong. The topographic map is awesome, you did a nice job! I used it last week many times to check my bed level. I'm the one who asked you to submit the PR to mainline.

I'm not planning to remove this feature, just the option for Origins other than Front Left. It's too much code for something pointless.

Cheers.

Alex.

NarimaanV, 2015-03-15:
Correct me if I'm wrong, but does homing position and origin position actually matter for anything besides something like this feature? As in if I changed my origin to FrontLeft and made my X and Y Min endstops into Max endstops, theoretically, wouldn't the printer behave the same? Still home to the endstops and start the print from there? If that's true, then we could do something else to this feature to make both sides of the argument happy. Theoretically, and I haven't looked at the exact code yet so this might completely not work, we could do this: 

Modify the feature to, by default, do the topographical map assuming the origin is FrontLeft, the same way it would be if the ONLY option was FrontLeft (i.e. Alex's idea), thereby reducing the code size. Then include a commented out option that would modify the feature to allow the origin to be any of the four corners, but in such a way that that part of code is only ever even touched IF the user wants to use a different origin, similar to how ABL only uses space if you enable the feature (i.e. Roxy's idea). This would increase the code size ONLY if the user wanted to use the option, which we could leave a note telling them it would, but then it's up to the user to either deal with more space taken or change their origin depending on their needs and/or preferences. Some users will have more than enough space to be able to sacrifice the extra space and leave their origin be, others will deal with changing the origin to get the feature and keep space needed down. Everyone's happy :D

Any thoughts? Is it feasible/possible? Or any thoughts on why the origin HAS to be FrontLeft (besides the idea of standardizing 3D printers, which is harder said than done)?

thinkyhead, 2015-03-15:
If the front-left can be figured out at compile-time, then perhaps front-left should be the default starting-point for the TOPO_MAP, but then the other options could be available as an extra parameter to G29.

Roxy-3DPrintBoard, 2015-03-15:
>If the front-left can be figured out at compile-time, then perhaps front-left should be the default starting-point for the TOPO_MAP, but then the other options could be available as an extra parameter to G29

I was thinking about this...  My guess is 1/2 of the printers have the origin at Front Left.  The rest are probably spread across the other 3 options.   This won't work all of the time, there are going to be exceptions:  But I think often times you will know the corner the origin is in from looking at the:

#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define X_HOME_DIR -1
#define Y_HOME_DIR -1

I like the extra parameter idea for the G29 but the obvious problem with that is it would be good to have a multiple letter value like LF (left front) or RB (right back) and right now that isn't supported.   Perhaps it could be described as how many clockwise 90 degree turns it takes to get your origin to the Front Left?   Something on the order of:  G29 O 2      to handle the Right Back case ????

Looking at the Configuration.h file, I see almost 300 lines of it dedicated to LCD panels.   My vote would be to move this one line to the area of the Configuration.h file that has all of the Auto Bed Leveling stuff.   And once the LCD Panel stuff is under control we might have a good answer for this one line of configuration information?

NarimaanV, 2015-03-15:
^^^ +1 The line should definitely be moved since the option for different origins for the topographical map is currently still in.

nophead, 2015-03-15:
A lot of machines have the origin at the centre, not in any corner.

thinkyhead, 2015-03-15:
@nophead So in that case the X and Y home position will be set to some negative (or positive for MAX) numbers in the Configuration. So it should still be able to determine where the physical "front-left" is located (which of course the printer "knows" – this is how it gets prints oriented the right way).

closed 2015-03-16 by thinkyhead