MarlinFirmware/Marlin #1610: CoreXY Y-homing error

Wurstnase, 2015-03-15:
Don't know if this is a problem with my ARM-version but I started again to port all new stuff. After that when I run a G28 Y the printer homes X again. Y moves work.

alexborro, 2015-03-15:
I will try that tomorrow... but last Friday it was working fine (with
latest Development version).

Cheers.

Alex.

Wurstnase, 2015-03-15:
Right. Last one works fine. Just tried the latest version from today.
My CorXY has max X and max Y end stops.

AnHardt, 2015-03-15:
Depending on when you pulled today this can be caused by #1606 and is fixed by #1614.

thinkyhead, 2015-03-15:
With that merged, let me know if this is still an issue.

Wurstnase, 2015-03-16:
Now the old homing-bug for coreXY is back.
With max endstops on X and Y I can only home X or Y when the other endstop is free.

alexborro, 2015-03-16:
I'm working on it right now... it has been 2h and I still cannot find the
issue.. that's the reason I fear mass changes in stepper.cpp

Cheers.

Alex.

Wurstnase, 2015-03-16:
Solved again by Alex!

boelle, 2015-03-17:
so we can close this one as fixed?

closed 2015-03-17 by alexborro