MarlinFirmware/Marlin #1613: Support multiple extruder drives but only one heater

daprice, 2015-03-15:
When attempting to configure Marlin with ```EXTRUDERS``` > 1 but using a board with only one heater output (in my case ```BOARD_RAMPS_13_EFB```), compilation fails with the errors below because it can’t find the pin for the second heater (because there is none). This should be a valid setup for hotends such as the E3D Cyclops (which has two filament inputs and therefore two extruder motors, but only one heater and nozzle) and potentially others, so it would be nice if Marlin could support it.

Probably either commands to set Extruder >=2's temperature should control Extruder 1 instead, or just be ignored entirely.

```
Arduino: 1.6.0 (Mac OS X), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

temperature.cpp:495:35: note: in expansion of macro 'HEATER_1_PIN'
   #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, v)
                                   ^
temperature.cpp:1298:9: note: in expansion of macro 'WRITE_HEATER_1'
         WRITE_HEATER_1(soft_pwm_1 > 0 ? 1 : 0);
         ^
fastio.h:29:40: error: pasting "DIO" and "-" does not give a valid preprocessing token
 #define _WRITE_NC(IO, v)  do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                        ^
fastio.h:44:105: note: in expansion of macro '_WRITE_NC'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                                         ^
fastio.h:67:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:495:29: note: in expansion of macro 'WRITE'
   #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, v)
                             ^
temperature.cpp:1298:9: note: in expansion of macro 'WRITE_HEATER_1'
         WRITE_HEATER_1(soft_pwm_1 > 0 ? 1 : 0);
         ^
pins_RAMPS_13.h:95:31: error: invalid suffix "_WPORT" on integer constant
   #define HEATER_1_PIN       -1
                               ^
fastio.h:29:48: note: in definition of macro '_WRITE_NC'
 #define _WRITE_NC(IO, v)  do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                ^
fastio.h:67:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:495:29: note: in expansion of macro 'WRITE'
   #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, v)
                             ^
temperature.cpp:495:35: note: in expansion of macro 'HEATER_1_PIN'
   #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, v)
                                   ^
temperature.cpp:1298:9: note: in expansion of macro 'WRITE_HEATER_1'
         WRITE_HEATER_1(soft_pwm_1 > 0 ? 1 : 0);
         ^
fastio.h:29:69: error: pasting "DIO" and "-" does not give a valid preprocessing token
 #define _WRITE_NC(IO, v)  do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                                     ^
fastio.h:44:105: note: in expansion of macro '_WRITE_NC'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                                         ^
fastio.h:67:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:495:29: note: in expansion of macro 'WRITE'
   #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, v)
                             ^
temperature.cpp:1298:9: note: in expansion of macro 'WRITE_HEATER_1'
         WRITE_HEATER_1(soft_pwm_1 > 0 ? 1 : 0);
         ^
pins_RAMPS_13.h:95:31: error: invalid suffix "_PIN" on integer constant
   #define HEATER_1_PIN       -1
                               ^
fastio.h:16:28: note: in definition of macro 'MASK'
   #define MASK(PIN)  (1 << PIN)
                            ^
fastio.h:44:105: note: in expansion of macro '_WRITE_NC'
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                                                                                         ^
[... 413 lines cut ...]
                       ^
temperature.cpp:495:29: note: in expansion of macro 'WRITE'
   #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, v)
                             ^
temperature.cpp:1298:9: note: in expansion of macro 'WRITE_HEATER_1'
         WRITE_HEATER_1(soft_pwm_1 > 0 ? 1 : 0);
         ^
fastio.h:44:36: error: 'DIO' was not declared in this scope
 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)
                                    ^
fastio.h:67:23: note: in expansion of macro '_WRITE'
 #define WRITE(IO, v)  _WRITE(IO, v)
                       ^
temperature.cpp:495:29: note: in expansion of macro 'WRITE'
   #define WRITE_HEATER_1(v) WRITE(HEATER_1_PIN, v)
                             ^
temperature.cpp:1321:35: note: in expansion of macro 'WRITE_HEATER_1'
       if (soft_pwm_1 < pwm_count) WRITE_HEATER_1(0);
                                   ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
```

AnHardt, 2015-03-15:
@daprice 
It may help if you define the same pins for all the other extruders.

But yes on long term we have to distinguish better between nozzles an 'filament pushers' and have do define them separately.

scotty1024, 2015-03-15:
Just wanted to step in here and mention I’m working on implementing the DGlass3D extruders for Marlin. One configuration that makes possible is:

One DGlass3D HPX2-MAX with One Cyclops

That would be:

One stepper motor
One PID
Two filaments

The HPX2-MAX runs one direction to push one filament and the other direction to p us the other filament.

Thus I’m interested in how y’all make this particular change. :-)

AnHardt, 2015-03-15:
Forgot to say. DISABLE_INACTIVE_EXTRUDER is a bad idea then - or not? - you have to try..

thinkyhead, 2015-03-15:
@scotty1024 Well. the abstraction of an EXTRUDER in Marlin still assumes that there will be pins for a motor plus a heater plus a thermistor. The E axis is implemented as a single coordinate system, and at the planner level all it cares about is moving the axis +forward and -backward. I've just been into the `stepper.cpp` code, and it handles moving axes at the lower level. It also manages situations where there are two motors for a single axis. So it will probably be in `stepper.cpp` where you deal with the E motor being directionally reversed.

As for the other part, where "switching the extruder" is actually only switching the direction of a single extruder, that's mostly a matter of interpreting the G-code a little differently for this setup. At the higher level it will switch the extruder in the usual way, but at the lower level it will talk to only the stepper motor instead of two. The pins for a second extruder motor should just be left as-is, and they will be ignored. (The problem being that signals intended for two motors being sent on a single set of pins can cause bad things to happen.)

I would start at that low level, where the E motor has a flag so it can be reversed. (Or, use the `active_extruder` value when `defined(DGLASS)`.) Then look at the planner code for the second extruder and deal with the DGLASS exception there.

scotty1024, 2015-03-16:
@thinkyhead All good thoughts, I'm chillin' letting the Kraken changes solidify a bit then going to dive in.

I've been working on a RESTful build service for your configuration front end to call. At present you pass three arguments (motherboard ID, configuration.h contents, configuration_adv.h contents) and four seconds later you get an error message or a hex file back. Next iteration 
 you can pass in a git tag.

thinkyhead, 2015-03-16:
@scotty1024 I'm looking forward to playing with that builder and continuing to make this tool more useful. I've been putting up some content on the new site wiki, and over the coming months I will move over all the documentation we have at present. I'd like to do some design mods on the wiki to put up the Marlin logo and make it look a little less generic and more readable on big screens, if there's some way I can get SFTP access.

scotty1024, 2015-03-16:
@thinkyhead I guess github filtered out my last attempt at getting you my email address.

Scotty1024 the at sign thingy the domain "me" with a TLD of "com".

I will happily set you up with accounts.

scotty1024, 2015-03-17:
@thinkyhead Well email doesn’t seem to be working. I’ve added an account for you if you can get to it before anyone else does. :-)

Username “scottl” password “scottl". You will be required to change the password upon first login.

Let me know of any permissions you need tweaked.

If someone else gets to it before you let me know and I will kick them out and we can try again.

MagoKimbra, 2015-03-17:
When I can I will build my firmware with the function SINGLENOZZLE. This function enables all calls to go to the only temperature hotend said. So you can have up to four extruders and a single hotend.

boelle, 2015-03-18:
Hmm... what happened to OP ?

is this an issue and if so can it be confirmed by others?

thinkyhead, 2015-03-19:
@scotty1024 I'm in no major hurry so no need to post info here, I'm juggling a lot of things. Let me email you now. I was not able to ssh into the domain with that information.

thinkyhead, 2015-03-19:
Oh wait, n/m I did get it - SSH not SFTP. Good.

daprice, 2015-03-19:
@boelle I’ve been busy and haven’t had a chance to try @AnHardt's workaround of setting the same pins for all extruders yet, but I’ll report back here when I do.

It seems like it could cause some issues since multiple PID loops would be trying to control the same heater, though.

AnHardt, 2015-03-19:
@daprice 
I admit - not one of my best ideas.

thinkyhead, 2015-03-20:
@daprice Quite right. If you set more than one extruder to use the same pins the results will be unpredictable. I killed a stepper driver by mistake simply by setting the STEP and DIR pins for more than one axis to the same values. It would be better to wait until there's a specific feature to support the configuration. For multi extruders that share a single heater, this will be similar to the various "DUAL" options. For something like a DGlass, it will be a little more involved, but along similar lines.

daprice, 2015-03-22:
I was able to work around the limitation for now by defining the same thermistor pin for all extruders (so Marlin would *think* they are all at the same temperature and still be able to detect and prevent cold extrusion), and an unused pin for heaters >=1 so commands to those heaters don't actually affect anything.

As long as all the extruders are set to the same temperature, this works fine. It might be nice to be able to set each extruder to a different temperature and then have Marlin use the temperature setting from whichever tool is active, but that isn't critical.

thinkyhead, 2015-03-28:
#1686 implements the code for a single nozzle+heater+sensor having two extruder motors. I'd still like to find a more general solution, that will allow defining which motors go with which nozzle+heater+sensor(s). I don't think the code can (yet) deal with something like the Kraken.

emartinez167, 2015-03-28:
Neither does it deal with the opposite case with two extruders and a single
motor like the DGlass HPX2-MAX.

thinkyhead, 2015-03-28:
@emartinez167 The DGlass is quite a special case, with a Servo to switch nozzles, and the motor reversing direction for the second nozzle. Someone who has one of these will have implemented it already. I think I'll hit up the guy on YouTube who posted a video of this setup working and see if his code is posted someplace.

MagoKimbra, 2015-03-28:
This too I had done in my firmware when we started printing in 4 colors with a single hotend. The first lead to 4 extruders, the second was that of singlenozzle and finally the third virtualize extruders.
For example I have 4 extruders (4 motor) but only two drivers, via relays controlled by a pin address the 4 outputs of a driver with two engines. In addition to firmware so I extruder, hotend and driver. So I have 4 extruders, Hotend 1 and 2 drivers (ramps 1.4). In the case of a double extruder single motor it would be equal, 2 extruders, hotend 1, and 1 driver. The pin to switch the relay in this case should command the servo that is passed through an extruder to another, but always the same driver is controlled and therefore always the same engine ....
Unfortunately, this modification changes dramatically to level mark of software because all part of Block changes ...
If and when we solve all the current problems and thinkyhead think is the case of inserting it we can do it ...

thinkyhead, 2015-03-28:
A comprehensive approach… I think it would be worthwhile to work out what all the required variables would be to best make a generic model of the parts which correspond with one another, how and when they are engaged, etc. Just writing it up in a gist at first…

nophead, 2015-03-28:
It's crying out for an OO approach with objects for heaters, motors, feed mechanisms and nozzles, etc with associations between them represented by pointers or references. Totally the opposite of how Marlin is written at the moment.

thinkyhead, 2015-03-29:
@nophead Pointers are a good thing, encapsulating things into structures and objects is a good thing, and I definitely vote for using more of both. Right now there are several arrays initialized with `[EXTRUDERS]` elements in them. There should only be one initializer `Extruder extruder[EXTRUDERS];` encompassing all of those. (Where an Extruder object would be a combination of one or more motors with one or more nozzles, etc.)

boelle, 2015-05-31:
any news on this one?

thinkyhead, 2015-06-09:
@boelle Still a lot of preliminary work before extruders can have more than one heater or more than one stepper. I'm thinking of a simple concept, like I described on Mar 28, but where you could even have two "tool" objects refer to the same "extruder" object, but with slightly different states. So for example, Tool A might be Extruder 0 with the motor moving forward and its servo in position 0°, while Tool B would be Extruder 0 with the motor moving the other direction and its servo in position 30°. Obviously we have a lot of preliminary stuff to make before we can do that.

boelle, 2015-06-29:
Looking at  this one would it not fall in the category where we agreed that new hardware should be supported by the makers and not us?

Wackerbarth, 2015-06-29:
@boelle - I disagree. Marlin needs to have the infrastructure to permit this. Then it would be up to the manufacturers to configure that infrastructure to match their hardware.

That said, this is a feature for, perhaps, Marlin 2.0.

boelle, 2015-06-29:
of course.... you are right 100%

and 2.0 might not even be in my lifetime as the progress is now

thinkyhead, 2015-07-03:
> might not even be in my lifetime as the progress is now

Sorry, I've been super busy! I will be doing more bug fixes soon to get back in the groove.

stanrob, 2015-07-23:
Did you fix your singlenozzle multiple driver issue? 
I am very interested un this because I bought a E3D Cyclops and I can't use it

KiteLab, 2015-07-23:
@stanrob 
Please read
https://github.com/MarlinFirmware/Marlin/issues/1613#issuecomment-84699527
Until then nothing has changed in that part of the code (as far as i know).

MagoKimbra, 2015-07-23:
I repet if you want try my marlinkimbra version. It has multiply extruder and singlenozzle option...

stanrob, 2015-07-24:
Yes I try it but I may have make mistake because it don't work on my printer

thinkyhead, 2015-07-24:
@MagoKimbra How easy do you think it will be to integrate your solution into Marlin? I would like to give it a go. Have you made many other (unrelated) changes in your branch?

MagoKimbra, 2015-07-26:
@thinkyhead I try merge my solution width Marlin...
I merge only singlenozzle. Then if you want I can also merge to the steps, feedrate, Accel differentiated by single extruder. But we do one thing at a time ...

thinkyhead, 2015-07-27:
@MagoKimbra Sounds like a good approach.

stanrob, 2015-07-28:
I try it and it seems to work well. However I didn't find out how to use the 2 extrudeurs

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better.
You should re-open this topic on [MarlinFirmware/MarlinDev](https://github.com/MarlinFirmware/MarlinDev) for proper followup from the development team. We suggest you to use the `MarlinFirmware/Marlin#<topic id>` markdown to link back to this original topic.

closed 2016-05-04 by jbrazio

arvinbagheri, 2016-11-22:
Dear all,
I want to run my 3d-printer with the 2nd extruder. My board is Arduino Atmega2560 with Ramps 1.4 shield. Before this I uploaded Marlin 1.1.0 RC2 for one extruder and everything was OK. Bat now, I want to use MarlinX2-master which is design for dual extruders. After modification of configuration.h according to my printer, I tried to compile the firmware but I got this error, I don't know what does this mean, please help me to find out the problem:
"Arduino: 1.6.7 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Marlin.h:21:0,

                 from sketch\Marlin_main.cpp:30:

sketch\Marlin_main.cpp: In function 'void setup_powerhold()':

fastio.h:68: error: '_WRITE' was not declared in this scope

 #define WRITE(IO, v)  _WRITE(IO, v)

                                   ^

sketch\Marlin_main.cpp:390:4: note: in expansion of macro 'WRITE'

    WRITE(PS_ON_PIN, PS_ON_AWAKE);

    ^

sketch\Marlin_main.cpp: In function 'void process_commands()':

fastio.h:68: error: '_WRITE' was not declared in this scope

 #define WRITE(IO, v)  _WRITE(IO, v)

                                   ^

sketch\Marlin.h:119:24: note: in expansion of macro 'WRITE'

   #define enable_x0()  WRITE(X0_ENABLE_PIN, X_ENABLE_ON)

                        ^

sketch\Marlin.h:123:26: note: in expansion of macro 'enable_x0'

   #define enable_x()   { enable_x0(); enable_x1(); }

                          ^

sketch\Marlin_main.cpp:1130:9: note: in expansion of macro 'enable_x'

         enable_x(); 

         ^

fastio.h:68: error: '_WRITE' was not declared in this scope

 #define WRITE(IO, v)  _WRITE(IO, v)

                                   ^

sketch\Marlin.h:135:22: note: in expansion of macro 'WRITE'

   #define enable_y() WRITE(Y_ENABLE_PIN, Y_ENABLE_ON)

                      ^

sketch\Marlin_main.cpp:1131:9: note: in expansion of macro 'enable_y'

         enable_y(); 

         ^

sketch\Marlin_main.cpp: In function 'void manage_inactivity()':

fastio.h:68: error: '_WRITE' was not declared in this scope

 #define WRITE(IO, v)  _WRITE(IO, v)

                                   ^

sketch\Marlin.h:120:24: note: in expansion of macro 'WRITE'

   #define disable_x0() WRITE(X0_ENABLE_PIN,!X_ENABLE_ON)

                        ^

sketch\Marlin.h:124:26: note: in expansion of macro 'disable_x0'

   #define disable_x()  { disable_x0(); disable_x1(); }

                          ^

sketch\Marlin_main.cpp:2611:9: note: in expansion of macro 'disable_x'

         disable_x();

         ^

fastio.h:68: error: '_WRITE' was not declared in this scope

 #define WRITE(IO, v)  _WRITE(IO, v)

                                   ^

sketch\Marlin.h:136:23: note: in expansion of macro 'WRITE'

   #define disable_y() WRITE(Y_ENABLE_PIN,!Y_ENABLE_ON)

                       ^

sketch\Marlin_main.cpp:2612:9: note: in expansion of macro 'disable_y'

         disable_y();

         ^

sketch\Marlin_main.cpp: In function 'void kill()':

fastio.h:68: error: '_WRITE' was not declared in this scope

 #define WRITE(IO, v)  _WRITE(IO, v)

                                   ^

sketch\Marlin.h:120:24: note: in expansion of macro 'WRITE'

   #define disable_x0() WRITE(X0_ENABLE_PIN,!X_ENABLE_ON)

                        ^

sketch\Marlin_main.cpp:2660:3: note: in expansion of macro 'disable_x0'

   disable_x0();

   ^

exit status 1
'_WRITE' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
"

Blue-Marlin, 2016-11-22:
Please ask there where you got "MarlinX2-master". Does not look as if this is one of our current branches or repositories.

_WRITE() is normally defined right before WRITE() in fastio.h

arvinbagheri, 2016-11-22:
what source do you suggest?

thinkyhead, 2016-11-26:
Apparently it comes from https://github.com/dob71/MarlinX2

thinkyhead, 2016-11-26:
Marlin `RCBugFix` has support for dual extruders. You can just enable the `SINGLENOZZLE` option.