MarlinFirmware/Marlin #1623: Question of adding Language to Marlin Firmware.

WWWTH, 2015-03-17:
Dear sir

    I have a question on adding language to Marlin.  if I would like to help adding a new language to Marlin.  How should I do ? Who should I contact ?

Thank you.

AnHardt, 2015-03-17:
@WWWTH 
Nice to hear that. Right now mayor changes in the 'Language - Font - System' are pending. (#1622)
In https://github.com/AnHardt/Marlin/blob/ace3c62a1c6b3665ab9c6c694749283b8f95f2c6/Documentation/LCDLanguageFont.md i give some hints about how it works and what to do.

Adding a new language is not to complicated.

As a first step take language_en.h . Copy it to language_xx.h, where xx represents your language. Remove the soruonding #ifndef - #endif statements.
Translate Strings. You don't have to translate all strings - the remaining will show in English.
Make a pull request.

Than ill have a look on your file and give some hints how to integrate and or improve.
Maybe i will make a pull request to your branch.

At this time contacting me seems to be a good idea. But we like to do things in the public, so here is the right place.

What language do you want to integrate?
Do you need special symbols (not US-ASCII) in your language?

WWWTH, 2015-03-17:
Anhardt, Thank you for your reply.  The language that I want to add is Thai. I will try to follow the step your recommend. 

I guess, I may need special symbols for Thai language. How should I specify the special symbols ?

AnHardt, 2015-03-17:
@WWWTH 
Oh! Thay!
That's one of the more complicated cases.
Lets see.
Is there a character based display (Hitachi) you could use? If yes - please give me a link to the datasheet. Otherwise we can only use the 12864 full graphic displays.
What symbols do you want to use? Is http://www.utf8-zeichentabelle.de/unicode-utf8-table.pl?start=3584 sufficient? Or, maybe, can you narrow down the range? Ending up with 128 (or less) extra symbols in a sequence will simplify things greatly.
Can all the symbols be drawn in a 5 pixel wide, 10 pixel high matrix? Maybe 11x10 is possible with tricks
>  How should I specify the special symbols ?
Depends on the answers we get above.

AnHardt, 2015-03-17:
@WWWTH 
I hope the needed symbols are all between U+0E00 to U+0E5F. For this case I prepared an initial Thai-font-for-Marlin at https://github.com/AnHardt/Thai-for-Marlin (You have write access).

AnHardt, 2015-03-17:
@WWWTH 
Currently we are not prepared to use Thai digit symbols.

WWWTH, 2015-03-17:
@AnHardt  
The symbols that I need are in between U+0E00 to U+0E5F.  Do I need to write now in those symbols to represent thai character. Please give me some advice and What do you means by "Not prepare to use Thai digits symbols" ?

AnHardt, 2015-03-17:
@WWWTH 
You can write your language.xx.h file with strings like "กขฃค". See the kana_utf8 example at https://github.com/AnHardt/Marlin/blob/tm-utf-minus-kanji/Marlin/language_kana_utf8.h.

At the moment we use the Arabic numbers '0' to '9' to write temperatures, coordinates, ... . 
Without further changes we can not use ''๐' THAI DIGIT ZERO to '๙' THAI DIGIT NINE.

WWWTH, 2015-03-17:
Thank you. Then, I will start to translate into my language according to the file. kana_utf8 .

For the number, it is not significant. We can use 0-9 in Arabic number

AnHardt, 2015-03-17:
The actual translation makes low to no sense until we now where and how to display.

WWWTH, 2015-03-17:
Ok, let 's see. One step at time. Thank you for your advise.

AnHardt, 2015-03-17:
@WWWTH 
Is there a character based devise what could display Thai?
Have you had a look into the font? Do you think it can be mad readable in 6x10?

WWWTH, 2015-03-17:
I have get back to you on that, I am trying to connect a guy who used to do this kind of thing in Thai

boelle, 2015-03-18:
will close this one... please open new issue or post a PR when time comes

closed 2015-03-18 by boelle

bathnsahin, 2017-08-03:
Hi, i try to add arabic language in marlin. But i cant find anyway. I found ISO number but probably i need mapper for arabic. Please help me about this. #5441