MarlinFirmware/Marlin #1627: compiling board issues

tris1986, 2015-03-17:
hi i am having the following errors when compiling

Arduino: 1.6.1 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from Configuration.h:4:0,

                 from Marlin.h:22,

                 from BlinkM.cpp:5:

boards.h:59:46: error: operator '==' has no right operand

 #define MB(board) (MOTHERBOARD==BOARD_##board)

                                              ^

pins.h:39:7: note: in expansion of macro 'MB'

 #elif MB(RAMPS_13_EFB) || MB(RAMPS_13_EEB) || MB(RAMPS_13_EFF) || MB(RAMPS_13_EEF)

       ^

In file included from Marlin.h:23:0,

                 from BlinkM.cpp:5:

pins.h:122:4: error: #error Unknown MOTHERBOARD value set in Configuration.h

   #error Unknown MOTHERBOARD value set in Configuration.h

    ^

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

boelle, 2015-03-17:
closing... dublicate entry

closed 2015-03-17 by boelle

AnHardt, 2015-03-17:
Dear @boelle 
Please read the issues. This one is entirely different from #1628.
Here he miss-typed the board-name. This was resolved by him self.
Else he couldn't open a new ticked (an hour later) where he didn't respect the need for integers in the z-probe offset.

So. Closing this ticked was ok - but for a completely different reason.

avluis, 2015-03-18:
@AnHardt agreed.

boelle, 2015-03-18:
I just tried to compare the 2 tickets and they looked the same to me minus the  lenght... so i was just to quick to judge and stamped it a dublicate