MarlinFirmware/Marlin #1628: compling board issues auto bed level stable 1.02

tris1986, 2015-03-17:
Arduino: 1.6.1 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from Marlin.h:22:0,

                 from BlinkM.cpp:5:

Configuration.h:397:41: error: floating constant in preprocessor expression

   #define X_PROBE_OFFSET_FROM_EXTRUDER -21.60

                                         ^

Configuration.h:437:9: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

     #if X_PROBE_OFFSET_FROM_EXTRUDER < 0

         ^

Configuration.h:397:41: error: floating constant in preprocessor expression

   #define X_PROBE_OFFSET_FROM_EXTRUDER -21.60

                                         ^

Configuration.h:442:13: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

       #if ((X_PROBE_OFFSET_FROM_EXTRUDER * AUTO_BED_LEVELING_GRID_POINTS) >= (RIGHT_PROBE_BED_POSITION - LEFT_PROBE_BED_POSITION))

             ^

Configuration.h:398:41: error: floating constant in preprocessor expression

   #define Y_PROBE_OFFSET_FROM_EXTRUDER -3.80

                                         ^

Configuration.h:446:9: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

     #if Y_PROBE_OFFSET_FROM_EXTRUDER < 0

         ^

Configuration.h:398:41: error: floating constant in preprocessor expression

   #define Y_PROBE_OFFSET_FROM_EXTRUDER -3.80

                                         ^

Configuration.h:451:13: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

       #if ((Y_PROBE_OFFSET_FROM_EXTRUDER * AUTO_BED_LEVELING_GRID_POINTS) >= (BACK_PROBE_BED_POSITION - FRONT_PROBE_BED_POSITION))

             ^

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

AnHardt, 2015-03-17:
@tris1986 
Please read: https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Configuration.h#L434

boelle, 2015-03-17:
also stable 1.02

we don't have a stable version yet so we assume you use 1.0.2

you should use the latest code as there are many fixes: https://github.com/MarlinFirmware/Marlin/archive/Development.zip

avluis, 2015-03-18:
@tris1986 as @AnHardt mentioned - keep your X & Y probe offsets as integers (no decimal values, no need to be precise).
Z offset can be as precise as needed - and you can tweak it on your slicer software as well.

boelle, 2015-03-18:
oki doki :-D

closed 2015-03-18 by boelle