MarlinFirmware/Marlin #1629: 3-points auto bed leveling issue - invalid conversion from 'int' to 'ProbeAction'

Maik93, 2015-03-17:
Hi,
I've found out an issue while setting up Z probing in 3-points mode:

Marlin_main.cpp: In function 'void gcode_G29()':
Marlin_main.cpp:2844: error: invalid conversion from 'int' to 'ProbeAction'
Marlin_main.cpp:2844: error:   initializing argument 4 of 'float probe_pt(float, float, float, ProbeAction, int)'
Marlin_main.cpp:2845: error: invalid conversion from 'int' to 'ProbeAction'
Marlin_main.cpp:2845: error:   initializing argument 4 of 'float probe_pt(float, float, float, ProbeAction, int)'
Marlin_main.cpp:2846: error: invalid conversion from 'int' to 'ProbeAction'
Marlin_main.cpp:2846: error:   initializing argument 4 of 'float probe_pt(float, float, float, ProbeAction, int)'

I've momentary bypassed it as follows:
if (1){ //if(enhanced_g29) {
  z_at_pt_1 = probe_pt(ABL_PROBE_PT_1_X, ABL_PROBE_PT_1_Y, Z_RAISE_BEFORE_PROBING, ProbeEngage, verbose_level);
  z_at_pt_2 = probe_pt(ABL_PROBE_PT_2_X, ABL_PROBE_PT_2_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, Probeverbose_level);
  z_at_pt_3 = probe_pt(ABL_PROBE_PT_3_X, ABL_PROBE_PT_3_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, ProbeRetverbose_level);
}
/*else {
  z_at_pt_1 = probe_pt(ABL_PROBE_PT_1_X, ABL_PROBE_PT_1_Y, Z_RAISE_BEFORE_PROBING, verbose_level);
  z_at_pt_2 = probe_pt(ABL_PROBE_PT_2_X, ABL_PROBE_PT_2_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, verbose_level);
  z_at_pt_3 = probe_pt(ABL_PROBE_PT_3_X, ABL_PROBE_PT_3_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, verbose_level);
}*/

avluis, 2015-03-18:
I can reproduce this if I have ENABLE_AUTO_BED_LEVELING defined and AUTO_BED_LEVELING_GRID disabled.

I do not have any grid points to add for manual probing points, so I used defaults - but it does demonstrate that the function void gcode_G29() needs to handle this if not using the grid method for ABL.

--
Arduino: 1.6.1 (Mac OS X), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

[non-error verbose redacted]

/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build9086982035705764180.tmp/Marlin_main.cpp:1988:8: error: #error "The given ABL_PROBE_PT_1_X can't be reached by the probe."
       #error "The given ABL_PROBE_PT_1_X can't be reached by the probe."
        ^
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build9086982035705764180.tmp/Marlin_main.cpp: In function 'void gcode_G29()':
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build9086982035705764180.tmp/Marlin_main.cpp:2333:103: error: invalid conversion from 'int' to 'ProbeAction' [-fpermissive]
         z_at_pt_1 = probe_pt(ABL_PROBE_PT_1_X, ABL_PROBE_PT_1_Y, Z_RAISE_BEFORE_PROBING, verbose_level);
                                                                                                       ^
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build9086982035705764180.tmp/Marlin_main.cpp:1369:14: error:   initializing argument 4 of 'float probe_pt(float, float, float, ProbeAction, int)' [-fpermissive]
 static float probe_pt(float x, float y, float z_before, ProbeAction retract_action=ProbeEngageRetract, int verbose_level=1) {
              ^
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build9086982035705764180.tmp/Marlin_main.cpp:2334:132: error: invalid conversion from 'int' to 'ProbeAction' [-fpermissive]
         z_at_pt_2 = probe_pt(ABL_PROBE_PT_2_X, ABL_PROBE_PT_2_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, verbose_level);
                                                                                                                                    ^
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build9086982035705764180.tmp/Marlin_main.cpp:1369:14: error:   initializing argument 4 of 'float probe_pt(float, float, float, ProbeAction, int)' [-fpermissive]
 static float probe_pt(float x, float y, float z_before, ProbeAction retract_action=ProbeEngageRetract, int verbose_level=1) {
              ^
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build9086982035705764180.tmp/Marlin_main.cpp:2335:132: error: invalid conversion from 'int' to 'ProbeAction' [-fpermissive]
         z_at_pt_3 = probe_pt(ABL_PROBE_PT_3_X, ABL_PROBE_PT_3_Y, current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS, verbose_level);
                                                                                                                                    ^
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build9086982035705764180.tmp/Marlin_main.cpp:1369:14: error:   initializing argument 4 of 'float probe_pt(float, float, float, ProbeAction, int)' [-fpermissive]
 static float probe_pt(float x, float y, float z_before, ProbeAction retract_action=ProbeEngageRetract, int verbose_level=1) {
              ^
Error compiling.

Wurstnase, 2015-03-18:
retract_action and verbose_level are optional parameters and should be named when used unordered.
I have fixed it already, Just need to merge it.

boelle, 2015-03-18:
looking forwared to the PR

plese mention this issue in the PR

Roxy-3DPrintBoard, 2015-03-18:
Is there a reason to have the 3 point leveling?   It would seem doing a grid with a 2 x 2 size is pretty close to the 3 point leveling but without all the problems (inaccuracies) in the unprobed corner?   If the goal is to simplify the code and configuration of the code, it would seem the 3 point leveling should be pulled???

boelle, 2015-03-18:
you only need 3 points as all what is done is correcting the plane

adding more points will just be a waste

Wurstnase, 2015-03-19:
Grid leveling is the one point in Marlin which makes it unique.

And again, we dont life in a perfect world. So using least squares methode is much more accurate.

boelle, 2015-03-19:
hmm... grid leveling or ore probe points does not matter

people tend to think that the more points the better, marlin does not compensate for a bend plate, 

it only compensates for at "tilted" plate...... and for that 3 points will do it

Wurstnase, 2015-03-19:
It doesn't compensate, but it minimized the error.

A 2D example for that.

blue: measured values
green: something like 3point leveling
red: least square methode

Least squares: y = 0.96*x + 1.16
2-Point: y = 0.9667*x + 1.1

[image]

Wurstnase, 2015-03-19:
Btw. this issue should be solved with #1630

closed 2015-03-19 by boelle