MarlinFirmware/Marlin #1632: Z_RAISE_AFTER_PROBING not declared in current scope when compiling with ENABLE_AUTO_BED_LEVELING and AUTO_BED_LEVELING_GRID

MaxOLydian, 2015-03-18:
When compiling with ENABLE_AUTO_BED_LEVELING and AUTO_BED_LEVELING_GRID defined, the following error occurs:

    Marlin_main.cpp: In function 'void gcode_G29()':
    Marlin_main.cpp:2342:77: error: 'Z_RAISE_AFTER_PROBING' was not declared in this scope
         do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], Z_RAISE_AFTER_PROBING);
                                                                                 ^
Temporary fix:
Adding

    #define Z_RAISE_AFTER_PROBING 5.0

to Configuration.h in auto bed levelling section allows proper compilation.

Related issue #1629- I encountered this error while compiling the first time with AUTO_BED_LEVELING_GRID not defined.  Switched to defining AUTO_BED_LEVELING_GRID which eliminated issue #1629, but produced this issue.

avluis, 2015-03-18:
This should had been added in Configuration.h by git when you merged.
Now, the related issue seems like a bigger issue - was ENABLE_AUTO_BED_LEVELING defined or not?

thinkyhead, 2015-03-19:
`Z_RAISE_AFTER_PROBING` is a new addition to `Configuration.h` so you can keep that as a permanent change.

closed 2015-03-19 by thinkyhead

MaxOLydian, 2015-03-19:
@avluis: I just grabbed the zip file and opened in Arduino IDE- I'm new to git and not yet confident with forks, merges, pulls, etc.  I modified Configuration.h for my printer, removed the comment in front of ENABLE_AUTO_BED_LEVELING, and commented out AUTO_BED_LEVELING_GRID in order to try the 3-point method first.  That compilation gakked on another error (issue #1629, already reported).  I then defined AUTO_BED_LEVELING_GRID and encountered the Z_RAISE_AFTER_PROBING issue.

Thanks for the responses- and if replying to a closed issue is poor form, please forgive me!

avluis, 2015-03-19:
@MaxOLydian It's all good (helps to close loose ends, gather additional info, etc) as long as we keep in on topic (within certain reasons) but not frowned upon if it helps discover any new issues (for which a new issues would have to be opened) or reopens the current issue (if a reported fix does not fix the original issue at hand).

While off topic - I really recommend learning git, GitHub, how to fork, creating branches, etc. Will help you keep your copy of Marlin updated (dev is always a moving target) and can help you figure out a configuration error (from your part) or an actual issue.