MarlinFirmware/Marlin #1633: Compilation error when selecting the russian language for reprap discount graphics controller

SkyFlyer2, 2015-03-18:
The code not compile - we get an error

language_ru.h:47:45: error: 'LCD_STR_THERMOMETER' was not declared in this scope
 #define MSG_BED                             LCD_STR_THERMOMETER " РљСЂРѕРІР°С‚СЊ"

Selection any other possible language is ok - successful compilation.

Ramps 1.4 with mega 2560 & Reprap Discount Full Graphics Controller

GrandK, 2015-03-19:
Change  'LCD_STR_THERMOMETER'  to  'STR_THERMOMETER'  in language_ru.h

SkyFlyer2, 2015-03-19:
Now compiled. However, no Russian characters, special symbols instead.

AnHardt, 2015-03-19:
@SkyFlyer2 
Russian language support needs special hardware (http://store.comet.bg/download-file.php?id=466) to work on 20x4  displays.
When #1622 is applied Russian language will be supported for all the 12864 displays.

SkyFlyer2, 2015-03-19:
Thanks, now it is clear for me. Will wait, is not so important.

thinkyhead, 2015-03-28:
The Big Language Pack has been merged, and the code seems very stable. Most likely this issue is all fixed!

closed 2015-03-31 by thinkyhead

SkyFlyer2, 2015-03-31:
This issue was not solved completely, the code still not compile.

If I comment this line - the code can be compiled. The cyrillic characters are present and ok.

#define MSG_BED                             LCD_STR_THERMOMETER " Кровать"

AnHardt, 2015-03-31:
@SkyFlyer2 
I'm sorry but i can't reproduce the error. Could you please load up your Configuration.h and the, maybe defect, language file somewhere.

SkyFlyer2, 2015-03-31:
Sorry, i used the old pre-configured file "configuration.h" with my printer settings, from previous commit. 

Now it is ok, the code can be compiled, i am used the configuration.h from last commit.

What to change to get the small fonts?

AnHardt, 2015-03-31:
@SkyFlyer2 
Ether comment in line 46 in 'dogm_lcd_implementation.h' or 
#define USE_SMALL_INFOFONT 
somewhere in your 'Configuration.h'.

AnHardt, 2015-03-31:
@SkyFlyer2 
How many of your other problems are solved by the new Configuration.h? :-)

SkyFlyer2, 2015-03-31:
Авто уровень по-прежнему не функционируют.)

Трудно с каждым новым совершить двигаться настройки принтера на новый configuration.h

SkyFlyer2, 2015-03-31:
Auto level is still non-functional.)

It is hard with each new commit move the printer settings to the new configuration.h

AnHardt, 2015-03-31:
@SkyFlyer2 
Pardon? I can't read Cyrillic nor i understand Russian.

SkyFlyer2, 2015-03-31:
Now in English)

Auto level is still non-functional, everything else looks fine.
It is hard with each new commit move the my printer settings to the new configuration.h

AnHardt, 2015-03-31:
Oh, sorry.
I expected something like a minor error in the language file. :-)

thinkyhead, 2015-04-01:
@SkyFlyer2 For updating configs (and other files) a good diff-and-merge tool is essential.