MarlinFirmware/Marlin #1634: Cannot reach target temperature beyond 260 deg C even though Configuration.h says so

emartinez167, 2015-03-18:
I am experiencing an issue. With an extruder quite capable of reaching over 400 deg C, I set the target temp to anything above 260 and the extruder just seems to turn off and cycles up and down in a repetitive pattern. I enabled debugging in Configuration.h by uncommenting the following line:

#ifdef PIDTEMP
  #define PID_DEBUG // Sends debug data to the serial port.

This is what debug writes out:
< 5:21:15 pm: echo: PID_DEBUG 0: Input 269.38 Output 255.00 pTerm 0.00 iTerm 0.00 dTerm 0.00
< 5:21:15 pm: echo: PID_DEBUG 0: Input 270.16 Output 51.41 pTerm 98.44 iTerm 1.65 dTerm 48.68
< 5:21:16 pm: echo: PID_DEBUG 0: Input 269.53 Output 255.00 pTerm 98.44 iTerm 1.65 dTerm 48.68
< 5:21:16 pm: echo: PID_DEBUG 0: Input 269.84 Output 255.00 pTerm 98.44 iTerm 1.65 dTerm 48.68
< 5:21:16 pm: ok T:269.8 /280.0 B:22.2 /0.0 T0:269.8 /280.0 T1:32.1 /0.0 @:127 B@:0
< 5:21:16 pm: echo: PID_DEBUG 0: Input 269.84 Output 255.00 pTerm 98.44 iTerm 1.65 dTerm 48.68
< 5:21:16 pm: echo: PID_DEBUG 0: Input 270.16 Output 34.37 pTerm 98.44 iTerm 1.65 dTerm 65.71
< 5:21:16 pm: echo: PID_DEBUG 0: Input 270.00 Output 50.62 pTerm 100.00 iTerm 3.32 dTerm 52.69
< 5:21:17 pm: echo: PID_DEBUG 0: Input 270.00 Output 54.93 pTerm 100.00 iTerm 4.99 dTerm 50.06
< 5:21:17 pm: echo: PID_DEBUG 0: Input 270.16 Output 47.78 pTerm 98.44 iTerm 6.63 dTerm 57.29

I would like to understand if this is an issue with the thermistor, or is there something else going on?
[image]

Wurstnase, 2015-03-18:
What kind of thermistor do you use?

emartinez167, 2015-03-18:
According to the documentation for the HPX2-MAX and self retracting extruders from GDlass3D, they are type 5:

// 5 is 100K thermistor - ATC Semitec 104GT-2 (Used in ParCan & J-Head) (4.7k pullup)

#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_1 5

Do they have some physical limitation?

Regards,

Ernesto.

Wurstnase, 2015-03-18:
Sure. Max temp for this sensor is 300°. Do you autotune your hotend for that temperature with M303?
http://www.atcsemitec.co.uk/gt-2-glass-thermistors.html

emartinez167, 2015-03-18:
I had tried before, but that led me to the now closed case of the runaway temps. I was trying to disable the runaway because when I tried to do the autotune the printer shut down due to runaway. Now that it has been resolved, I will attempt to do the autotune again.

Regards,

Ernesto

boelle, 2015-03-18:
let us know how that goes so we can either close this one or upscale it to a potential and later verified issue

emartinez167, 2015-03-18:
Will do tonight when I get back home.

Regards,

Ernesto

emartinez167, 2015-03-19:
I have done the PID auto tune , recompiled with the new values and the
behavior is still there, even trying with the second extruder. I am
suspecting the thermistors might not be up to the task as I can see that at
some point near the peak of the thermal curve the behavior becomes erratic.
Do you guys favor any thermistor model/brand in particular or would
switching to a different measuring device help?

emartinez167, 2015-03-23:
Could this be related to the Temperature issues being experienced under #1209?

Wurstnase, 2015-03-23:
No, if you can't reach the temperature then there is a.) the heater to weak, b.) the power supply to weak, c.) the thermistor damaged or d.) you made some max_pid configs. I tend to b or d, because your curve is really flat at the end. Normally it should overshot by 10 to 20%.

emartinez167, 2015-03-23:
I have seen rather erratic curves lately (spotty, and jumpy) which led me
to think the thermistor was at fault. I'll check power consumption .

thinkyhead, 2015-03-23:
@Wurstnase @emartinez167 If you can help figure out the most recent date when this problem didn't exist, I will make a tag for that and then explore the diff closely!

Wurstnase, 2015-03-24:
@emartinez167 could post his config.h. But @thinkyhead i don't think there is a problem with the code. I have the same thermistor and my extruder reached without any problem more than 270°. Well, I'm using a Due, but the issue should then be the same.

emartinez167, 2015-03-24:
I am also suspecting it is a hardware issue. Stuck for now as I need to order new thermistors, and going to a remote site for work until Friday. Hope to get to work in this over the weekend.

Regards,

Ernesto

AnHardt, 2015-03-24:
@emartinez167 
If you have current Marlin and compile it with Arduino 1.0.6  - its most likely not a defect thermistor but a bug in Marlin.

emartinez167, 2015-03-24:
Currently compiling with 1.6.1. I will try 1.0.6 later when I come back
from my trip.

Wurstnase, 2015-03-24:
@AnHardt Which bug you are talking about?
The heater have not enough power. You can see that on the picture in the open post.
My guess again, the power supply have not enough power or the power is limited in the config.h.
[Here](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Configuration.h#L175) or [here](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Configuration.h#L176)

emartinez167, 2015-03-24:
Manufacturer confirmed the thermistor can only go up to 300 deg so whatI
think we see in the thermistor failing (reading becomes erratic beyond a
certain point do it could be damaged?)

AnHardt, 2015-03-24:
@Wurstnase 
See #1634.
@emartinez167 
Don't try 1.0.6.

Wurstnase, 2015-03-24:
@AnHardt This is #1634?!?

AnHardt, 2015-03-24:
F..! #1660

emartinez167, 2015-03-24:
LOL!

Regards,

Ernesto

emartinez167, 2015-03-24:
I have been following #1660... So there is no resolution yet?

I will still put new thermistors when I get back, just to be sure we have a good reference.

Regards,

Ernesto

nophead, 2015-03-25:
It just looks like bad PID tuning causing it to oscillate to me.

I think maybe the lack of ADC resolution at high temperatures may upset the PID stability because steps in the temperature will create large D values. What happens if you set kD to zero?

thinkyhead, 2015-03-28:
Awaiting more feedback on this and #1695 since the code has been patched to fix up some buffer overruns and other issues.

emartinez167, 2015-03-28:
I just finished applying the changes and compile went good; however I have to go out. I will report back in a few hours.

Regards,

Ernesto

emartinez167, 2015-03-28:
I can now print. I am still having an issue where the PID autotune is failing due to high temperature, but I suspect it is a bad thermistor. This can be closed as far as I can tell. I will test with a new thermistor (in fact, I got an E3D v6 incoming next week) so I will tell you once I get a chance to test against that one.

Regards,

Ernesto

closed 2015-03-28 by thinkyhead