MarlinFirmware/Marlin #1641: Improve the management of the printer type

Drakelive, 2015-03-18:
Hi

I like the work done by #MagoKimbra (https://github.com/MagoKimbra/MarlinKimbra/blob/master/MarlinKimbra/Configuration.h) on his Marlin Fork 
I'm sure this greatly simplifies the configuration of a printer
He define a type printer :

/***********************************************************************\
**************************** Define type printer **********************
***********************************************************************/
#define CARTESIAN
//#define COREXY
//#define DELTA
//#define SCARA
#if defined(CARTESIAN)
      #include "Configuration_Cartesian.h"
#elif defined(COREXY)
     #include "Configuration_Corexy.h"
#elif defined(DELTA)
     #include "Configuration_Delta.h"
#elif defined(SCARA)
     #include "Configuration_Scara.h"
#endif

You think that you can implement this approach in MarlinFirmware?

Regard 
Drake

Wurstnase, 2015-03-18:
A third file to maintain? I don't see any benefit.

boelle, 2015-03-18:
me neither.... and we are going for a web interface to do config anyway.... so this one is a no go i think

thinkyhead, 2015-03-19:
I do understand the appeal, but I agree that fewer config files is better. That said, some of the code in the configuration files is not meant to be edited by users (code that conditionally sets defines for LCDs and things like that), and that code might end up in an additional file along with some #if / #error blocks that make sure configuration options are within bounds and throw errors, to catch more potential problems at compile-time.

closed 2015-03-19 by thinkyhead

alexborro, 2015-03-19:
Completely agree!

All checks and non-editable defines must be moved to another file.

Scott, any suggestions for this new file name?

Cheers

Alex

boelle, 2015-03-19:
why not use the web config?

less confusion factor, but of course it has to be completed first

AnHardt, 2015-03-19:
How about 'postConfiguration.h'?
Could be called at the very end of Configuration.h, right after the inclusion of thermistortables.h

Wurstnase, 2015-03-19:
Call it sanityChecks.h or something like this so normal users don't want to look at it.

alexborro, 2015-03-19:
I agree to SanityCheck.h...

thinkyhead, 2015-03-19:
I'll get the ball rolling on this with a fairly simple PR shortly.

thinkyhead, 2015-03-20:
#1651 is my first crack at moving invariant configuration code into a separate file (`Conditionals.h`) and moving most error-checking into its own file (`SanityCheck.h`). This reduces the size of the configuration files significantly.

thinkyhead, 2015-03-20:
#1651 takes the approach of moving invariant configuration code into one file and sanity checks in another. This reduces the size of the configurations significantly, resulting in 3200 fewer lines of code to maintain.

AnHardt, 2015-03-20:
>  fairly simple PR
:-)