MarlinFirmware/Marlin #1644: OT: Cant home Z

boelle, 2015-03-18:
using this fork to dial in and get numbers etc:

https://github.com/buserror/Marlin

and my config is here: http://pastebin.com/x0Zs5Zzu

one strange thing... : I cant home Z

i use pronterface and i can home X and Y... i can move Z manual up and down... turn bed on and off... extruder same... extruder fan also...

i have checked endstops with M114 and they all work

one thing i noted also... when i jog manual upwards to hit the endstop i cant move it down again

its a mendel90 hence top mounted  endstop..... anyone that can help me figure why G28 Z does not work or why hitting home Z in pronterface does not work either?

i know its an fork but its what is know to work on the OMC controller... i wanted to get it to work so i have a base to compare from should marlin not work...

thinkyhead, 2015-03-19:
It sounds like the homing direction might be contrary to the endstop position...?

boelle, 2015-03-20:
you are on to something,

just got up and it hit me that the board does not have connectors for max endstop... just endstops.

so it might not even be possible on a mendel90 where endstop is top mounted.

i put OT in the subject at i count this as an low priority thing... i will be able to print fine... i just need to zero it before each session

but i will have another eyeball at the config and keep in mind that there is no difference between max and min endstop...

boelle, 2015-03-20:
you might wonder i have set home pos to 0 and using top mounted endstop for Z

it was the only way i could get it somewhat working..  prepping to get out the door for work

but when home i have this in mind

try and change #define INVERT_Z_DIR true
ie set it to false.

leave this one as is: #define Z_HOME_DIR 1

alse leave this one as is: #define Z_MAX_LENGTH 391.40

change this one: #define Z_HOME_POS 0
i will set it to: Z_MAX_LENGTH
so that it will always be the same as the other

thinkyhead, 2015-03-20:
Endstop voodoo... :chicken:

boelle, 2015-03-20:
very strange at least

just wanted to have this working so i have a base to compare from

boelle, 2015-03-20:
that idea is out the window.... now directions have reversed... ie + in pronterface makes it go down

got in a state where it can home Z, but travel are reversed....

pastebin in first post are updated.... 

any ideas as why it keeps teasing me? was the version of marlin i link to made to use top mounted endstops?

nophead, 2015-03-20:
You need to change the pin definitions to make your limit switch a Z_MAX and define the Z_MIN to be -1.

boelle, 2015-03-20:
So i just change place on these 2 ?
#define Z_MIN_PIN           59
#define Z_MAX_PIN          -1

nophead, 2015-03-20:
Yes and have the home pos equal to the max and the homing direction
positive.

boelle, 2015-03-20:
that one nailed it :-D

thanks a bunch to @nophead

closed 2015-03-20 by boelle