MarlinFirmware/Marlin #1645: Graphic LCD support makes Marlin too big for 128k boards.

unlimitedbacon, 2015-03-18:
Without graphic LCD support enabled (for instance, for the RepRapDiscount LCD or the VIKI 2.0), Marlin compiles to about 80 kib. With it enabled, it compiles to about 135 kib. This is just slightly too large to fit on a 128 kib board. With a little bit of optimisation, we could probably get this small enough to fit.

Many boards are moving to 256 kib, but some prefer to use the AT90USB series of MCUs with integrated USB support. Atmel does not make one of these with more than 128k of flash.

AnHardt, 2015-03-18:
@unlimitedbacon 
You are using a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER?
Not a REPRAP_DISCOUNT_SMART_CONTROLLER.
The name is just to long and to similar - i know.

To save about 5441 Bytes right now apply #1621, go to dogm_lcd_implementation.h and undefine USE_BIG_EDIT_FONT.

AnHardt, 2015-03-18:
@unlimitedbacon 
If you have no't updated to Arduino 1.6.1 try that and save another about 3.5K.

With plain Marlin standard Configuration.h just defined in REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
and the above tweaks i'm compiling today to 115,678 bytes.

chris-bo, 2015-03-19:
Commenting line 9 in ultralcd_st7920_u8glib_rrd.h (#pragma GCC optimize (3)) saves another 10K!
Is there any reason for changing optimizations in this file? I noticed no issues.

mr-sneezy, 2015-03-19:
Hi, I'm one the Marlin users effected by code size once graphic LCD support is enabled on a recently bought Melzi 2.0 board. In my case it was just by a few bytes... 
I did compile with a late version of Arduino IDE.

There are a few suggestions above now, which one is the least useful to have do you think, or the first one that should be commented out to save bytes ?
Martin

AnHardt, 2015-03-19:
@mr-sneezy 
The above mentioned #1621 is already in since this morning.
BIG_EDIT_FONT is close to useless, but helps if you need eyeglasses otherwise.

Other changes (1622) to the display-font-language system are pending, but memory consumption will be about the same.

AnHardt, 2015-03-19:
@chris-bo 
Indeed! I got about -8000 bytes - but only with Arduino 1.6.1.

This has to be tested. Will affect overall timing. Could be critical for, for example, the fast moving deltas.

alexborro, 2015-03-19:
Increasing the Graphic LCD processing time is waayyy critical!!! Even in
cartesians bots. If you are sure about that, use a conditional #define.

Cheers.

Alex.

AnHardt, 2015-03-19:
@alexborro 
I'm far from turning at this wheel. My last message was ought to be warning.

AnHardt, 2015-03-19:
Hm.
Had a closer look. Somewhere i had seen a warning about ignoring GCC optimization.

With Arduino 1.0.6 i get:
/ultralcd_st7920_u8glib_rrd.h:9: warning: ignoring #pragma GCC optimize

Is upgrading to Arduino 1.6.1 the way to get enough speed for the fast deltas?

Is commenting out the way to get the speed we are used to - but saving 8KB?

How can a test look like?

thinkyhead, 2015-03-20:
I just recently merged a PR that reduces the menu code by ~6 or 7 KB, so that may also help. We're always looking for more ways to shave off precious bytes…

ntoff, 2015-03-21:
Mentioned it elsewhere, not sure if it was answered (sometimes github doesn't seem to inform me of responses) but maybe get rid of the animated fan logo? I don't see a reason to have the little spinny fan icon. Apologies if it's already been removed, I'm still using Marlin from a month or 3 ago.

Roxy-3DPrintBoard, 2015-03-21:
> I just recently merged a PR that reduces the menu code by ~6 or 7 KB, so that may also help. We're always looking for more ways to shave off precious bytes…

If they have the Auto Bed Leveling turned on, we can easily prune 25Kb of code by re-writing the Least Squares Fit algorithm.   It is solving for the general case and we don't need that.

thinkyhead, 2015-03-21:
@Roxy-3DPrintBoard Do you feel like having a crack at whittling that algorithm down? If we can eliminate the need for the `qr_solve.*` files that would be a nice extra benefit.

alexborro, 2015-03-21:
@Roxy-3DPrintBoard , that would be nice to optimize the Least Square Fit algorithm to our needs. Do you know somebody who could help on this??

@thinkyhead , we will need to store the algorithm somewhere else.. I think it don't matter if in a separate file or among other stuffs.

Cheers.

Alex.

Roxy-3DPrintBoard, 2015-03-24:
> @Roxy-3DPrintBoard , that would be nice to optimize the Least Square Fit algorithm to our needs. Do you know somebody who could help on this??

I'm toying with this right now.  It is an interesting problem!   Right now, I'm assuming that a 1 or 2 second time to do a very accurate Bed Level Correction matrix is acceptable if it can shrink the code footprint significantly.   But I would like to hear confirmation of that.

alexborro, 2015-03-24:
Even 3 seconds are nothing if just once in the print beginning..

brainscan, 2015-03-24:
I'd be happy if it took longer than that, it's peanuts compared to the time most jobs take to print. Also if it improves quality an extra few seconds is nothing.

Roxy-3DPrintBoard, 2015-03-25:
Score!!!  I need to do much more testing.  But it looks like it is possible to get a better solution for the Bed Level Correction Matrix than the QR_SOLVE.CPP package provides.   And at a much reduced code foot print!!!

nophead, 2015-03-25:
Sounds impressive but by what quality is the result better than a least squares fit. Are you minimising a different quality function?

Roxy-3DPrintBoard, 2015-03-25:
There are multiple solutions to describe the plane through the sample set of points.   The plane is described by the equation ax+by+cz+d=0.    As you adjust things, the summation of the least square error rises and falls.  There are many peaks and valleys.   The QR_SOLVE.CPP package is not finding the most optimized 'minimum error'.   It is just finding a very good one.    

And there is another problem...  When it solves the matrix of equations, it returns an optimized value (for the solution that it found) for d.   Right now, in the code, the Z Offset is being set to what ever is in the Configuration.h file.   But as it turns out, when doing the Least Squares Best Fit, that number needs to shift a little bit.  The QR_SOLVE package returns the more optimized number but it is being ignored.  The current code just does     current_position[Z_AXIS] = zprobe_zoffset;    It should be using the 'better' number.

nophead, 2015-03-25:
Ah, I see, thanks for the explanation.

I had also wondered about the validity of setting Z to zprobe_offset. My original implementation uses -d / c for the Z value at the origin.

Roxy-3DPrintBoard, 2015-03-25:
> I had also wondered about the validity of setting Z to zprobe_offset. My original implementation uses -d / c for the Z value at the origin.

Oh Cool!  I didn't know you did the original implementation!  Right now I'm playing with different data sets and I've found one where it can optimize the least squares fit very well but it has to move d way too much to be usable.   So what that tells me is there has to be some limitation built in to limit how much d can move.   In the very original code it actually used the bed level correction matrix to calculate how much different the numbers were because they were offset from the nozzle.  That isn't being done any more.  I suspect it should be.

But anyway...  I'm having a good time playing with it!   :)

nophead, 2015-03-25:
I did the original implementation on my own hardware and software: http://hydraraptor.blogspot.co.uk/2011/04/auto-bed-leveling.html. It's done in the host rather than the firmware. I didn't have anything to do with the Marlin version other than point out some other bugs in it that have been fixed.

I am not sure how you can have a least squares plane that is at the wrong height. Surely -d/c must lie close to the of the average height of the points and must at least be between the extremes of the points?

Roxy-3DPrintBoard, 2015-03-26:
> I am not sure how you can have a least squares plane that is at the wrong height. Surely -d/c must lie close to the of the average height of the points and must at least be between the extremes of the points?

My guess is you have had calculus?   The summation of squared error terms gives you a 2nd order equation to start with.  But even with 4 points (and not the 25 points I'm working with) you have 3 different ways to put a plane through 3 of the 4 points.  Using that as a starting point you probably can lift one corner of the plane and get a more minimal least squares summation.  If you go lift a different corner of the plane, you probably can find another minimal least squares summation.

Expand that logic to 25 points.  There are going to be a bunch of different solutions.   And you don't even have an equation to differentiate to look for places where the first derivative is zero (local peak or valley).  You have to do this by just looking at individual points.  

If this is of enough interest to you, I have some code and data sets you can pull into Microsoft Visual Studio and take a look at it yourself.   But the Reader's Digest version is this:  By starting with  a and b of the ax+by+cz+d=0 set to zero, I can very consistently get better (a smaller) least squares summation of the data set compared to what the current code can do.   

If the current code made use of the d coefficient that QR_SOLVE returns, it would beat the algorithm I currently have 1/2 the time.  However, with that said, I've found some data sets where d has to change so much it is unusable.   That may be why the original author did not use it because (s)he obviously knew what (s)he was doing!   

I'm going to see if I can set the limit on how much d can change, and if it is too much of a shift, go look for another very small least square summation near by.   

And you know the strange thing???  I don't even need more memory in my PrintrBoard!   I've still got 40 Kb of headroom!  :)

thinkyhead, 2015-03-26:
Rather than using least-squares, one could just take all the squares produced by the grid points and split each one into two nice flat triangles. Get the offset angle of all the triangles and then average the angles together to get the final result.

Or is that pretty close to what is already done?

thinkyhead, 2015-03-26:
@Roxy-3DPrintBoard You coincidentally mentioned a line which is controversial.

    current_position[Z_AXIS] = zprobe_zoffset;

You will notice that this line appears immediately after the line:

    current_position[Z_AXIS] = corrected_position.z;

So there's talk of just getting rid of the line that uses `zprobe_zoffset`. Does that dovetail with your understanding, or is there some valid reason that odd line is included?

nophead, 2015-03-26:
@thinkyhead,
  That might be a reasonable heuristic but least squares fit is an algorithm that minimises the larger errors at the expense of more smaller errors. That seems a reasonable criteria for getting the first layer laid down, particularly if the outer probe points match the bounding box of the object. When the probe offset prevents that the parts of the bed that cannot be reached might have a worse Z offset because had they been probed they would contribute greatly to the sum of squares. I.e. points on the outside are very sensitive to the tilt of the plane.

@Roxy-3DPrintBoard,
  Yes I understand basic calculus. I used to win prizes for maths at school but that was a long time ago now!

There may be more than one solution that gives the least sum of squares but there is only one correct minimum value. If you are able to beat the current code and find a lower sum then it must have bugs.

I still don't understand what you mean by solutions with an unusable value of d. a, b and c are the normal vector of the plane and define its tilt. In the normalised form (where a^2+b^2+c^2=1) d is the distance to the origin. Changing d moves the plane up and down along the normal vector. The bounding values of d must be when the plane is above all the points and below all the points. d must always lie between these two extremes because once outside of that range all the squares are increasing.

Roxy-3DPrintBoard, 2015-03-26:
> current_position[Z_AXIS] = corrected_position.z;
So there's talk of just getting rid of the line that uses zprobe_zoffset. Does that dovetail with your 
understanding, or is there some valid reason that odd line is included?

I think there is value to using a value different than Z_PROBE_OFFSET_FROM_EXTRUDER.  But from the playing around 
I have done I think I'm convinced this needs very careful thought.   More detail below!

> There may be more than one solution that gives the least sum of squares but there is only one correct minimum 
value. If you are able to beat the current code and find a lower sum then it must have bugs. 

Let me try it with different words:   There are many 'local' solutions to the least sum of the errors squared.   
And I agree there usually is one solution that is better than all of the other solutions in that regard.   But 
"No!" I donot agree that because I'm finding iterative solutions that are better than the QR_SOLVE 'solution' that 
means the iterative solution is wrong.

First, I have sample code and data sets that demonstrate exactly that point.  The code loops through the data set 
and computes the Least Square Error summation for the set using the specified coefficients.  And it then does the 
same calculation using the coefficients printed out by Marlin while doing a G29.   I'm pretty confident I'm 
finding better solutions than what QR_SOLVE is finding in many cases.  I would be honored if you would pull them 
into your compiler and take a look at the them.   But if you want to do that, give me a day or two to clean things 
up a little bit!  :)  And you will also need to go change the print outs that Marlin's G29 does to print with 8 
digits of precision.

Secondly, pull QR_SOLVE into your text editor.   The function Marlin calls for the Least Squares Fit is:   

double *qr_solve ( int m, int n, double a[], double b[] )

This function calls dqrls() to perform the least squares fit.   dqrls() is called with these parameters:

 ind = dqrls ( a_qr, lda, m, n, tol, &kr, b, x, r, jpvt, qraux, itask );

Notice the 5th parameter tol.   This is the tolerance it is supposed to use to solve the equation.   So...   This 
may be confirmation it is not looking for the 'Best' solution. Just a very good one.

> I still don't understand what you mean by solutions with an unusable value of d. a, b and c are the normal 
vector of the plane and define its tilt. In the normalised form (where a^2+b^2+c^2=1) d is the distance to the 
origin. Changing d moves the plane up and down along the normal vector. 

I apologize for how long this post is.  But here is the data set that creates the situation where 'd' needs to get 
altered by too much to me useful.  My Z_PROBE_OFFSET_FROM_EXTRUDER is -3.85

struct point data[] = {
{ 20.0000991, 19.0000991, 4.3065242},
{ 62.0000991, 19.0000991, 4.1739020},
{ 104.0000991, 19.0000991, 4.0449175},
{ 146.0001068, 19.0000991, 3.8554094},
{ 188.0001068, 19.0000991, 3.6288599},
{ 20.0000991, 59.0000991, 4.1252846},
{ 62.0000991, 59.0000991, 3.9979536},
{ 104.0000991, 59.0000991, 3.8590474},
{ 146.0001068, 59.0000991, 3.6599485},
{ 188.0001068, 59.0000991, 3.4466283},
{ 20.0000991, 99.0000991, 3.9400760},
{ 62.0000991, 99.0000991, 3.8150606},
{ 104.0000991, 99.0000991, 3.6735084},
{ 146.0001068, 99.0000991, 3.4545657},
{ 188.0001068, 99.0000991, 3.2548053},
{ 20.0000991, 139.0001068, 3.7512297},
{ 62.0000991, 139.0001068, 3.5984330},
{ 104.0000991, 139.0001068, 3.4502663},
{ 146.0001068, 139.0001068, 3.2197480},
{ 188.0001068, 139.0001068, 2.9941906},
{ 20.0000991, 179.0001068, 3.5412168},
{ 62.0000991, 179.0001068, 3.4231464},
{ 104.0000991, 179.0001068, 3.2111489},
{ 146.0001068, 179.0001068, 3.0309016},
{ 188.0001068, 179.0001068, 2.7785553}
};

Let's start with the Marlin verbose output as the points were collected:
[... 25 lines cut ...]

 5.4412059948  a= 0.0015821000 b= 0.0016920000 c= 1.0000000000 d=-3.8500000000    delta=0.0000001000
17.5461092701 G29 using Z_PROBE_OFFSET
 0.7127565664 G29 using QR_SOLVE optimized 'd' offset

In this case I'm still much better than what is currently happening with Marlin ignoring the QR_SOLVE.CPP returned 
'd' coefficent.  But I'm much worse than what could be done if the returned 'd' was actually usuable.

> If you are able to beat the current code and find a lower sum then it must have bugs.

So... I only posted the one data set that is causing problems.  I haven't posted all the other data sets where my 
iterative algorytim clearly beats the QR_SOLVE.CPP solution.  Let me know if you want to look at those!  But right 
now you should have everything you need to rethink this statement!  :)  Please let me know what you decide!

> The bounding values of d must be when the plane is above all the points and below all the points. d must always 
lie between these two extremes because once outside of that range all the squares are increasing.

Yes, in theory I agree this is true.  However...  Look at that data set I posted up above.  QR_SOLVE (or Marlin) 
is claiming a value for the d: coefficient of -4.5517477.   There are no sampled points that big.  So what gives?  
 Using the  d=-4.5517477 number does indeed return a very good least squares error summation when used with the 
returned 'a' and 'b' numbers.  

And in the above data set, I did mangle my bed leveling very bad just to have a data set that was very tilted.   
It took me forever to get it level again because the G29 Topology report was not presenting me accurate 
information.  It had the report rotated 180 degrees because of that pre-processor issue you helped me with.

nophead, 2015-03-26:
I think you misunderstood what I meant by "If you are able to beat the current code and find a lower sum then it must have bugs".

I mean if you can beat QR solve then QR solve has bugs.

nophead, 2015-03-26:
Regarding d, I think it cannot make sense if the distance from the inclined plane to the origin is bigger than all the Z values from the X Y plane.

Is what you are summing (aXi + bYi + cZi - d)^2 ?

nophead, 2015-03-26:
Also c should be close to 1, not actually 1 unless a and b are 0.

Roxy-3DPrintBoard, 2015-03-26:
Marlin calls QR_SOLVE() with c=1 and fixed.   I kind of disagree with this too because it really is going to end up at something like .999 or .998 with a fairly level bed.

And, Yes, I'm summing the 'error' between the plane equation and what is measured.  But the negative numbers caused a bug so I'm re-doing things...

nophead, 2015-03-26:
Yes C will be very close to 1 but then A and B are similarly close to 0 unless your bed is on a massive slant in 3D printer terms. We are talking about small fractions of a mm over something like 200mm.

Roxy-3DPrintBoard, 2015-03-26:
Yes, but from an error perspective..  Especially because the error is being squared...  Which distance is best to use for the error?   The plane normal direction to the point (which is a little bit shorter) or the vertical distance?

nophead, 2015-03-26:
The plane normal direction, which I think is (aXi + bYi + cZi - d)^2 where Xi, Yi, Zi are the probed points.

nophead, 2015-03-26:
Note it is only the distance when a,b and c are normalised, which they aren't if c is fixed at 1. You would need to divide by sqrt(a^2 + b^2 + c^2) if it is not 1.

Roxy-3DPrintBoard, 2015-03-26:
OK...  Understood...  I'm going to redo things to make my stuff better....    But QR_SOLVE is returning a 'd' constant that is outside of the bonds we both agree should be respected.

Roxy-3DPrintBoard, 2015-03-26:
> So there's talk of just getting rid of the line that uses zprobe_zoffset. Does that dovetail with your understanding, or is there some valid reason that odd line is included?

@thinkyhead 

I think there is value in using the coefficients returned by what ever solves the Least Squares Best Fit algorithm.   A much better solution (mathematically) is possible by allowing the offset to be set by the algorithm as part of the solution.  But the algorithm also needs to be sensitive to how much this offset can be adjusted.  The nozzle must be close enough to the surface of the bed to force good adhesion and if that constant is adjusted just for the sake of getting the numbers better for the least square fit it is self defeating.

I'm starting to understand things much better.  This is a really great group of people!  It is helpful to have people that know so much to guide your own thinking!  I expect soon I'll have a solution that takes much less code space and produces at least as good of results as the current code base.  I'll keep you posted!

nophead, 2015-03-26:
Maybe it is simply broken, like the rest of the levelling was when I first started looking at it. 

"The nozzle must be close enough to the surface of the bed to force good adhesion and if that constant is adjusted just for the sake of getting the numbers better for the least square fit it is self defeating."

The least squares value is a measure of how close the plane is to the measured Z values so I can't see how a lower sum can give a worse extrusion result. Something is wrong with the maths if that happens.

AnHardt, 2015-03-26:
I believe zprobe_zoffset must be somewhere.
     current_position[Z_AXIS] = zprobe_zoffset + corrected_position.z;
or something like that?

Roxy-3DPrintBoard, 2015-03-26:
> The least squares value is a measure of how close the plane is to the measured Z values so I can't see how a lower sum can give a worse extrusion result. Something is wrong with the maths if that happens.

Yes.  I agree.  That is good insight.  I need to circle back and figure out why QR_SOLVE is doing that.   
 
> I believe zprobe_zoffset must be somewhere.

It is.  Inherently, the numbers being sent into the least squares fit algorithm are all going to be offset by that much from zero.

nophead, 2015-03-26:
In my implementation my probe function subtracts the z offset of the probe from the measured z, so the plane equation I make is the actual plane of the bed, not one floating above it by the probe offset. Then I add -d / c to the z values during the transformation, whereas Marlin just does a rotation, so never uses d.

I can't get my head around whether that is the same but this
```
current_position[Z_AXIS] = corrected_position.z;
// put the bed at 0 so we don't go below it.
current_position[Z_AXIS] = zprobe_zoffset;
```
definitely looks wrong to me as it throws away the rotated version of the current z. 

Also what is the current position at the time it does this? Why does z happen to equal the probe offset?

thinkyhead, 2015-03-26:
The pending PR #1700 is fixing up all the leveling issues I can find. Give it a scan if you have some time. I will add the `zprobe_zoffset` into the equation. I think the bad line was just supposed to say `+= zprobe_zoffset` instead of `= zprobe_zoffset`.

nophead, 2015-03-26:
The abs on this line is bogus:
https://github.com/MarlinFirmware/Marlin/blob/9e8427f253158c59ea01fbd3b694200ec7e8a557/Marlin/Marlin_main.cpp#L1120.

I pointed this out before and I thought it got fixed. If the Z coefficient
is negative, the points are in the wrong winding order. To fix it the
normal vector should be inverted, not just its Z coefficient.

thinkyhead, 2015-03-26:
@nophead Is the solution _only_ to remove the `abs()` or do we need to mess with the normalize function (or its results)?

thinkyhead, 2015-03-26:
How do you feel about this? Proper?

    current_position[Z_AXIS] = corrected_position.z + zprobe_zoffset;

nophead, 2015-03-26:
Instead if abs it should be something like this:
```
planeNormal = planeNormal.z > 0 ? vector_3( planeNormal.x,  planeNormal.y,  planeNormal.z)
                                                 :  vector_3(-planeNormal.x, -planeNormal.y, -planeNormal.z);

```
Regarding  zprobe_zoffset, I don't know, its confusing the hell out of me.

The probed values are returned without correcting for the offset. But as far as I can see those values are only used to calculate the plane normal. I don't see how the plane offset from the origin is calculated or corrected for. Presumably this line, but I can't see how it is equivalent to what I did with the d value from the plane equation that I derived from the plane normal.

nophead, 2015-03-26:
Or perhaps less code with
```
if (planeNormal.z < 0) {
    planeNormal.x = -planeNormal.x;
    planeNormal.y = -planeNormal.y;
    planeNormal.z = -planeNormal.z;
}
```

nophead, 2015-03-26:
Also there is no need to normalise the two vectors here before crossing them as the result is normalised.
https://github.com/MarlinFirmware/Marlin/blob/9e8427f253158c59ea01fbd3b694200ec7e8a557/Marlin/Marlin_main.cpp#L1117

thinkyhead, 2015-03-26:
vector_3 planeNormal = vector_3::cross(pt1 - pt2, pt3 - pt2).get_normal();

?

nophead, 2015-03-26:
Yes I think that is all it needs followed by the check for negative Z and the three negations.

thinkyhead, 2015-03-26:
I think the point of the z offset is because the nozzle is above the bed, and it wants to correct the coordinate so it's at the level of the bed.

nophead, 2015-03-27:
Yes but what happens to the Z information that the nozzle gained? Say all
the points are 1mm high. How does that end up offsetting the Z axis by 1mm?
It only seems to be used to calculate the plane normal, which is just a
direction to rotate to.

thinkyhead, 2015-03-27:
@nophead Hmm! Well, if it's only used in getting the normal, then it just needs to be consistent. Have you tried it both with and without that being added and found it makes no difference?

nophead, 2015-03-27:
I have never used the code in Marlin, only my own Python version, which definitely works.

Adding zprobe_zoffset must offset all Z values by the probe offset but I only offset them by the distance of the plane from the origin. I don't see how the two methods are equivalent.

thinkyhead, 2015-03-27:
Well, the thing is, not only are these values being used to do the calculations, they also go into `current_position[]` and are passed in `plan_set_position`. So the important thing, I think, is that whatever goes into these variables, it should be the true nozzle position at all times. Otherwise people start their prints and printing starts in the air – or it _could_.

nophead, 2015-03-27:
So why is the current position changed by the probe offset without it having moved?

I am not saying it is definitely wrong, but having written the first version, I can't understand how this one works.

nophead, 2015-03-27:
To correct the bed you need to tilt the model by rotating it and also offset it by the difference between the the probed z values and the probe offset. As far as I can see the probed z values are only used to rotate it. I can't see how they influence the offset, which seems to just be equal to the probe offset.

thinkyhead, 2015-03-27:
@nophead I admit I have not run this code in my head to follow all the effects of these functions. But I feel certain that it's important that the `current_position` be set properly when the process is complete, even if it doesn't matter for the math of leveling. I just want to make sure that we don't see any more printing in the air after leveling because the Z coordinate is incorrect.

nophead, 2015-03-27:
Well the planner normally keeps track of the current position. When the rotation matrix is set the current position has to be corrected slightly to account for the fact it is now being rotated. The two things I can't understand are why the current position suddenly needs to change by the probe offset. That would imply it gets messed up during probing, but why? The probe is a series of normal moves, why does it mess up the planner's current position?

The other thing is how does the difference between the probed position of the bed and the expected position of the bed get compensated for? Can somebody confirm it is? For example can you put a thick sheet of glass on you bed and have it compensated for automatically? If so where is the code that does that?

[18 more replies]

closed 2015-05-31 by boelle