MarlinFirmware/Marlin #1647: Compile error when setting TEMP_SENSOR_BED 1

detroitfire, 2015-03-19:
When I set TEMP_SENSOR_BED to 1 I get a compile error:

http://pastebin.com/PbL2x1e9

Wurstnase, 2015-03-19:
Setup your heatbed-pin

clefranc, 2015-03-19:
Is it complicated to throw a compile error when a pin is not define for the sensors (or other)?

AnHardt, 2015-03-19:
@clefranc 
here it is:
>pins.h:721:33: error: invalid suffix "_RPORT" on integer constant
>     #define HEATER_BED_PIN     -1    // NO BED
>                                 ^

from the above pastebin.

Do you really wan more errors?
But i admit - not really easy to find in between the others.

AnHardt, 2015-03-19:
@clefranc 
No, it isn't so hard to make a test.
We could check for more dependencies.
#if !(TEMP_SENSOR_BED == 0) && !( HEATER_BED_PIN == -1)
  #errror("No pin for bed heater defined.")
#endif

More of a problem is the littering of Configuration.h with this.
We are already discussing possible solutions in #1641.

thinkyhead, 2015-03-26:
Please feel free to add more tests to `SanityCheck.h`.

closed 2015-03-26 by thinkyhead