MarlinFirmware/Marlin #1649: Serial protocol documentation?

natevw, 2015-03-19:
I'm having trouble finding the protocol when talking directly over the raw serial port. I'm able to send my board (from Printrbot LC+) a certain subset of Gcode, like "G0 X1" will move the head, but I don't know how I could e.g. query the current temperatures or get a readout of its current position.

For example with Grbl, [it's documented](https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl) that sending a `?` responds with current status, `$$\n` gets you a settings dump, etc.

natevw, 2015-03-19:
Ah, I think I've partially found my answer. Probably got too stuck on the reprap wiki page… the [Marlin GCodes documentation](https://github.com/MarlinFirmware/Marlin/blob/Development/Documentation/GCodes.md) contains M codes like:

* M27 - Report SD print status
* M114 - Output current position to serial port
* M119 - Output Endstop status to serial port
* M105 - Read current temp (found via https://github.com/kliment/Printrun/blob/f2b96fca5b11f8f663f5e8d03e631fe5cb0b4f16/printrun/pronsole.py#L1401)

So basically, everything's probably there you just have to know what to all grep for ("report", "output", "read", perhaps I've still missed others…)

closed 2015-03-19 by natevw

thinkyhead, 2015-03-20:
@natevw We're working on getting the G-Code documentation for Marlin more complete, grouping the codes by category, and so forth, but there's a lot of editing still ahead of us. So, yeah, at the moment a page search on the RepRap wiki is probably the fastest way to find specific functions.

RobolutionMx, 2018-08-08:
Hi wich format are you using to send GCODE?? char? string??