MarlinFirmware/Marlin #1654: AD595  maxtemp bed error

ta1dr, 2015-03-20:
Hi All

I used Ad595 circuit and termocupler also I connected A3 pin on ramps and modify  pin number on pin_ramps_13....
it is correct work because normaly room temparature 25C  and I see extruder and bed temparature same 25 C  but LCD last row write err:MAXTEMP BED   also printer not work

I test termocupler I used hotair gun and warm bed , increased temperature

how can I solve this error ?
regards
Ahmet

thinkyhead, 2015-03-21:
Potentially related to the MINTEMP error mentioned at #1227. It's possible there's a bug causing the temperature value to get overwritten. Investigating...

thinkyhead, 2015-03-21:
Try #1660 and see if it helps. It might already be merged into the Development branch by the time you read this.

thinkyhead, 2015-03-28:
Awaiting feedback on this and testing with the latest code. We've patched up some buffer overruns and other issues that may have been contributing to this behavior.

ta1dr, 2015-03-28:
yes I tried latest code it is work well  AD595  no any error  ....
thank all

thinkyhead, 2015-03-28:
@ta1dr I'm glad to hear it is working well. We'll just leave that code alone for a while, I think.

closed 2015-03-28 by thinkyhead