MarlinFirmware/Marlin #1659: Dual Z driver errors at compile

Fixx, 2015-03-21:
Using RUMBA board, only configured current development branch pin_RUMBA for Z2 driver and Confadv #define Z_DUAL_STEPPER_DRIVERS

Arduino: 1.6.1 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

stepper.cpp: In function 'void __vector_17()':

stepper.cpp:137:44: error: expected ';' before ',' token

   #define Z_APPLY_STEP(v,Q) Z_STEP_WRITE(v), Z2_STEP_WRITE(v)

                                            ^

stepper.cpp:562:13: note: in expansion of macro 'Z_APPLY_STEP'

             AXIS ##_APPLY_STEP(!INVERT_## AXIS ##_STEP_PIN,0); \

             ^

stepper.cpp:570:9: note: in expansion of macro 'APPLY_MOVEMENT'

         APPLY_MOVEMENT(z, Z);

         ^

stepper.cpp:137:44: error: expected primary-expression before ',' token

   #define Z_APPLY_STEP(v,Q) Z_STEP_WRITE(v), Z2_STEP_WRITE(v)

                                            ^

stepper.cpp:562:13: note: in expansion of macro 'Z_APPLY_STEP'

             AXIS ##_APPLY_STEP(!INVERT_## AXIS ##_STEP_PIN,0); \

             ^

stepper.cpp:570:9: note: in expansion of macro 'APPLY_MOVEMENT'

         APPLY_MOVEMENT(z, Z);

         ^

In file included from Marlin.h:21:0,

                 from stepper.cpp:24:

fastio.h:44:24: error: expected primary-expression before 'do'

 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)

                        ^

fastio.h:67:23: note: in expansion of macro '_WRITE'

 #define WRITE(IO, v)  _WRITE(IO, v)

                       ^

stepper_indirection.h:92:30: note: in expansion of macro 'WRITE'

 #define Z2_STEP_WRITE(STATE) WRITE(Z2_STEP_PIN,STATE)

                              ^

stepper.cpp:137:46: note: in expansion of macro 'Z2_STEP_WRITE'

   #define Z_APPLY_STEP(v,Q) Z_STEP_WRITE(v), Z2_STEP_WRITE(v)

                                              ^

stepper.cpp:562:13: note: in expansion of macro 'Z_APPLY_STEP'

             AXIS ##_APPLY_STEP(!INVERT_## AXIS ##_STEP_PIN,0); \

             ^

stepper.cpp:570:9: note: in expansion of macro 'APPLY_MOVEMENT'

         APPLY_MOVEMENT(z, Z);

         ^

fastio.h:44:24: error: expected ';' before 'do'

 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)

                        ^

fastio.h:67:23: note: in expansion of macro '_WRITE'

 #define WRITE(IO, v)  _WRITE(IO, v)

                       ^

stepper_indirection.h:92:30: note: in expansion of macro 'WRITE'

 #define Z2_STEP_WRITE(STATE) WRITE(Z2_STEP_PIN,STATE)

                              ^

stepper.cpp:137:46: note: in expansion of macro 'Z2_STEP_WRITE'

   #define Z_APPLY_STEP(v,Q) Z_STEP_WRITE(v), Z2_STEP_WRITE(v)

                                              ^

stepper.cpp:562:13: note: in expansion of macro 'Z_APPLY_STEP'

             AXIS ##_APPLY_STEP(!INVERT_## AXIS ##_STEP_PIN,0); \

             ^

stepper.cpp:570:9: note: in expansion of macro 'APPLY_MOVEMENT'

         APPLY_MOVEMENT(z, Z);

         ^

stepper.cpp:1183:1: error: expected '}' at end of input

 }

 ^

stepper.cpp:1183:1: error: expected '}' at end of input

stepper.cpp:1183:1: error: expected '}' at end of input

stepper.cpp:1183:1: error: expected '}' at end of input

Error compiling.

alexborro, 2015-03-21:
I'm working on this issue among a feature I'm implementing (Dual Z Endstop
to automagic align both Z axis during home).
I will release it soon next week.
If you wanna correct by now, replace:

#define Z_APPLY_STEP(v,Q) Z_STEP_WRITE(v), Z2_STEP_WRITE(v)

by

#define Z_APPLY_STEP(v,Q) { Z_STEP_WRITE(v); Z2_STEP_WRITE(v); }

Cheers.

Alex.

Fixx, 2015-03-21:
Thanks Alex, I can wait till week.

closed 2015-03-21 by Fixx

thinkyhead, 2015-03-21:
#1663 fixes this issue.