MarlinFirmware/Marlin #1664: CoreXY kinematics not working

dteeter, 2015-03-22:
I just downloaded the latest development code and found that when trying to move the x axis, only the "Y" stepper turns (i.e. the stepper plugged into the "Y" pins on RAMPS) and when trying to move the y axis, only the "X" stepper turns. With COREXY enabled (i.e. #define COREXY) you would expect a coordinated movement between the  "X" and "Y" steppers in order to make a single axis move.  This test was run on a  RAMPS 1.4 board.

closed 2015-03-22 by dteeter

thinkyhead, 2015-03-22:
Curious, what was your resolution?

dteeter, 2015-03-22:
Hi Scott,

It's likely there is still a problem in the firmware but I decided to close
the issue until I more throughly prove that it's not caused by my
setup. The test was done with a brand new ramps board. Since yesterday, I
reverted back to a firmware that I've been using on an older corexy printer
and I'm seeing movement with both motors now, but there are still some
glitches and I want to get everything working correctly with the known good
firmware before I go back and test with the new stuff.

-Dan

dteeter, 2015-03-23:
Hi Scott,

OK, the development firmware is working fine for me now on CoreXY and I've
got a Reprap Discount Smart Controller that is working as well. The problem
I was having was caused by improperly configured polulu stepper controllers
on a new RAMPS board. For some reason, the position of the pots is
different with these polulu's compared to my other board. I used Nophead's
advice to set the pots at .4V and that did the trick.

Just for fun, at the bottom of this image is the x/y axis of the new CoreXY
printer I'm testing. The old CoreXY printer that's been working since this
past summer is above.
[image: Inline image 1]

If you're ever down in the Bay Area, fell feel to stop by the Bay Area
RepRap meeting (usually at Techshop in San Jose on Wednesday evenings)

-Dan
​

thinkyhead, 2015-03-23:
Thanks for the update @dteeter, glad you worked it out! I hope to get down towards San Jose this summer, time allowing. I'll be sure to stop by the Techshop and see what's happening.