MarlinFirmware/Marlin #1667: Please keep 3-point leveling

woodsmoke, 2015-03-22:
I'm not a coder. I really like 3 point leveling from a mechanical arrangement point of view. 
Your mechanical setup can be *much* simpler:
A Z probe is always diagonally offset from the nozzle (XY plane). 
Most people have an engage-able probe that ends up lower than the nozzle. 
My Z probe is higher than the nozzle and It has no fiddly engage/disengage steps, so it's robust.
3 point leveling allows probing at the 3 bed corners which are compatible with the diagonal offset.
During each Z probe the nozzle is outside the bed extents and can drop below the bed plane.
The grid would require probing the fourth, incompatible point and crashing the nozzle.

I second the 3 point leveling is the quickest, simplest, most straight forward approach and it's all that's needed. Grid mode looks to me like a confusion of the issue. It's your hardware that needs improving if you can't generate coherent readings from a flat bed with 3 probe points.

woodsmoke, 2015-03-22:
How do I make this a feature request?

Roxy-3DPrintBoard, 2015-03-22:
> I second the 3 point leveling is the quickest, simplest, most straight forward approach and it's all that's needed. Grid mode looks to me like a confusion of the issue. It's your hardware that needs improving if you can't generate coherent readings from a flat bed with 3 probe points.

I suspect both are needed just because the user base is so polarized on the subject.   It would probably be best to put them in separate files and only pull in the desired method.   With that said, I object strongly to the "Its all that is needed" and the "Its your hardware that needs improving" mentality.     

Open Source is all about the ability to take the code in what ever direction is needed.   I would rather improve the code than work on mechanical issues with my printer.

boelle, 2015-03-22:
yep, you dont have to worrie.... 3 point is here to stay...

more points dont make it better

boelle, 2015-03-22:
people arguing for more points thinks that marlin can compensate for a bent bed... but it does not..

my argument has always been to get a flat bed rather than messing around

its ok if it tilts etc.... that is what the bed leveling are for.

i have just gotten my mendel90 up and running again... and i spend the better of an hour getting it perfect level manual... benefit is that i don't need automatic bed leveling.

Roxy-3DPrintBoard, 2015-03-22:
> people arguing for more points thinks that marlin can compensate for a bent bed... but it does not..

No.  Not everybody arguing for more points thinks that.  The fact of the matter is there are measurement errors and repeatability errors when probing the bed.   More points spread across the bed area helps compensate for those errors.

And I can tell you first hand that I have supported many many people over at 3DPrintBoard to get Auto Bed Leveling going.   I don't push hard but when people have problems with adhesion on parts of bigger prints, switching away from 3 Point Leveling to the Grid Method  almost always cures the problem with no further effort.

> my argument has always been to get a flat bed rather than messing around

Agreed.   But no bed is perfectly flat.   That is what the Topology Map is all about in the G29 command.  It lets you know where the imperfections are for your bed and what you need to do to level it better before the software takes over.

> its ok if it tilts etc.... that is what the bed leveling are for.

Agreed.  But the more level it starts off before the software has to take over, the better.  If for no other reason than the back lash on the Z-Axis nuts and threaded rods.

> i have just gotten my mendel90 up and running again... and i spend the better of an hour getting it perfect level manual... benefit is that i don't need automatic bed leveling.

Even in this case, the Topology Map of the Auto Bed Leveling is still beneficial to you.  It really does help a lot when manually leveling the bed!   :)

brainscan, 2015-03-22:
As one of the people Roxy helped get this working, in my own experience with my printer I can get consistent prints with grid but not with 3 point. I use 5x5 because I've had better results than 3x3. So in my case why would I want to spend money swapping parts just to switch to 3 point probing?

woodsmoke, 2015-03-22:
By 'all that's needed' I mean you don't need any more than that, rather than we ought to get rid of anything extra, sorry for the confusion. More sample points for a more accurate result seems fine, albeit sampling the same 3 points several times seems most straight forward. It sounds like the minimal 3 point option stays now, so that's great.

nophead, 2015-03-22:
So it seems people have beds that are not flat and then probe them with
probes that are inaccurate and have poor repeatability.  So Marlin gets
ever more complex and bug ridden to cope with rubbish hardware and no
longer fits in the the majority of electronics platforms out there. Madness!

alexborro, 2015-03-22:
I think it is cheaper to source a board with 256Kb instead of a metal bed machined flat to 6 microns, specially if you have a large bed like I do (350mm x 350mm)... Madness **may** be thinking that precision mechanics is cheaper than software.

nophead, 2015-03-22:
Yes but no matter how much software you write you will never print a flat object on a curved bed. And if you probe is inaccurate you won't achieve 6um accuracy either.

ntoff, 2015-03-23:
>i have just gotten my mendel90 up and running again... and i spend the better of an hour getting it perfect level manual... benefit is that i don't need automatic bed leveling.

Same. I don't ever need to mess with my bed on my Mendel90 Sturdy, not unless I change something.

I sit there for a good long while with automotive feeler gauges checking multiple points on the bed, level it, tighten it down and there it'll stay until next I play with it.

First layer heights of 0.06? Easy.

gregrebholz, 2015-03-23:
I have no idea why I'm wading into this, but "you will never print a flat object on a curved bed" is not necessarily true.  There's no reason Marlin and the host couldn't use a topology map to print a minimal "raft" on a warped/distorted/bent/cratered bed to provide a level surface on which to begin layer-0 of the model.  I'm happy for folks with excellent hardware, but "never" is such a strong word and seems out of place in the maker movement.

I put in quite a bit of time leveling my 13x17" aluminum bed but probably would have been happy if there was a simple way to let software deal with my "rubbish hardware".

thinkyhead, 2015-03-23:
Just thinking aloud here, one economical way to compensate for probe "reading errors" is to re-probe at the same point multiple times, then average the readings. In fact, probe down until the switch engages, records that Z, then probe upward until the switch disengages, and record that Z....

lrpirlet, 2015-03-23:
There is another aspect that I have never seen discussed yet... I have a Prusa i3.
I have an issue with the bed to hot-end distance. The distance is independent from the position of the Y coordinate. The distance is proportional to the position of the X coordinate. I have integrated Roxy g29 code in my (old 1.0.0) firmware. Using 5 by 5 grid, it is quite clear that the distance between the hot-end and the bed is following a curve where the extreme points on the X axle are about 0.1 lower than the middle point on the X axle...
NO, the bed is flat. I turned the glass 90 degrees and I got the same pattern... 
No, the smooth rods are NOT bended. I rotated both X rods, one at the time, in 3 steps of about 60 degrees... the pattern does NOT change...
Now, looking at the architecture of the machine, the hot-end, the extruder an the motor are all on the same side of the plane containing both rods... Yes there is a torque that changes the distance of the hot-end to the bed... I proved it using a pendulum (made of a length of wood) pinched in the extruder... In that configuration, the map generated shows a constant distance (well, I mean within the 6 sigma variation) for all 25 probed points...
The solution would be to have a balanced extruder/hot-end over the X axle. The workaround would be to have some software correction that could compensate that variable distance... 
I am NOT a good coder (read I can code given time, trial and error... expect a working code in about one year from now that will work for me alone...). 
Anyone would have a recommendation over the many g29 code showing up?

Roxy-3DPrintBoard, 2015-03-23:
> Just thinking aloud here, one economical way to compensate for probe "reading errors" is to re-probe at the same point multiple times, then average the readings. In fact, probe down until the switch engages, record that Z, then probe upward until the switch disengages, and record that Z....

This is a very interesting idea ThinkyHead!   I suspect you do cancel some of the error.  But I also suspect you introduce new amounts of error because of back lash in both the switch and the Z threaded rods.   And, a very natural place to investigate this and get an answer would be to add an option to the M48 command to do 'Dual Measurements' of each point probed.  My guess would be it helps some hardware configurations and hurts others.  But either way, it is an interesting idea that deserves some investigation!

thinkyhead, 2015-03-23:
@Roxy-3DPrintBoard Well actually I totally expected them to trigger differently in each direction. The question is, does that produce any useful data, or is the disengage position completely unrelated to the bed height … or is it just erratic?

ntoff, 2015-03-24:
>UPDATE: This idea does not work! The problem is the micro-switches are deliberately designed with a lot of hysteresis in the lever movement of the switch. 

Exactly why I rip that stupid lever off my microswitches. You can get  a LOT more accuracy out of them.

nophead, 2015-03-24:
Yes the lever amplifies and inaccuracy but without it you need more force.
I use low force microswitches to avoid bending the bed / axes.  They are
quite expensive and only available in the larger size though. I find them
repeatable to about 0.01mm.

ntoff, 2015-03-24:
Would optical end stops be better? Since they require no force at all.

nophead, 2015-03-24:
I don't think they are as accurate because they are analogue. The tab closes off a 0.1mm slit. As the light level drops the voltage on the collector of the photo transistor rises until it hits a threshold. The exact point it trips will be influenced by ambient IR, supply voltage and temperature.

Capacitive, inductive and magnetic are all the same principal. They are all analogue measurements that can be affected by stray fields and are then converted to a digital signal by electronics that depend on supply voltage and temperature. That is why I prefer a switch, which is truly digital and not affected by fields.

thinkyhead, 2015-03-25:
I think we're going to keep 3 point leveling, so I'll close this. But please continue to discuss.

closed 2015-03-25 by thinkyhead

redsqrly, 2015-03-28:
Just an idea on microswitches, I'm using one i got from a pc mouse. Its much smaller than the original endstop microswitch and requires less force to close even without the lever. I ran repeatability tests with both and the mouse switch was miles better.

ntoff, 2015-03-28:
How about a spring and a paperclip? Zero force. I might actually try that, get a small spring, print a holder that allows me to screw it in and out for the adjustment and a paperclip on the moving axis. Down side is it would require one wire to each, instead of both wires running to the single switch.

alexborro, 2015-03-28:
All my probes are made of an optical switch and a pin.. the pin touch the
bed and triggers the optical switch.. The force applied on bed is just the
pin weight (less than 1g).

thinkyhead, 2015-03-29:
Nice. I assume the switch is enclosed so it's unaffected by ambient light…? That is a nice economical (and 3D-printable) solution which I shall definitely try.

alexborro, 2015-03-29:
You don't need to enclose the switch. It uses a Infrared beam much stronger
than ambient light. The switch is triggered as soon as you interrupt that
beam.

Cheers.

Alex.

nophead, 2015-03-29:
Th exact point it switches will be affected by ambient IR from sunlight or
halogen lighting. This has been a problem with optos on Reprap in the early
days and why it moved to switches.

Also the slit in an Opto is 0.1mm, whereas a good quality microswitch gives
a repeatability of about 1/10th of that.

alexborro, 2015-03-29:
Sorry @nophead but this is not the behavior I see on the optical switch I
have made.

I did not tested it under direct light because it is inside the machine.
But I have achieved the repeatability of 7 microns, which is more than
enough to me.

I could say it is far better than microswitchs, mainly because it applies
no force at all on the bed. A regular microswitch applies about 50g to 80g
of downforce.

Cheers.

Alex

nophead, 2015-03-29:
Is that 7 microns from one minute to the next or 7 microns from day to day at different temperatures and sunlight levels? 

The microswitches I use are only 15g and I am beginning to think the force required to squash the filament is quite significant, so the bed needs to be reasonably stiff anyway.

However you could be right as it very difficult to find any figures for repeatability of opto ,capacitive, inductive and hall sensors. They are all analogue and subject to influence from stray fields but no info about how much. On the other hand switches are not affected by any fields or supply voltage. Possibly temperature affects them due to expansion, but again no data.

alexborro, 2015-03-29:
Although I didn't run a dozen of formal repeatability tests in different
conditions, I have those optos running in my bots for months - actually
more than one year in one of them - and I could tell you I do not need to
adjust the offset at all..  For months. I just need to adjust it when I do
some maintenance on hotend or nozzle.

So yes, they are pretty accurate. I suggest you to give it a try.

Cheers.

Alex.