MarlinFirmware/Marlin #1668: Marlin 1.0.2 - Ramps 1.4 - enabling EEPROM disbales extruder motor

harerama, 2015-03-22:
On my Ramps 1.4 (Chinese clone), when I enable EEPROM (#define EEPROM_SETTINGS), I cannot extrude anymore. I am using an ATMega2560 (clone as well).
I tried M502 then M500 to reset the EEPROM settings.
Disabling EEPROM fixes the problem (it tries both with and without CHIT_CHAT, problem persists).
Without EEPROM support, everything works fine.

I don't think this is the same problem as https://github.com/MarlinFirmware/Marlin/issues/1398 (maybe I am wrong).

boelle, 2015-03-22:
try the latest code instead: https://github.com/MarlinFirmware/Marlin/archive/Development.zip

harerama, 2015-03-23:
I confirm dev version (I took the one from github directly) solves the pb - thanks.

closed 2015-03-23 by harerama