MarlinFirmware/Marlin #1671: Least Squares Best Fit Algorithm (for ABL) Should Be Shrunk

Roxy-3DPrintBoard, 2015-03-22:
People are complaining about  Auto Bed Leveling taking too much code space.  The bulk of the bloat happens when you turn on the Grid Based Auto Bed Leveling because it pulls in a Least Squares Best Fit library.   

I suggest when there is nothing more urgent, we take a look at rewriting that functionality and shrink it down.  I suspect we can free up 10Kb of code space without really trying too hard.  (The current Least Squares Best Fit code solves for the general case.  We don't need to do that!)

nophead, 2015-03-22:
What is the special case optimisation you have in mind?

Roxy-3DPrintBoard, 2015-03-22:
I'm not sure...   But a simple (and small) iterative search solution probably can be done very fast and accurately.   I'm going to toy with that direction soon if somebody does not have a better idea.

Roxy-3DPrintBoard, 2015-03-28:
Just a quick note...  Progress is being made and the code size can be shrunk significantly!

nophead, 2015-03-28:
Good work. Did you resolve why the d value from qr_solve doesn't lie between the probed values?

Roxy-3DPrintBoard, 2015-03-28:
> Good work. Did you resolve why the d value from qr_solve doesn't lie between the probed values?

Not yet.   That really does not make sense to me.   But with that said, I can get the same exact numbers that QR_SOLVE gets, so it may make sense to link it into my Marlin and start using it for my own prints to confirm 'correctness'.    

I'm debating how to proceed.   One option would be to (Gasp!!!!) add a #define to control whether Marlin_Main.CPP pulls in QR_SOLVE or the much shrunk piece of code.  It would not be difficult to make the calling convention identical.  Over time...  Once it is known that things are working correctly that could be pulled out.  But in the mean time, people without a need for the smaller code footprint could keep using the larger and more proven code.

nophead, 2015-03-28:
Well it won't be correct until the plane equations is used to calculate the Z offset and that requires the d issue to be resolved.

Roxy-3DPrintBoard, 2015-03-28:
> Well it won't be correct until the plane equations is used to calculate the Z offset and that requires the d issue to be resolved.

I agree...   And I'm working on that.

nophead, 2015-03-29:
Ah, I see. My origin is in the middle of the bed so the probe points always straddle it. I hadn't realised they might not when the origin is in the corner and probe offset precludes getting to that corner.

Roxy-3DPrintBoard, 2015-03-29:
Well, the good news is we were worried about something that is not an issue!  At least it is understood and not causing confusion any more!    :)

Roxy-3DPrintBoard, 2015-04-02:
A quick question for those that understand the Arduino compiler.    What is the right way to make this code drop in with minimal code changes?   I don't think I can have the exact same name and calling convention because Arduino will compile both modules and try to pull them both in.

Do we have to use different names for the function that does the work so only one module gets pulled in?

alexborro, 2015-04-02:
I think the easiest way is using #ifdefs.

You choose what code to compile.

Cheers

Alex

thinkyhead, 2015-04-04:
@Roxy-3DPrintBoard Definitely __#ifdef__ conditional blocks. Either include the old-style qr_solve code, or include your code. It can be structured just the same way, if the calling conventions are good as they are, or use whatever makes sense, then conditionalize the calling code also.

Roxy-3DPrintBoard, 2015-04-04:
> @Roxy-3DPrintBoard Definitely #ifdef conditional blocks. Either include the old-style qr_solve code, or include your code. It can be structured just the same way, if the calling conventions are good as they are, or use whatever makes sense, then conditionalize the calling code also.

OK!  That is easy enough to do.  Its just that a lot of work has been done to remove those #ifdef's so I was hoping to not clutter things up.   Probably the thing to do is leave Marlin_main.cpp alone and just put an #ifdef #else to turn on the original contents of QR_SOLVE.cpp or the (small) iterative code tacked on the bottom of it.  That would leave the main line code alone.  The one downside is we would be altering the QR_SOLVE.cpp file from its original state.

alexborro, 2015-04-04:
I prefer tweaking the qr_solve.cpp than marlin_main.cpp. 

The qr_solver is suppose to have such algorithms, so no matter one or two versions. You can use a #define in configuration.h to choose which version... it makes the test and debugging easier.

Cheers.

Alex.

thinkyhead, 2015-04-04:
I agree with @alexborro on placement.

I'm almost inclined to see things like the alternative `gcode_G29` function that comes with `MESH_BED_LEVELING` move into the `mesh_bed_leveling.*` files since that would work… but then actually I wouldn't mind splitting out _all the GCode handlers_ into at least one separate 'gcode.cpp` / `gcode.h` unit.

Roxy-3DPrintBoard, 2015-04-05:
> I'm almost inclined to see....

Agreed...  Marlin_main.cpp would still be longer than it should be.

And most certainly agreed on Mesh stuff being a separate file!

closed 2016-05-04 by jbrazio