MarlinFirmware/Marlin #1679: Safe Z Homing at coodinates failure

woodsmoke, 2015-03-24:
Core XY printer, Marlin development downloaded 24/03/15. Arduino 1.05

  #define Z_SAFE_HOMING 
    #define Z_SAFE_HOMING_X_POINT 16    // X point for Z homing when homing all axis (G28)
    #define Z_SAFE_HOMING_Y_POINT 60    // Y point for Z homing when homing all axis (G28)

G28 just triggers a standard X-Y-Z home with-out moving to safe Z homing coordinates

thinkyhead, 2015-03-24:
Are you using `G28` by itself, or `G28 Xn Yn` ?

woodsmoke, 2015-03-24:
Just G28. It doesn't replicate? Here is my config:

//===========================================================================
//============================= Mechanical Settings =========================
//===========================================================================

// Uncomment this option to enable CoreXY kinematics
#define COREXY

// Enable this option for Toshiba steppers
// #define CONFIG_STEPPERS_TOSHIBA

// The pullups are needed if you directly connect a mechanical endstop between the signal and ground pins.
#define ENDSTOPPULLUP_XMAX
#define ENDSTOPPULLUP_YMAX
#define ENDSTOPPULLUP_ZMAX
#define ENDSTOPPULLUP_XMIN
#define ENDSTOPPULLUP_YMIN
#define ENDSTOPPULLUP_ZMIN

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool X_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
//#define DISABLE_MAX_ENDSTOPS
//#define DISABLE_MIN_ENDSTOPS

// For Inverting Stepper Enable Pins (Active Low) use 0, Non Inverting (Active High) use 1
#define X_ENABLE_ON 0
#define Y_ENABLE_ON 0
#define Z_ENABLE_ON 0
#define E_ENABLE_ON 0 // For all extruders

// Disables axis when it's not being used.
#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false
#define DISABLE_E false // For all extruders
#define DISABLE_INACTIVE_EXTRUDER true //disable only inactive extruders and keep active extruder enabled

// If you motor turns to wrong direction, you can invert it here:
#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR true
#define INVERT_E0_DIR true
#define INVERT_E1_DIR true
#define INVERT_E2_DIR true
#define INVERT_E3_DIR true

// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

#define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.

// Travel limits after homing (units are in mm)
#define X_MAX_POS 194
#define X_MIN_POS 0
#define Y_MAX_POS 154
#define Y_MIN_POS 0
#define Z_MAX_POS 144
#define Z_MIN_POS -30

#define X_MAX_LENGTH (X_MAX_POS - X_MIN_POS)
[... 138 lines cut ...]

//// MOVEMENT SETTINGS
#define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
#define HOMING_FEEDRATE {50*60, 50*60, 200, 0}  // set the homing speeds (mm/min)

// default settings

#define DEFAULT_AXIS_STEPS_PER_UNIT   {100,100,4000,174}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {500, 500, 5, 25}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.

#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000   // E acceleration in mm/s^2 for retracts
#define DEFAULT_TRAVEL_ACCELERATION   1000    // X, Y, Z acceleration in mm/s^2 for travel (non printing) moves

// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
// #define EXTRUDER_OFFSET_X {0.0, 20.00} // (in mm) for each extruder, offset of the hotend on the X axis
// #define EXTRUDER_OFFSET_Y {0.0, 5.00}  // (in mm) for each extruder, offset of the hotend on the Y axis

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK                20.0    // (mm/sec)
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                 5.0    // (mm/sec)

Wurstnase, 2015-03-24:
#define Z_PROBE_OFFSET_FROM_EXTRUDER 4 // **-below (always!)**

replace it with

#define Z_PROBE_OFFSET_FROM_EXTRUDER -4 // -below (always!)

woodsmoke, 2015-03-24:
@Wurstnase See issue #1667

thinkyhead, 2015-03-24:
Another item for `SanityCheck.h`. — That offset must always be negative.

woodsmoke, 2015-03-24:
Please don't do that, for the reasons discussed.

thinkyhead, 2015-03-25:
@woodsmoke When would `Z_PROBE_OFFSET_FROM_EXTRUDER` be positive? Can it be positive and result in code that makes any sense?

nophead, 2015-03-25:
@woodsmoke explained earlier that his probe is higher than his nozzle and diagonally offset from it. His nozzle can move outside of the bed area so he can probe three corners without the nozzle hitting the bed. A neat solution if you have a machine with over travel.

woodsmoke, 2015-03-25:
This is it! The bleeding edge of Open Sore Hardware Movement. 
Personal savings are a prerequisite. When I have personal savings I'm going to take up coding.

woodsmoke, 2015-03-25:
Let's steer this back towards the safe homing coordinates issue. Last time this feature failed was because the probe offset caused the safe homing coordinates to fall outside the travel extents. So it just homed at 0. Soft endstops are still present, but some associated code has been removed from config:

#define X_MAX_LENGTH (X_MAX_POS - X_MIN_POS)
#define Y_MAX_LENGTH (Y_MAX_POS - Y_MIN_POS)
#define Z_MAX_LENGTH (Z_MAX_POS - Z_MIN_POS)

Has that got anything to do with it?

Wurstnase, 2015-03-25:
you will find this in Conditionals.h. I think this could be go back to config.

thinkyhead, 2015-03-26:
You are allowed to have a negative `Z_MIN_POS` (and the others as well) which should give the right limits and the correct max_length values. With `Z_PROBE_OFFSET_FROM_EXTRUDER` set to a positive value and the travel limits set far enough, the firmware should at least attempt do all the right stuff, but it's possible there is some code to prevent Z going below 0. Also I see that the `M851` gcode won't allow values outside of -5 to -15, so you can't use that currently.

Wurstnase, 2015-03-26:
Just edit this limits. I've done this and it works.

thinkyhead, 2015-03-28:
We can re-open this if there's still more to do to get everything working smoothly.

closed 2015-03-28 by thinkyhead

woodsmoke, 2015-03-29:
I still can't get a safe home at coordinates to work. I'm not having any luck narrowing it down either.

thinkyhead, 2015-03-30:
Some of the homing code has been tweaked, so give the latest a try. Might not fix it, but at least one known homing bug was addressed.