MarlinFirmware/Marlin #1682: Dev version and bed levelling - change with last version for movements (G1) between G28 and G29

harerama, 2015-03-24:
I just upgraded to the latest dev version. I noticed one change that broke my init G-code.
I had the following: 
  G28
  G1 Z10.0 F{travel_speed} ;move the platform down 10mm
  G29
Now, with the new version, the G1 move is NOT substracted from the final Z height, which results a wrong first layer Z (in this case, my printer started at Z=10+layer height which messed up the print).

The same init sequence was working perfectly previously.
I just need to remove the G1 line to fix the problem, and I am fine with it, but you may want to lay the stress on this functional change.

As far as I remember I copied a starting sequence from some webpage, so there may be many users in the same situation.

Regards

harerama, 2015-03-24:
Actually, it seems to be something in the G29 command and not in the G1 in between - I'll add further details

thinkyhead, 2015-03-24:
@harerama Reset your homing offsets and see if that helps.

harerama, 2015-03-24:
I tried M502, if that is what you mean. My home offsets are all set to 0.
If I send just G28, everything works fine (but no levelling of course), then if I add a G29, my print starts about 1cm above the bed...
If you have a working start G-code sequence that includes auto bed levelling,I can try it (I have a Prusa i3 with amps 1.4).
My bed is almost perfectly flat, so I can avoid the G29 completely, but there must be a way to have it work.

thinkyhead, 2015-03-24:
Which of the bed leveling options are you using?

brainscan, 2015-03-24:
After you tried M502 did you save the settings? Think it's M500. I had a similar problem a long time ago but my start sequence is just G28, G29. Sorry if you have tried this, just seems like the same as my old issue and this was the fix.

harerama, 2015-03-24:
I will try again tonight - I am using grid levelling

Sniffle, 2015-03-24:
I just uploaded the firmware with the latest commits, the issue seems to me that the new raise before retracting probe isn't being taken into account for the Z height and z probe offset.

harerama, 2015-03-24:
Does it mean that you can reproduce the problem ?

Sniffle, 2015-03-24:
every time i go to print, the printer heats, runs through g28, then g29 and then begins printing in the air ~5mm which is what my rais before retracting is set to.

Sniffle, 2015-03-24:
At this point hte gap is reliable, I am working towards my new adjusted offset and it's looking like the gap is 4mm, where that number comes from i don't know.  5 which was my raise after probing was too much and caused a minor crash, i went from that to double my original offset, which was still a crash but less to make the hotend raise.  so now i am down to 4mm + my original offset to make it print.

epatel, 2015-03-25:
Looking at this, if the mesh bed leveling could be at fault here. 
The default search height down to the bed is 4 mm (`MESH_HOME_SEARCH_Z `).

Are you using `MESH_BED_LEVELING`?

Note! It is not possible to compile the mesh + auto bed leveling, and a `G29` by itself does not activate the mesh compensation. To activate mesh compensation one need to run through probing all points, either via the display (if `MANUAL_BED_LEVELING ` was enabled) or a number of `G29 S2`.

alexborro, 2015-03-25:
Is there any document about how to use the Mesh Leveling? I wanna try it
tomorrow..

Sniffle, 2015-03-25:
I finally got the new arbitrary of set setup and now I can reliably print
again.

My offset went from 4.2 to 6.5,  how this happened I have no clue but math
has gone wrong somewhere by 2.3mm

epatel, 2015-03-25:
@alexborro I think I described it shortly in the pull request. Maybe got lost in all the text.

*The bed level option is added last in the Prepare menu. When selecting it the hotend will travel to all points and one use the encoder to move it up/down. Move it down until the usual paper thickness from the bed and press the encoder to "enter" that value, keep doing that until all points have been entered. The mesh can be saved in the EEPROM so storing them is good. A G28 homing will show if the mesh is active or not. If it does not travel to the first measure point (for sync position with mesh) the mesh is not active.*

https://github.com/MarlinFirmware/Marlin/pull/1619

Later I also added `G29`, with which one can do this without a display. 

`G29` or `G29 S0` Print mesh info

`G29 S1` Initiate probing, will do a homing + travel to first probe point. Now use Printrun or what you use to slowly lower the hotend until it touches the bed, i.e. with a paper between to feel when it's close enough.

`G29 S2` Save current height for the current point, then travel to next probe point. Repeat the manual lowering until touches.

I use a paper and do the *goto next point* when the paper feel loose, which gets the compensation a little too loose so I compensate that in Slic3r with "Z offset" (in Printer Settings) set to `-0.05`. 

When probing and `MESH_HOME_SEARCH_Z` set at 4 the Z endstop is considered being at 4 mm and it is possible to search downward (yes, this will press on the Z endstop unless it is an opto) to 0. It might be possible to have the endstop below the bed, but I think G28 might need to change as the initial movement to the first probe point could be bad so an extra lift might be needed.

alexborro, 2015-03-25:
Got it, but how do I use my Z Probe to do an automatic probing?

epatel, 2015-03-25:
@alexborro I don't have that setup so I have not added that, sorry. 

But I think it shouldn't be to hard to add, wither taking code from the auto bed leveling parts or optionally add the mesh there.

alexborro, 2015-03-25:
I will try to implement it tomorrow.

epatel, 2015-03-25:
@alexborro Cool! Please let me know how it works, and feel free to ask I'll help if I can.

thinkyhead, 2015-03-25:
@Sniffle If you've figured out the printing in the air problem, please contribute to this conversation! https://github.com/MarlinFirmware/Marlin/issues/1682#issuecomment-85595876

harerama, 2015-03-25:
Here is the log of the terminal:
  - G28: if I print after that everything is ok (but no bed levelling)
  - then G29
  - then I manually go down Z for the nozzle to almost touch the bed (true manual homing)
=> the Z value is @ -2.7, so I would have printed 2.7 mm above the bed

I don't know what this 2.7 value comes from...
_______________
Send: M501
Recv: echo:V17 stored settings retrieved (366 bytes)
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z3990.00 E758.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X250.00 Y250.00 Z1.75 E25.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X3000 Y3000 Z100 E10000
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P1000.00 R1000.00 T1000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Z0.40 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P15.13 I1.43 D40.17
Recv: echo:Filament settings: Disabled
Recv: echo:Z-Probe Offset (mm):
Recv: echo:   M851 Z-7.30
Recv: ok
[...]
Send: M114
Recv: X:87.00 Y:148.99 Z:7.28 E:951.41 Count X: 87.00 Y:149.00 Z:7.22
Recv: ok
[...]
Send: G28
Recv: ok
[...]
Send: M114
Recv: X:169.00 Y:101.00 Z:7.30 E:951.41 Count X: 169.00 Y:101.00 Z:7.30
Recv: ok
[...]
Send: G29 V4
Recv: G29 Auto Bed Leveling
Recv: Bed X: 20.00 Y: 20.00 Z: 7.28
Recv: Bed X: 130.00 Y: 20.00 Z: 7.20
Recv: Bed X: 130.00 Y: 150.00 Z: 7.06
Recv: Bed X: 20.00 Y: 150.00 Z: 7.14
Recv: Eqn coefficients: a: -0.00073138 b: -0.00106613 d: 7.31540012
Recv: Mean of sampled points: 7.16992521
Recv: 
Recv: Bed Height Topography:
Recv: +-----------+
Recv: |...Back....|
Recv: |Left..Right|
Recv: |...Front...|
Recv: +-----------+
Recv:  -0.03058 -0.10802
Recv:  +0.02757 +0.11103
Recv: 
Recv: planeNormal x: 0.000731 y: 0.001066 z: 1.000000
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: 1.000000 0.000000 -0.000731
Recv: -0.000001 0.999999 -0.001066
Recv: 0.000731 0.001066 0.999999
Recv: ok
Recv: echo:endstops hit:  Z:7.14
[...]
Send: M114
Recv: X:86.99 Y:148.99 Z:7.35 E:951.41 Count X: 87.00 Y:149.00 Z:7.13
Recv: ok
[...]
Send: G1 X100 Y100 F6000
Recv: ok
[...]
Send: G1 Z0
Recv: ok
[...]
Send: G91
Recv: ok
Send: G1 Z-1.0000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-1.0000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-1.0000 F200
Recv: ok
Send: G90
Recv: ok
[...]
Send: G91
Recv: ok
Send: G91
Recv: ok
Send: G1 Z1.0000 F200
Recv: ok
Send: G90
Recv: ok
[...]
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
Send: G91
Recv: ok
Send: G1 Z-0.1000 F200
Recv: ok
Send: G90
Recv: ok
[...]
Send: G91
Recv: ok
Send: G91
Recv: ok
Send: G1 Z0.1000 F200
Recv: ok
Send: G90
Recv: ok
[...]
Send: M114
Recv: X:100.00 Y:100.00 Z:-2.70 E:951.41 Count X: 100.00 Y:100.00 Z:-2.88
Recv: ok

harerama, 2015-03-25:
Here is my full Configuration.h - in case it helps

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.2"
#define STRING_URL "reprap.org"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 115200

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_13_EFB
#endif

// Define this to set a custom name for your generic Mendel,
// #define CUSTOM_MENDEL_NAME "This Mendel"

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
[... 759 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially 

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H

thinkyhead, 2015-03-25:
???

   Recv: echo: M851 Z-7.30

???

thinkyhead, 2015-03-25:
Mulling this over… When `G29` begins, my guess is that it just assumes you did `G28` and that the Z axis hasn't been moved. So maybe doing a `G1` before `G29` isn't truly "supported." But it should be… Your log indicates that the firmware seems to be aware of the Z coordinate. If that's the case, when starting a print, the firmware should be moving to the starting coordinate for the print. It should move Z from its known position to 0 for the first layer.

harerama, 2015-03-25:
@thinkyhead: too early for me to spot easy math...

I removed the G1 between G28 and G29, so this is not the problem.
The sequence for the log above is:
 G28
 G29
 G1 Z0 (the nozzle is 2.7 mm above the bed)
 several G1 to touch the bed (so down 2.7 mm)

SkyFlyer2, 2015-03-25:
I'm also have this problem. This issue occur after adding command "do_blocking_move..." into retract probe function.

That all, what i know.

thinkyhead, 2015-03-25:
So again, very odd. It ends up at `Z_RAISE_AFTER_PROBING` - `Z_PROBE_OFFSET_FROM_EXTRUDER` = ~2.7mm above the bed in the case where you don't move the bed down.

scotty1024, 2015-03-25:
I can add that this may not be a recently added bug. I have a Printrbot Full Metal Simple running their branch of firmware and out of the blue one day in Feb it started printing in air after the G29. No firmware change, it's running the factory code it came with. I switched from CURA to Simplify3D for its slicing but couldn't really see anywhere I got the startup wrong and going back to CURA didn't get me back to working. I'm thinking EEPROM calibration setting? That's all I've "changed" and do so frequently to keep calibration tight.

harerama, 2015-03-25:
I am also using EEPROM.

Everything was working good until I upgraded Marlin to dev version because I had trouble getting EEPROM to work in my previous version.
This may be a hint though, and I will try when I have time to deactivate EEPROM and run G29.

gaussgss, 2015-03-25:
Looks like Z distance difference equals to half distance between extruder and probe.

harerama, 2015-03-25:
Same problem without EEPROM support.
The problem seems to come from the last raise after last probe point - which is not taken into account...

SkyFlyer2, 2015-03-25:
Yes, this is the case. The last climb of the sample (do_blocking_move ...) is not taken into account and marlin thinks that there was no lift.

But how to fix it?

epatel, 2015-03-25:
@harerama Could you perhaps checkout tag `1.0.2` and see if that behaves like this? The we would know if it was introduced before or after that point. If after one could binary search a little among the merges into Development until we find which commit caused this. Before? Yeah then maybe use tag `1.0.1`. Well, this can be a bit tedious I know. git has a command for this but I think maybe manual could work here http://git-scm.com/docs/git-bisect

harerama, 2015-03-25:
Found the bug in Marlin_main.cpp, in function set_bed_level_equation_lsq (@line 1103)
Delete the following lines:
    // put the bed at 0 so we don't go below it.
    current_position[Z_AXIS] = zprobe_zoffset; // in the lsq we reach here after raising the extruder due to the loop structure

I don't understand the purpose of this line, nor the comment.

Again, this WAS working with 1.0.2 at some point.

epatel, 2015-03-25:
The commit https://github.com/MarlinFirmware/Marlin/commit/05932e445883c4fdf5620f90d503a3745639ffbc

thinkyhead, 2015-03-26:
@harerama Good catch. I wasn't looking anywhere near that function. Clearly that line doesn't belong, but maybe it was intended to follow the `plan_set_position` line…? I will remove it if no one else has made a PR yet.

@epatel I see that you're correct. The `zprobe_zoffset` value is never initialized unless EEPROM is in use. I'm confused and annoyed by the way that `gcode_SET_Z_PROBE_OFFSET` and the Z offset interact, however. The custom M code requires a value between -15 and -5 but then it sets the offset to the negative if the input, so it will be between +5 and +15. So yes,  `float zprobe_zoffset = -Z_PROBE_OFFSET_FROM_EXTRUDER` is correct, but I just find it weird that it insists on being the opposite sign to everything else. So I'm going to see about making that more consistent as well.

thinkyhead, 2015-03-26:
I could use some feedback on the lines that read:

    #elif defined(Z_PROBE_ALLEN_KEY)
      engage_z_probe();

Should this be replaced by:

    #elif defined(Z_PROBE_ALLEN_KEY) || defined(SERVO_LEVELING)

…? It seems like this is correct, because the `engage/retract z_probe` functions customize their behavior based on both of these flags.

thinkyhead, 2015-03-26:
@epatel I see that same line also in the `set_bed_level_equation_3pts` function. I will ask @alexborro whether it was meant to be intentional or not.

Ziggy2013, 2015-03-26:
If there are changes to the way the zprobe_zoffset value is to be handled, this value can be negative OR positive depending on how the Z Probe is physically implemented on the printer.  

Also the LCD displayed "Z Offset" parameter is currently only displayed as a positive float value. It should be possible to set this value negative as well as positive to allow for different Z Probe implementations.

Ziggy2013, 2015-03-26:
@epatel I see that you're correct. The zprobe_zoffset value is never initialized unless EEPROM is in use. 

I believe this assessment is not correct. In configurationstore.h there is this code.  

// configurationstore. h code

#ifdef EEPROM_SETTINGS
  void Config_StoreSettings();
  void Config_RetrieveSettings();
#else
  FORCE_INLINE void Config_StoreSettings() {}
  FORCE_INLINE void Config_RetrieveSettings() { Config_ResetDefault(); Config_PrintSettings(); }
#endif

If EEPROM_SETTINGS is not defined the inline code becomes ResetDefault()  which initialises the zprobe_zoffset = -Z_PROBE_OFFSET_FROM_EXTRUDER;

thinkyhead, 2015-03-26:
@Ziggy2013 I agree we should allow a wider range. How about -20 to +20?

Ziggy2013, 2015-03-26:
@thinkyhead +/- 20mm should be enough as a default range.

This would also mean the LCD "Z Offset" menu item should allow the Z Offset (which is actually zprobe_zoffset) to be set between -20 to +20.  In ultralcd.cpp the Z Offset menu item statement is

#ifdef ENABLE_AUTO_BED_LEVELING
    MENU_ITEM_EDIT(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, 0.0, 50);
  #endif

Obviously the 0.0, 50 lower/upper limits in this statement need to be changed to the defined defaults for the Z probe offset. Unfortunately it's not a simple matter to make the lower limit negative. The handling of MENU_ITEM_EDIT also needs to be modified to allow negative values.

thinkyhead, 2015-03-26:
@Ziggy2013 I've applied those changes to #1700. And actually, I'm pretty sure that MENU_ITEM_EDIT has supported negative values for a while, but I may be mistaken. I know code was added that figures out the range, limits between 0 and that size, but then adds the low value, so it should work, but I will double-check.

Ziggy2013, 2015-03-26:
@thinkyhead  Even just allowing negative integers as the lower limit is a great improvement - thanks

thinkyhead, 2015-03-27:
@harerama Have you checked out the latest iteration of #1700? There's some minor confusion (mainly just my own) about what to do with the `zprobe_zoffset`, and whether it should be getting added to the previous `current_position[Z_AXIS]` at those lines that were removed or just left out entirely. In other words, should it be this…

    current_position[Z_AXIS] = corrected_position.z + zprobe_zoffset;

or just this…?

    current_position[Z_AXIS] = corrected_position.z;

Ziggy2013, 2015-03-27:
Just in case this particular wrinkle has been overlooked? There is also the Z Home Offset value as set by M206 which has to be handled through all these calculations.

harerama, 2015-03-27:
@tinkyhead: I am currently using (without any problem) the second one:
    current_position[Z_AXIS] = corrected_position.z;
Looking at the value I have for zprobe_offset (7.3 mm), it certainly shouldn't be added to the current position, otherwise I would have either printed in the air, or crashed my bed...

thinkyhead, 2015-03-27:
@harerama I've finally merged #1700 and used the code you suggest. Give it a try when you have a chance, and watch out for temperature behavior. I tried to address that, but the cause was obscure – probably a buffer overrun from declaring too-short arrays.

Sniffle, 2015-03-27:
I'll also test it when I get home

Sniffle, 2015-03-28:
So, I know from previous versions of marlin my Z-probe offset was -4.3mm, 

I flashed just a few min ago after #1700 had been merged, I left my offset of -4.3 for my Z offset, I ended up changing it using M851 to -8.05 to allow a .038 feeler gauge to have a moderate drag when placed under the hotend nozzel.  

my procedure for checking the offset to make sure it is correct is to G28, G29 E, G1 X150 Y150 Z0 F3000

At the end of which i check with the feeler gauge to see if the hotends were placed properly.

thinkyhead, 2015-03-28:
I'll be monitoring the issue queue all weekend, in case there's anything needing to be fixed.

alexborro, 2015-03-28:
Ok, we have 2 offsets in your case:
4.3 x 8.05

Which one is closer to your real offset (distance between *triggered* probe
and the tip) ?

Cheers

Alex.

Sniffle, 2015-03-28:
-4.3 is the measured offset

-8.05 is the printable offset meaning if I want it to be properly spaced to
have a successful print I use this.

alexborro, 2015-03-28:
So we can tell that previous version of Marlin has more accurate probe
offset.

Sniffle, 2015-03-28:
Yes, this all started with the code to raise the hotend before retracting
the probe at the end of a G29.  Before that it relied on the measured
offset.

thinkyhead, 2015-03-28:
@Sniffle So, maybe I just missed it, but what are your `RAISE_*` values?

Sniffle, 2015-03-28:
Off the top of my head I believe
6
5
5

I'll edit this post tomorrow to confirm or correct but I am pretty sure
that's it.

oysteinkrog, 2015-03-28:
Hmm, I had this issue (Z0 causes nozzle to hover several mm above bed) yesterday using head.
Testing now with the same config I have the opposite problem, where Z0 causes head crash.
Do I need to update the Z offset?

thinkyhead, 2015-03-28:
@oysteinkrog There is some confusion concerning where the firmware "thinks" it is positioned after a `G29`, and how that is accomplished. The best bet is to look at the bed leveling code at the last point when it worked reliably, which is perhaps 2 weeks ago, and see what the heck it was doing. There was a point when the bed leveling equation functions ended with the line `current_position[Z_AXIS] = zprobe_zoffset` – and this was apparently incorrect. But it might have been working for some.

In any case, I would rather have a look at the older code and truly know what it was doing, and that it was doing it correctly, and be able to say exactly why it's screwed up now, and have it solidly fixed, so… I'll be back in a bit with hopefully some answer.

thinkyhead, 2015-03-29:
Specifically, the `set_bed_level_equation` functions from the 14th both end with these lines:

    current_position[Z_AXIS] = corrected_position.z;

    // put the bed at 0 so we don't go below it.
    current_position[Z_AXIS] = zprobe_zoffset; // in the lsq we reach here after raising the extruder due to the loop structure

    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);

Code from 2 weeks ago also didn't have the Delta bed leveling, but that would only pertain to DELTA configurations.

thinkyhead, 2015-03-29:
The same line `current_position[Z_AXIS] = corrected_position.z;` was also in the code on March 1. It is possible that it's the right thing to do, even though it makes the previous line superfluous.

[52 more replies]

closed 2015-08-03 by boelle