MarlinFirmware/Marlin #1683: PID tuning in Configutation.h is useless.

AnHardt, 2015-03-24:
'Configuration.h' line 187 (and all the other configs):
       #define PID_dT ((OVERSAMPLENR * 10.0)/(F_CPU / 64.0 / 256.0)) //sampling period of the temperature routine

a.) OVERSAMPLENR is not defined here. OVERSAMPLENR is defined in 'thermistortables.h' - what is linclude in line 730.
b.) https://github.com/thinkyhead/Marlin/blob/Development/Marlin/Conditionals.h#L393 is the fix.

I suggest to either #include "thermistortables.h" in time and bemove b.) or define  PID_dT in 'temperature.h'.

Wurstnase, 2015-03-24:
Just a question: How does PID_dT works? 

When I'm right the timer0 runs with 62.5kHz and for the temp-isr on a 128 counter. So we are at ~2ms. That means it takes 16*12*2ms ~ 38.4ms for each cylce. PID_dT otherwise is set to ~200ms.

Maybe @ErikZalm has an idea?

nophead, 2015-03-24:
OVERSAMPLENR doesn't need to be defined when you define PID_dT.  If it is defined where PID_dT is used it will work.

thinkyhead, 2015-03-24:
Actually, you will notice, if you search the code for `PID_dT`, that the value in `Configuration.h` is totally ignored anyway.

thinkyhead, 2015-03-25:
/**
     * Sampling period of the temperature routine
     * This override comes originally from temperature.cpp
     * The Configuration.h option is basically ignored.
     */
    #ifdef PID_dT
      #undef PID_dT
    #endif
    #define PID_dT ((OVERSAMPLENR * 12.0)/(F_CPU / 64.0 / 256.0))

nophead, 2015-03-25:
Probably should be
```
#undef PID_dT
#define PID_dT ((OVERSAMPLENR * 12.0)/(F_CPU / 64.0 / 256.0))
```
It isn't something that the user should be able to change. It needs to match the code in temperature.cpp. The undef is only to allow old configs to be ignored.

AnHardt, 2015-03-25:
And it should go back to temperature.cpp.

closed 2015-03-26 by thinkyhead