MarlinFirmware/Marlin #1685: Option to disable the buzzer.

cntrprt, 2015-03-24:
Could you add an option to disable the buzzer when used as feedback for the control inputs.?  
I don't mind it for alarms and such, but I hate it when used as feedback for the control wheel.
Thanks.

thinkyhead, 2015-03-25:
@cntrprt What display are you using? There are two options in `Configuration.h` that may be of use to you:

    //#define LCD_FEEDBACK_FREQUENCY_HZ 1000	// this is the tone frequency the buzzer plays when on UI feedback. ie Screen Click
    //#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 100 // the duration the buzzer plays the UI feedback sound. ie Screen Click

If you're using a non-graphical LCD you can set LCD_FEEDBACK_FREQUENCY_DURATION_MS to 0 and it will silence the feedback. For graphical displays currently this option doesn't have any effect. So if you have one of those you can just comment out the call to `lcd_implementation_quick_feedback` in `ultralcd.cpp` as a workaround until we have that updated.

ntoff, 2015-03-25:
So that's why changing the frequency on my full graphics display doesn't do anything.

Time to go hunting through the firmware methinks (I'm using a horribly old but reliable version of Marlin)

cntrprt, 2015-03-25:
I'm using the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Thanks

thinkyhead, 2015-03-26:
#1693 will make it possible once it is merged.

closed 2015-03-26 by thinkyhead

thinkyhead, 2015-03-31:
#1752 makes the click sound much less obtrusive, more like it used to be