MarlinFirmware/Marlin #1690: Bed leveling is not exact

SkyFlyer2, 2015-03-25:
May be this is not a bug, just my printer issue, I don't know. But this problem occurs in all versions Marlin, that I tested (1.0.0 RC2 - 1.0.2). 

Now I'm use the optical Z-probe, before was the micro-switch. Result the same, you can see problem on video.

Any ideas?

http://youtu.be/feP8gsw45oU

nophead, 2015-03-25:
Are your Y bars exactly parallel. If not the bed will rotate around the Y
axis as it moves in Y, which is not compensated for.

SkyFlyer2, 2015-03-25:
On video you can see, that heated bed is not moving, only x-carriage.

What you mean, says Y is not compensated? I see compensation when move along X-axis, and along the Y-axis also.

Everything is moving while performing auto leveling, so as i understand, all discrepancies must be measured and compensated

nophead, 2015-03-25:
The compensation assumes the bed is a flat plane but inclined. It rotates
the model to match the incline of the bed.

If your bed rotates around Y as it moves along Y it appears to be twisted
when it is probed, even if it is perfectly flat. That means if you get any
three corners level, the fourth will not be level. That type of error
cannot be compensated for with a fixed rotation matrix. It would need to
apply a different rotation at each end of the Y axis and interpolate all
the way along. I.e. not a simple matrix multiplication by a constant matrix.

The recently added mesh compensation would fix the nozzle being at the
right height but I think it would print a distorted object. You really need
to make your Y bars exactly level at each end and of course perfectly
perpendicular to your other axes as that sort of error is also not
compensated for with ABL. It only perfectly compensates for a perfectly
flat bed inclined relative to the axes but with motion exactly parallel /
orthogonal with the axes.

SkyFlyer2, 2015-03-25:
Sorry, but I don't understand what means "bed rotates around Y".

Currently, without bed auto leveling procedure z-offset is better, than with auto leveling.

The design of this printer is not perfect, I will do a new, dibond. This printer helps  print plastic parts for the new.

nophead, 2015-03-25:
If the four ends of your Y rods are not all at exactly the same height then one side of the Y carriage will rise relative to the other as it moves along Y. I.e. the bed tilts more or less as it moves. That change in tilt is a small rotation around the Y axis of the machine.

To fix it you need to put shims under one of the Y bars so they are exactly parallel. Then you can get all four corners of the bed to be the same distance to the nozzle instead of only three.

Roxy-3DPrintBoard, 2015-03-25:
>  That change in tilt is a small rotation around the Y axis of the machine.

I guess it depends on what the machine's problem is.  But it would seem to me that it would be a rotation around the X axis of the machine.

nophead, 2015-03-25:
We are talking about Mendel style axis arrangement.

For the Y carriage to rotate around X while travelling along the Y bars I think the bars would need to be curved. If it tilts left right due to them not being vertically parallel that is a rotation around Y.

SkyFlyer2, 2015-03-25:
Y axis must be parallel relative to what? Relative to each other or...?
It is still difficult to understand.

nophead, 2015-03-25:
To stop the bed rotating (making it appear to be twisted when probed) the Y bar ends must all be at the same height, to a similar accuracy as you want your bed to be flat to. I.e. the bars must be exactly parallel to each other in both the XY and YZ planes.

People seem to completely neglect how accurately the Y bars need to be aligned. It was very difficult when they were slung from a studding frame, which is why I stood them on a flat sheet with Mendel90. Even then if you stand the machine on a twisted surface it is sometimes necessary to put a paper shim under one corner as the base is not infinitely stiff.

SkyFlyer2, 2015-03-25:
But why without bed autoleveling all movements is much better? I disabled this feature, then I carefully do the manual bed leveling. 
Then i do test, shown on the video, again. Result is better, than you have seen with auto leveling enabled.

Of course, i will try to made all axes aligned, but this is not so easy. In addition, rods are crooked a little,  they were damaged during shipping.

How do you think, 8mm enough for Mendel90, or necessary to increase its diametr?

nophead, 2015-03-25:
I don't know why it is worse but @Roxy-3DPrintBoard just found a bug in the grid version.

Yes I think 8mm is enough as that is what we ship in Mendel90 kits. They do need to be precision hardened ground rods with h6 tolerance to work with the bearings.

thinkyhead, 2015-03-26:
Manual bed leveling with always be more accurate than bed leveling using an electronic probe switch.

SkyFlyer2, 2015-03-26:
Yes, these shafts i have.

The main reason why I enabled bed auto leveling, was a pit in the middle of a heated bed. This is due the Z-carriage is very weak and under the weight of the extruder X-carriage is slack.

nophead, 2015-03-26:
With a pit in the middle I can imagine the least squares fit may improve
the middle at the expense of making the corners worse. Particularly because
the probe is offset from the nozzle, so the probe points might be further
from some corners than others. Those corners don't get a vote in the LSF,
so will probably end up worse off.

thinkyhead, 2015-03-26:
Such is a good case for mesh bed leveling, but ultimately you'll want to fix the X axis so the rods don't sag. The current mesh leveling routine is manual-only, but it could be automated at the cost of accuracy.

boelle, 2015-04-01:
i think this is an hardware issue and we can close it as its not a thing with marlin firmware

CptanPanic, 2015-04-27:
I was thinking that a good addition to the topological report, would be a printout of the topology after the Bed Level Correction Matrix is applied.  That way you can see the actual magnitude of the defects that can't be compensated for.  I was seeing if I could calculate this myself in matlab, but I am unable to figure out the math.

Roxy-3DPrintBoard, 2015-04-27:
- I was thinking that a good addition to the topological report, would be a printout of the topology after the Bed Level Correction Matrix is applied. That way you can see the actual magnitude of the defects that can't be compensated for. I was seeing if I could calculate this myself in matlab, but I am unable to figure out the math.

That is an interesting idea.   But I suspect you would have to re-probe the bed with even more resolution using the correction matrix to get an interesting map.   The math is easy, especially in something like MatLab.    You just load a 3x3 matrix with the bed correction matrix.   Probably the way to do it is have a for() loop for x and y.   You just multiply a 1x3 matrix of your [x,y,z] times the 3x3 correction matrix.  MatLab will give you a new [1x3] matrix with the new coordinates.

boelle, 2015-05-31:
will close as this again seems like bad assembly /adjustment of printer

closed 2015-05-31 by boelle

nicksears, 2016-02-08:
I don't know if this needs to be reopened, but I have a similar issue - I need more precision with the z probing. I have theoretically perfect probe resolution where I use my nozzle as the probe, but since I'm probing a very small area (I'm printing onto a 38mm SEM stub) and the z probe values are only in tens of microns, I think it's throwing it off.Is there any way to make the z-probing more precise?

Blue-Marlin, 2016-02-08:
@nicksears 
The theoretical resolution is one (micro) step in z-direction. For a i3 with M5 spindle 1mm/4000 = 0.0025mm, for typical Delta 1mm/80 = 0.0125mm.
What you see on the console output is just a rounded value. Internally we calculate with the full precision of a __float__ value.

nicksears, 2016-02-08:
Interesting, I'm also working with a kudo3d and noticed that they said they added a digit to the z position in their update to the firmware (http://www.kudo3d.com/downloads/ down at the bottom of the page)

I'm using 3 point leveling because I've heard there was a bug in the mesh leveling calculations. When I did mesh it only showed (0.2, 0.1, -0.3, -0.2) or something. Now it outputs:

10:41:55.053 : Bed Level Correction Matrix:
10:41:55.057 : +0.999984 +0.000000 +0.005676
10:41:55.057 : +0.000024 +0.999991 -0.004245
10:41:55.061 : -0.005676 +0.004245 +0.999975

I can't do probe repeatability/bumping, so I'm going super slow (6mm/MIN) and lifting 1mm between so there's plenty of time for the bed to settle vibration from moving. I'm fairly sure the probing is near perfect. if not is there any way to know with the 3 point? If I had to I'd turn on the bump/repeat probing, and multiple times if possible...

Bio-Printer, 2016-02-08:
A couple of comments.   First, you maybe getting confused by thinking grid based and mesh auto bed leveling are the same thing.   They are not.    Mesh Bed Leveling will not produce a Bed Level Correction Matrix.

Given you need very accurate Z-Probing, I suggest you do several M48's at different locations with and without multiple legs of travel.    The reason to do this is it will help you understand the limitations of your Z-Probe and how much you can trust it to give you the bed locations accurately.   What ever error is reported by M48, expect even more uncertainty when using the Z-Probe to do Auto Bed Leveling just because you will be probing multiple points and it is possible for the error of each measurement to compound.

Blue-Marlin, 2016-02-08:
Indeed the change in precision only changes the output to the serial console. Compare line 167 and 177 of MarlinSerial.h and MarlinSerial.h.bak. That's all. As said this has __nothing__ to do with the precision of the internal calculations.

To get more samples of your tiny area you could try AUTO_BED_LEVELING_GRID. 
Narrow down the area to probe and set a high AUTO_BED_LEVELING_GRID_POINTS. 
Maybe you have to decrease MIN_PROBE_EDGE.

Then you still will not probe 3 points multiple times, but a lot of points near each other - averaged by a least squares calculation.

nicksears, 2016-02-08:
Awesome! thanks for all your help!

For those interested I was able to get this to work by turning on "Z_MIN_PROBE_REPEATABILITY_TEST" in configuration.h. I used an xyz offset from extruder of 0 since the probe is my nozzle. I raise 1 before homing and 0.5 before/during/after probing. I set z homing speed to 0.5*60.

In configuration_adv.h i set home bump to 0.5 and the divisor of 5 gives a bump speed of 0.1*60=6mm/min (really slow) I think this is all I had to do. 

This gave me a standard deviation of 0.002110, which is 2um I assume? and the smallest I've seen...using a 10x bump divisor lowered the standard deviation to 0.001713!
I have a ball screw with a pitch of 5mm and the usteps/mm=3840 and have a theoretical resolution of 0.25um/ustep

thinkyhead, 2016-02-11:
@Blue-Marlin I see you have a marlin-firmware.org redirecting to the Marlin project. Is that for official use or just something of your own?

Roxy-3DPrintBoard, 2016-02-11:
@nicksears 

> This gave me a standard deviation of 0.002110, which is 2um I assume? and the smallest I've seen...using a 10x bump divisor lowered the standard deviation to 0.001713!

Hi Nick!   Yes.  Your Z-Probe setup is giving you a 2 um standard deviation.    What every you are printing, you are good to go!

Blue-Marlin, 2016-02-11:
@thinkyhead 
Welcome back!
marlin-firmware.org is owned by @AnHardt. He reserved if when marlinfirmware.org went down. He is extremely unhappy with how Marlin evolved over the last few month. Our internal Marlin is diverging fast and he's thinking about publishing it. Maybe he'll use Marlin-Firmware.org, maybe he'll use a completely different name, mayme it'll stay internaly. To ask him today seems not to be a good idea if you want a yes. He slammed some doors and went off with the dog for hours - after reading the Marlin postbox.

thinkyhead, 2016-02-19:
> Welcome back!

Great to be here. Focusing like a laser on tasks is helping.

> @AnHardt is extremely unhappy with how Marlin evolved over the last few month
> He slammed some doors and went off with the dog for hours - after reading the Marlin postbox.

Meanwhile, the @MagoKimbra version of Marlin has been evolving at quite a pace, including now support for Due in a separate fork. Recent changes to MarlinDev make it a bit harder to take advantage of the work being done in forks. I've proposed to @MagoKimbra that I can step in to integrate his many changes, but I realize this will be a very meticulous side-project. In the long run I hope that we will be able to restore the ecosystem so that Marlin forks can have closer parity to the main branch.

nicksears, 2016-02-19:
A final note, I thought I had down this to work, but I hadn't. My probing was very precise and was working great to find z0, but my leveling must have just been close enough that I didn't notice right away that out want actually adjusting z throughout the layer. 

I turns out, right after running g29, I do a "g28 y" which evidently completely throws out the probe leveling! I have to "home" y (max) every so often during a print to wipe the nozzle, so I changed this to "g1 y150" and voila!

brainscan, 2016-02-19:
It's always been the case that g28 shouldn't be used after g29.

Bio-Printer, 2016-02-19:
> I turns out, right after running g29, I do a "g28 y" which evidently completely throws out the probe leveling!

As BrainScan pointed out, it has always been that way.   Probably this needs to change.  But right now, the G28 clears (or really...  Sets to identity) the Bed Level Correction matrix.  So once you enter the G28 code for any reason, you lose the information to do the Grid Based bed compensation.

nicksears, 2016-02-19:
Ya, it looks like that idea is being debated (https://github.com/MarlinFirmware/MarlinDev/issues/28) and I agree, g28 shouldnt reset the matrix, but I suppose you should be able to do "G1 X0 Y0 Z0" etc...

Bio-Printer, 2016-02-19:
Yeah...   But it is important to keep perspective.   When the Auto Bed Leveling first came out, the author was trying to preserve old behavior.  G28 was supposed to home and reset the machine.   If an auto bed level correction matrix was active, it might affect later movements of the machine, or for that matter, maybe even the home operations.

But things have evolved and it is too limiting to clear the Auto Bed Leveling information ever time you home the machine now.