MarlinFirmware/Marlin #1692: 'DIOE2_ENABLE_PIN_RPORT' not declared compile error

woodsmoke, 2015-03-25:
Marlin downloaded 25/03/15 11am GMT. Arduino 1.05, CoreXY, #1679 for config

'DIOE2_ENABLE_PIN_RPORT' not declared in this scope on line 2549 in Marlin_main

danielmartinez, 2015-03-25:
If you comment this line #define Z_DUAL_STEPPER_DRIVERS at Marlin/Configuration_adv.h:101 I think that you will be able to compile.

Commit https://github.com/MarlinFirmware/Marlin/commit/0ce35766855c53c95df9d7e065f774534937df96 makes Z_DUAL_STEPPER_DRIVERS defined by default, but before of that commit it was undefined by default, so if you doesn't have dual drivers for Z axis then you must comment this line.

woodsmoke, 2015-03-25:
Confirmed, do I close this issue now or once the new commit has been made?

Wurstnase, 2015-03-25:
I made a pull request #1694 
Saw this allready yesterday, but forgot it again.

thinkyhead, 2015-03-26:
Merged, so it should compile now.

closed 2015-03-28 by thinkyhead