MarlinFirmware/Marlin #1695: Heated bed won't go over about 48 celsius on recent commits

dattas, 2015-03-25:
I've been testing out the current development branch on a few of my 3d printers on and off. I want to say about a week ago the heated bed was fine, but if I pull the latest commits my heated bed won't go above 48 celsius. I've also tried turning on bed pid tuning on recent commits as well with no success, I have the same problem.
If I go back to older versions of the development branch this issue doesn't exist so the hardware appears to be fine. I'll try to work this weekend to find the commit that this first started in, but I wanted to get the issue created so I wouldn't forget to do that!

My current testing setup is a Rambo with a MK2b heated bed.

AnHardt, 2015-03-25:
#1634 ?

SkyFlyer2, 2015-03-26:
Check heated bed voltage.
On my bed voltage now is only 10.8v (dav commit), before it was 12.8v. Temperature cannot reach more than 115 degrees, also very long heating.

dattas, 2015-03-26:
This appears to be 100% in Marlin. If I go to an older version of Marlin (from say a week ago) this issue doesn't exist. Even when using bed PID autotune on the newer versions I can't reach what older firmwares do (over 100 fairly easily). It isn't a problem with the thermistor (verified that it is reporting the correct temperature with 2 other forms of temperature measurement) and it isn't an pid problem as it exists without the bed pid being enabled. I will once again try to pin down the exact commit that this was introduced in but it may be sometime this weekend.

Wurstnase, 2015-03-26:
Which Arduino Version do you use for compiling?

dattas, 2015-03-26:
I'm using Arduino 1.6.1. I've just done some testing, lots of flashing and found out that commit 3703694e3017356acae81d90788945666694823e is where this bug was introduced. 3d0a060a7a998b91d815a822bdf7774e3a8b9f6a does not have this issue, the bed heats normally, but when I flash with 3703694e3017356acae81d90788945666694823e I have this problem.

SkyFlyer2, 2015-03-26:
I also think that this bug somewhere in this commit. I have the feeling that the bed was supplied pulsed voltage, respectively, much longer warm. LED blinks on the bed, this should not be.

thinkyhead, 2015-03-26:
Looking very closely at the code…

thinkyhead, 2015-03-26:
Well, you can see there's _almost_ nothing that changes the logic in that commit. And it fixes a bug in the min/max test. There's just one thing I can see being problematic.

    static unsigned long raw_temp_value[EXTRUDERS] = { 0 };

This should probably be:

    static unsigned long raw_temp_value[4] = { 0 };

Because some of the logic that deals with these values is only conditional on:

    #if HAS_TEMP_n ...

SkyFlyer2, 2015-03-26:
This is part of my "confiiguration.h".

----
// This defines the number of extruders
#define EXTRUDERS 1
........
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1
.........

Is it?

thinkyhead, 2015-03-26:
Looks ok to me! Will keep tracking this issue... Something about the heater logic is very touchy.

Wurstnase, 2015-03-27:
I have similar issues. Yesterday I used first time PID for my bed. My bed makes always some clicking noises when switching, not an issue by Marlin, but I can hear it. And while heating it's switching the whole time.

thinkyhead, 2015-03-28:
The latest code patched up any potential buffer overruns with temperature arrays by making them all 4 elements long, which makes sense since theoretically you could use heater 2 and no others. However, if this is related to PID logic, then we may still have to track that down. Awaiting further testing and feedback.

SkyFlyer2, 2015-03-28:
I can test this only in tuesday. The easiest way to check that all ok, just measure the voltage on the heated bed. Must be about 12v (if power supply 12v). Also, the led on it should not blink at all.

dattas, 2015-03-29:
As of f47fc879fe47c3597ad47cb4026888c028c1c55e (was too lazy to pull individual commits, so I just gabbed the latest at the time) this appears to be fixed for me. Thank you @thinkyhead !

closed 2015-03-29 by dattas

thinkyhead, 2015-03-30:
:smiley:

SkyFlyer2, 2015-03-31:
I tested - it seems that everything is fine now.