MarlinFirmware/Marlin #1699: G29 Topographical Map Issue

Roxy-3DPrintBoard, 2015-03-25:
Currently, the G29 has this code to print out a header for the Topographical Map if the TOPO_ORIGIN is set to OriginFrontLeft.   The problem is the C pre-processor is always thinking the comparison of (TOPO_ORIGIN == OriginFrontLeft) is true.   So, first, the header is printing out when it shouldn't.  But it is also keeping the other 3 cases (like mine of TOPO_ORIGIN==OriginBackRight) from working correctly.   Here is the code that is failing:

        #if TOPO_ORIGIN == OriginFrontLeft
          SERIAL_PROTOCOLPGM("+-----------+\n");
          SERIAL_PROTOCOLPGM("|...Back....|\n");
          SERIAL_PROTOCOLPGM("|Left..Right|\n");
          SERIAL_PROTOCOLPGM("|...Front...|\n");
          SERIAL_PROTOCOLPGM("+-----------+\n");
          for (yy = auto_bed_leveling_grid_points - 1; yy >= 0; yy--)
        #else
          for (yy = 0; yy < auto_bed_leveling_grid_points; yy++)
       #endif 

I don't think this is an Arduino problem because this code in Microsoft's Visual Studio also generated unexpected results.  Specifically, if you put this code into an Aruduino program or a Microsoft Visual Studio C file, it does not do what you expect:

enum { OriginBackLeft, OriginFrontLeft, OriginBackRight, OriginFrontRight };
#define TOPO_ORIGIN OriginBackRight
printf("TOPO_ORIGIN=%d\n", TOPO_ORIGIN);
#if (TOPO_ORIGIN == OriginFrontLeft)
    printf("TOPO_ORIGIN does not equal OriginFrontLeft!!!! \n" );
#endif

This code snippet will print out "TOPO_ORIGIN does not equal OriginFrontLeft!!!!"  even though TOPO_ORIGIN is clearly set to OriginBackRight!

NarimaanV, 2015-03-25:
Wait isn't that because @thinkyhead got rid of the TOPO_ORIGIN option?

nophead, 2015-03-25:
#if is evaluated by the C preprocessor which knows nothing about enums or any C symbol names. Any name name that is not defined as a preprocessor macro is silently converted to 0 when evaluated in a #if expression.

Roxy-3DPrintBoard, 2015-03-25:
Yeah, I'm sure its a pre-processor issue.  Or stated differently, an assumption was made that the pre-processor is 'smarter' than it really is.   I've run into that before trying to do floating point things and such.

nophead, 2015-03-25:
Yes I don't think the preprocessor has changed much since ANSI C days. It's use is frowned upon by modern C++ programmers.

enums and if() would work and being constant any dead code would be eliminated at compile time. Or #defined constants with #if for the old fashioned way.

thinkyhead, 2015-03-26:
I suppose there could be a slow migration to using const instead of all these preprocessor defines. One way to do this would be to make an intermediate file that turns every configuration #define into the required const values. Would the compiler also be smart enough to throw away unused local and global variables?

nophead, 2015-03-26:
Probably, but it would need testing. You still have the problem that dead code still needs to compile, so all the symbols it uses must exist at compile time. That doesn't make always sense for features or hardware that people don't have.

Roxy-3DPrintBoard, 2015-03-26:
It sure seems to me that the pre-processor should be smart enough to handle this kind of thing.  And if it can't be smart enough, it should warn you.   But because changing it would break so much code, they probably don't dare make any changes.   

> Would the compiler also be smart enough to throw away unused local and global variables?

Stepping back and looking at it from a higher level:   I think this is dangerous because every compiler is going to treat optimizations differently.  And in fact, the same compiler is going to behave differently depending on the mode it is.  For example, Microsoft's Visual Studio does totally different optimizations if you are debugging (stepping and setting break points) the code versus producing a release version.   So the code size and speed varies dramatically.   

Personally, I don't think the #defines with the matching #ifdef and #endif are that bad.  In a sense, that is what was being attempted except to use the code generator to cut pieces of the code out instead of using the pre-processor to do it.

nophead, 2015-03-26:
The pre-processor is by definition something that happens before compilation so has no access to the syntax or semantics of C/C++. I have used it to process other languages, such as Forth, which would not work if it understood the text it was transforming. After ANSIfication though the preprocessor became aware of C tokens, rather than just treating the source as text, so I wrote my own. 

I don't think removing dead code caused by if(constant expression) counts as optimisation. I think it is guaranteed by C++ in order to be able to have less reliance on the preprocessor. Similarly const int was added for the same reason.

thinkyhead, 2015-03-26:
If the aim is to make the code prettier, we might try moving features into their own files, but there are so many cases where features intermingle with each other in a single function, depending on locals, that it could not be applied universally. The recent Mesh Bed Leveling feature is a fair example. It adds two files to the project, but still requires conditionals in `Marlin_main.cpp`. Its own implementation of `gcode_G29` could be moved to the mesh bed leveling file, but that harms clarity. Overall, I think the code is in pretty good shape. It just needs some better formatting, and I am doing that piecemeal.

thinkyhead, 2015-03-26:
So… as to the main topic, do we want to put back `TOPO_ORIGIN`?

Roxy-3DPrintBoard, 2015-03-26:
> If the aim is to make the code prettier, we might try moving features into their own files, but there are so many cases where features intermingle with each other in a single function, depending on locals, that it could not be applied universally. 

My opinion is the thing to do is put #define's at the highest level possible.  And hide the less important ones at a lower level.   So, in that view of the world, important, global #define's are in Configuration.h.   But ones focused on a particular topic get pushed down.  In fact, I would argue that 3-point bed leveling and grid based bed leveling should be separate files and each of those can have #defines that are only important to them.

> So… as to the main topic, do we want to put back TOPO_ORIGIN?

I certainly think it should be part of the code base.  And just as a sanity check, what if the Grid Based and 3-Point bed leveling were each in their own file?   The #define's like TOPO_ORIGIN would be down in the Grid Based file where it doesn't bother anybody unless they are doing something with Grid Based leveling.  Even the 3-Point leveling crowd would not ever see it.

thinkyhead, 2015-03-27:
@Roxy-3DPrintBoard The LCD options have been reduced a bit by the addition of `Conditionals.h` but I agree, more can still be done. However there is no enthusiasm for adding `Configuration_myFeature.h` files, nor for mixing configuration options into headers with corresponding `.cpp` files. The aim is to have configuration pretty much all in one place, so that all you need to configure a printer with some version of Marlin is `Configuration.h` and `Configuration_adv.h` plus the latest sources.

Natealus, 2015-03-28:
I'm curious if my issue has something to do with this but it is different. I got all my bed leveling stuff working with a servo moving an arm in place and out of the way. One issue I've come across though is that if I set the X or Y probe offset more than +/- 17 then it won't compile saying that it cant reach the probe distance on Y above that...but as it goes through the G29 its not even really getting close to its edge towards the Z tower. 

Seems like it wants to start the grid up at the "front" of the printer opposite of the Z tower and doesn't quite make it back all the way. I wish I could put in the true offset values but we're talking -62 mm Y offset due to my mounts and bulk of the Kraken. 

I realize that technically and physically this is not ideal but I'm working on a different design to cantilever the end stop under the center of the Kraken block or in other words dead center of the effector. I believe this is the best location to get accurate measurements but the implementation and making sure there isn't and unwanted flex in the system shall be tricky.

But I was wondering if this could at all be related to the origin settings discussed in this topic? If not I'll repost in a separate issue.

thinkyhead, 2015-03-28:
@Natealus What are your bed dimensions? What are your MIN and MAX values for X and Y? Is your probe to the left or the right of the nozzle?

Natealus, 2015-03-28:
@thinkyhead 280mm circular bed. So here's another kicker I had to do some heavy code comparisions with the kolossel mini cause that's where the delta leveling stuff was designed originally. There was a missing line of #define MIN_PROBE_EDGE 10 // The probe square sides can be no smaller than this. 

And this section was different...seems to get automated in the code. Otherwise it would be 140 mm each way on X and Y.
// Travel limits after homing (units are in mm)
#define X_MAX_POS DELTA_PRINTABLE_RADIUS
#define X_MIN_POS -DELTA_PRINTABLE_RADIUS
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS
#define Y_MIN_POS -DELTA_PRINTABLE_RADIUS
#define Z_MAX_POS MANUAL_Z_HOME_POS
#define Z_MIN_POS 0

Bed center is 0, 0 on Deltas as well
#define MANUAL_X_HOME_POS 0
#define MANUAL_Y_HOME_POS 0
#define MANUAL_Z_HOME_POS 367.55 // For delta: Distance between nozzle and print surface after homing.

thinkyhead, 2015-03-28:
@Natealus #1721 should clear that up.

Natealus, 2015-03-28:
Well increasing the printable radius to 140 did allow me to bring the Y probe offset to -20 but no further. I'm designing a better system to get the probe closer to the nozzle or center of the effector. But its a pretty tight fit with all the kraken and wires and such lol.

Natealus, 2015-03-29:
I've increased the variable for the probable distance. Guess since its negative it would be decreasing it because it missed the build plate. Its now -31 but now it is probing just about the whole bed which is much better. I'm going to do a test print and see how the layers come out. Back in an hour and half or so.

thinkyhead, 2015-03-29:
@Natealus I think a better approach is needed for delta leveling, anyway. Three-point leveling is actually fine, but 6 point leveling (3 points one way, then the other) might be even better.

Natealus, 2015-03-29:
I'm using 10 point grid leveling actually. Does about 10x10. I say about cause it seems to trim some near the edge. But I'd have to do precise measurements to really say if its circular or not...or put a sharpie on the Z-probe to mark a piece of paper...lol

lrpirlet, 2015-03-29:
It looks to me that there is an WRONG assumption in configuration.h... 
At least the left, right, behing and front in the comment needs to be defined... Just like the topology "left, right, back and front wording" needs carefull definition...
Here are my probe offsets
  #define X_PROBE_OFFSET_FROM_EXTRUDER -19     //-19.20 Probe on: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -9      //-9.40  Probe on: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -2.33   //-2.33  -below (always!)I know that my  My probe is on the RIGHT and BEHIND the hot-end...
I know that the value in my configuration.h are ok... Yes, the safe homing brings the probe right in the middle of the bed before probing.
In fact, my origin 0,0 is in the BACK and on the RIGHT... 

I guess that the definition of right would be between origin and hot-head and front would mean between origin an hot-end... 

The WRONG assumption is that the origin is always in the front left and that when incrementing X and Y the hot-end go back and right relative to the origin...

Natealus, 2015-03-29:
Yeah for a delta the origin should be the center because the radius calculation is center relative to towers to get a flat movement. On the current Dev build my machine is doing "front-right" origin or basically right by the Y tower. 

I've commented out #define Z_RAISE_AFTER_PROBING    20 because it seems the Z height after homing is basically what that variable is. It ignores my manual Z value and thus won't probe to the bed if it only goes down 20mm rather than 365mm. That particular issue has been present for days though since I first got all the probe hardware hooked up.

Natealus, 2015-04-01:
@thinkyhead I know you've been looking at the bed code alot and fixing up alot of stuff...are those two issues happening on other types or just Delta? I just re-configured the newest dev build and the g29 is still only probing a quarter pie of my plate near the Y tower.

I noticed the arbitrary coordinates just below the grid leveling and tried to change a couple to be negative variables since center and nozzle at bed is 0, 0, 0 and the opposite corner from where it's currently probing would be -140, -140, 0. Am I correct in guessing that with other printer types; 0, 0, 0 is in a corner of the square build plate and the values only go positive from there? It seems like it's only probing positive X position values...and oddly enough -Y position values. But changing the arbitrary points didn't seem to change the probing pattern at all.

On a secondary note I was able to increase the Y Probe Offset to -31 which is half way to being actual lol. But especially with it only probing a quarter of the circle it's getting a very wonky calculation for trying a print in the center.

And on a thirdly note...kudos to the hard work you guys. :) I know its a lot of headache to look through line after line after line of code to see where a particular oddity happens. Hell I've been getting a headache myself and I hardly venture into Marlin_main.cpp. Good work guys. :)

Natealus, 2015-04-01:
@boelle  this isn't exactly a new feature...this is bug fixing for bed leveling code.

boelle, 2015-04-02:
So we should remove the feature request label and add verified bug instead? or add verfied bug on top of the feature request?

Natealus, 2015-04-02:
I could create a new issue specific to the problem if you'd like but yeah I am still experiencing the "Quarter-Pie" scan rather than a whole bed scan on my Delta. Which dramatically throws the calibration off. It needs to probe the whole circular bed and compare the furthest points to the center to get an appropriate delta radius calculation.

boelle, 2015-04-02:
nah its np... its more that we get the right label so in the end it lands on a bug fix milestone and not a feature request one.... the last one will mean it goes on the back burner until bugs are fixed and we dont want that :-D

Natealus, 2015-04-02:
Hehe I was like...but...but...but... :) I've taken my printer off printing to test out all this bed leveling stuff. It's a pretty complicated system to implement. Lots of sequences have to orchestrate just right. I'm trying to be detailed with my descriptions so the coders can find the issues easier. Of course that's relative when it comes to thousands of lines of code lol.

thinkyhead, 2015-04-03:
@Natealus Gosh has it been 2 days since I visited the issue queue?

At the moment I am preparing a PR which does not so much as just cleans up and documents the code around G28 and G29, but I have not as yet done any bugfixing. However, my unread issues list is long, and perhaps by the end of it I will find someone posted a solution – best to check the PRs first…

I'l check in again in a bit.

Natealus, 2015-04-03:
Its been kinda quiet code wise the last couple of days...its been eerie...lol. I've just been trying to test and verify in posts.

lrpirlet, 2015-04-16:
If that is of any help, and if the logic still holds, I would suggest modifying the output of the G29 topology map to read: 

```C++
        // Show the Topography map if enabled, note Origin is O
        if (do_topography_map) {

          SERIAL_PROTOCOLPGM(" \nBed Height Topography: \n");
          SERIAL_PROTOCOLPGM("|+-----------+\n");
          SERIAL_PROTOCOLPGM("Y|...Back....|\n");
          SERIAL_PROTOCOLPGM("^|Left..Right|\n");
          SERIAL_PROTOCOLPGM("y|...Front...|\n");
          SERIAL_PROTOCOLPGM("|O-----------+\n");
		  SERIAL_PROTOCOLPGM("O---x-->--X---\n");
```
The Origin would be the corner desigated with an O, both X and Y axis would be represented to get a normalized orthogonal set of axis (assuming Z axis grows when going up and away... 

Now, if the Origin is in the back-right as in my machine, I only need to turn the printer around before to observe-it... (or to turn the print around if that is easier)

BTW, the comments in the Configuration.h should also be modified. Whenever there is a reference to front, back, left or right, I would suggest making a note explaining that those locations are correct with the Origin in the front left...

nophead, 2015-04-16:
References to front, back, left or right still make sense when the origin is in the centre. It doesn't have to be front left. The only difference is the coordinates range from negative to positive.

boelle, 2015-08-03:
This issue was moved to MarlinFirmware/MarlinDev#16

closed 2015-08-03 by boelle