MarlinFirmware/Marlin #1701: Z Probe designs with Z Endstop triggered while retracted - feature request

Ziggy2013, 2015-03-26:
Marlin currently does not support Z Probe designs where the Z Endstop Min is always triggered while the Z probe is retracted. An example of a Z Probe design using an Opto endstop is shown here

http://forums.robo3dprinter.com/index.php?threads/auto-bed-leveling.1590/#post-16959

Similar Z Probe designs could be developed using hall effect switches etc.

The issue is that Marlin sees that the Z Endstop looks like it has been triggered immediately the Z Probe is "engaged". Therefore the probing does not work.

thinkyhead, 2015-03-26:
I wonder if – like NC endstop switches – you have to invert the state. I'll check to see if the probe is implemented in the same way as the endstop.

thinkyhead, 2015-03-26:
Looks like `Z_MIN_ENDSTOP_INVERTING` also applies to the Z probe. Try changing its value.

Ziggy2013, 2015-03-26:
Inverting doesn't work because the probe is using the endstop to detect when it touches the bed. So the sequence is

Probe stowed = endstop triggered
Probe Extended  = endstop is open
Probe touches bed = endstop triggers
Probe retracted = endstops triggered again

The solution I found was to turn off all endstops before engaging the probe, then turning the endstops back on immediately after the probe was extended. Alternative (and better) solution would be to stop the engage_z_probe() function checking the endstop before the probe is actually engaged.

Changes in Marlin_main.cpp

  do_blocking_move_to(x - X_PROBE_OFFSET_FROM_EXTRUDER, y - Y_PROBE_OFFSET_FROM_EXTRUDER, current_position[Z_AXIS]);

enable_endstops(false); // mod - turn endstops off
  
  engage_z_probe();   // Engage Z Servo endstop if available
  
enable_endstops(true); // endstops on again

  run_z_probe();
  float measured_z = current_position[Z_AXIS];
  retract_z_probe();

thinkyhead, 2015-03-26:
Waiting for the probe to be engaged means basically adding a brief delay right after the probe is extended. Or, in your case, you could check the state of the probe switch before, then extend it, and wait for the switch state to change.

Ziggy2013, 2015-03-26:
In that case a generic solution is to do what I did. Turn off the endstops before engaging the probe. Then turn them back on after it is engaged.  This works reliably and does not introduce any delay.

But I will have a look at why the engage_z_probe() function checks the Z endstop too early.

thinkyhead, 2015-03-26:
@Ziggy2013 Well, there's just no delay after the servo is told to deploy, so it probably just needs a short delay to wait for the servo to finish its move.

Ziggy2013, 2015-03-27:
@thinkyhead 

After some investigation and testing I now have a better understanding of what is happening in this situation where the Z endstop is triggered while the probe is stowed.

The Arduino servo library uses timer interrupts to send pulses to the servo(s) when a servo.write is requested. In the case with Marlin this means the servo will be moving while other functions are executing. In fact a servo typically takes about 200-300 milliseconds to move position, so Marlin can be handling many other functions while the servo moves.  If the endstops are checked during this 200-300 ms while the servo is moving, then the Z probing screws up.

IMO the most straightforward way to prevent other Marlin functions being executed while the servo moves is to introduce a loop type delay() immediately after the servo.write. This delay needs to be long enough that the servo can finish moving.
 
As it happens there is already a configuration parameter in Marlin which introduces a loop type delay() after the servo.write.  In configuration.h  the parameter

  #define PROBE_SERVO_DEACTIVATION_DELAY 300      // the value is in milliseconds

is meant to allow the servo to be turned off when not being used.  But the way this delay is implemented in engage_probe() and elsewhere means the servo can be given time to finish moving before anything else is executed.

So the solution turns out to be simple.  For Z probe designs where the Z Endstop is triggered while stowed the 

  #define PROBE_SERVO_DEACTIVATION_DELAY 300  

parameter must be defined with a value in milliseconds which allows the servo to finish moving before anything else is executed.  It would be good if this explanation were added to configuration.h. The name of the parameter could/should also be changed to PROBE_SERVO_ACTIVATION_DELAY as this better describes what it actually does.

Hope this all is clear.

thinkyhead, 2015-03-27:
Makes sense. I will look at implementing this soon.

alexborro, 2015-03-28:
I have implemented all these routines.  It turns the servo power off after
movement (and a delay). You can use it for any other purpose, but the
primary one is to avoid the servo flickering. So do not change its name.

Cheers.

Alex.

thinkyhead, 2015-03-28:
@alexborro Are you saying the code is already in-place, just that the delay needs to be longer? My reading comprehension is awful today.

Ziggy2013, 2015-03-28:
@alexborro My understanding is that the servo moves after the servo.write is called (as the servo pulses are interrupt driven by a timer).  

So the delay() after the servo.write needs to be long enough to allow the servo to finish its move (via the interrupts).  If the delay() is not long enough, the servo could be in an unknown position when the code following the servo.write is executed. Having the delay() kinda defeats the purpose of the interrupt driven servo.write but otherwise the servo will not be in a known position.

In other words I don't think having the delay() after the servo.write is optional

closed 2015-09-25 by AnHardt