MarlinFirmware/Marlin #1703: Fan Start when Homing!!

Suduku, 2015-03-26:
Hi, I have a MM2 with Taurino power and Ramps 1.4 for 24V from ReprapDiscount.
I upgrade Marlin with new version, and I have a problem with the D9 Fan, that worked fine with the older version.

Now when I do a homing (X,Y or Z) the Fan start, I change nothing with electronics and if i Flash the board with the older marlin Version, all work fine again.
It's a Firmware configuration But I don't see where I must change something to avoid having the Fan start when homing.

Classic config : BOARD_RAMPS_13_EFB

Thanks for the answer

AnHardt, 2015-03-26:
I can confirm this.
Fan-display will show 39%.
It's independent of sending G28 per menu or per serial.
Can't trigger the failure by simply jogging x,y or z.
Can trigger with homing one axes alone.
 BOARD_RAMPS_13_EFB - Arduino 1.6.1

Can NOT confirm the failure with Arduino 1.0.6 !!!

Not again - please.

AnHardt, 2015-03-26:
The first time i can see this is with https://github.com/MarlinFirmware/Marlin/commit/0ce35766855c53c95df9d7e065f774534937df96 New Feature: Z_DUAL_ENDSTOPS.

AnHardt, 2015-03-26:
Some hints what could be related.

Marlin_main.cpp:1783: warning: array subscript is above array bounds
https://github.com/MarlinFirmware/Marlin/blob/0ce35766855c53c95df9d7e065f774534937df96/Marlin/Marlin_main.cpp#L1783
https://github.com/MarlinFirmware/Marlin/blob/0ce35766855c53c95df9d7e065f774534937df96/Marlin/Marlin_main.cpp#L388
https://github.com/MarlinFirmware/Marlin/blob/0ce35766855c53c95df9d7e065f774534937df96/Marlin/Marlin_main.cpp#L247
https://github.com/MarlinFirmware/Marlin/blob/0ce35766855c53c95df9d7e065f774534937df96/Marlin/Configuration.h#L504

Marlin_main.cpp:4267: warning: the address of 'float code_value()' will never be NULL
https://github.com/MarlinFirmware/Marlin/blob/0ce35766855c53c95df9d7e065f774534937df96/Marlin/Marlin_main.cpp#L4267

/pins_RAMPS_13.h:42:1: warning: "Z2_STEP_PIN" redefined
/pins_RAMPS_13.h:43:1: warning: "Z2_DIR_PIN" redefined
/pins_RAMPS_13.h:44:1: warning: "Z2_ENABLE_PIN" redefined

AnHardt, 2015-03-26:
First aid would be:
'<' instead of '<=' in
https://github.com/MarlinFirmware/Marlin/blob/0ce35766855c53c95df9d7e065f774534937df96/Marlin/Marlin_main.cpp#L1783

AnHardt, 2015-03-26:
Fan error has gone with #1704.

@alexborro @thinkyhead 
Would you please have a look on to the other warnings.

thinkyhead, 2015-03-26:
@AnHardt I have my Arduino set to verbose, but I don't see warnings. Is there some way I can get those to emit within the IDE?

AnHardt, 2015-03-26:
@thinkyhead 
Same problem her wit 1.6.1 - but my 1.0.6 compiles with -Wall.

AnHardt, 2015-03-27:
@thinkyhead 
Search for ....\Arduino\hardware\arduino\avr\platform.txt
and change the options.

C:\Users\ah\AppData\Roaming\Arduino15\preferences.txt
build.verbose=true
console.length=2000

thinkyhead, 2015-03-27:
I have verbose already set. However I still don't see warnings in the console output. I will try Arduino 1.6.1.

AnHardt, 2015-03-27:
@thinkyhead 
With 1.6.1 - a total different experience than with 1.0.6 where you hardly can find  anything but 'only initialized variables can be placed into program memory area'.

Now i can see there are some more warnings.
More tomorrow.

.https://gist.github.com/AnHardt/f0d6bc2587e8945f29a0

thinkyhead, 2015-03-27:
Ok, I used `-Wall` and got rid of all the warnings with the default configuration. I will try all the example configs later and clean up their warnings as well.

closed 2015-03-27 by thinkyhead