MarlinFirmware/Marlin #1706: MAXTEMP triggered after PowerOn

floyd871, 2015-03-26:
After flushing the last DEV-Branch in my Delta Printer (Megatronics V3) the Printer stops after Powering on:
----------------------
Error:0
[ERROR] Error:0

: Extruder switched off. MAXTEMP triggered !!
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
[ERROR] Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
----------------------
After a bit look in the Code i have revert the changes made in https://github.com/MarlinFirmware/Marlin/pull/1689

Now i could heat up  the Extruder and Headbed again.

AnHardt, 2015-03-26:
Hello @floyd871 
What IDE do you use (1.0.6, 1.6.1)?
What temp-sensors do you have (Nr.)?

floyd871, 2015-03-26:
IDE 1.6.1
Extruder / Bed: 1

thinkyhead, 2015-03-26:
@floyd871 Trying to isolate these temperature issues, I have a suspicion. Starting with the latest code, try changing these lines:

    #ifdef TEMP_SENSOR_1_AS_REDUNDANT
      #define TEMP_SENSOR_COUNT 2
    #else
      #define TEMP_SENSOR_COUNT EXTRUDERS
    #endif

To:

    #define TEMP_SENSOR_COUNT 4

If this helps, I will make the necessary changes in #1700 .

AnHardt, 2015-03-26:
Error AND fix confirmed. 
Switching to 1.0.6 now and have a test.

Yes. Overwriting other variables gives configuration and compiler depended different errors. :pray:

thinkyhead, 2015-03-26:
@AnHardt I guess we're stuck with 4 elements for all the temperature-related code, because the code allows you to have thermistor 2 set to something while thermistor 0 is set to 0.

AnHardt, 2015-03-26:
@thinkyhead 
Don't worry - just some bytes of RAM.
I think the major part of #1689 can be reverted. I only shifted the the error to an other place. But please don't forget the MAX6675 and the PSTR().

thinkyhead, 2015-03-26:
#1707

AnHardt, 2015-03-26:
As far as i see also OK with 1.0.6 and my 6 standard configurations - including MAX6675.

thinkyhead, 2015-03-27:
Ok, just need to see if #1707 solves things for @floyd871 too…

floyd871, 2015-03-27:
Now i have a complete different situation, the Printer powers up with no Error. But heats up the Extruder and Hotend immediate. Set to extrem high temperatures. I could only switch off:

[image]

thinkyhead, 2015-03-27:
@floyd871 Well, that's not good…!

I will have another thorough look at the temperature code over the weekend and get this sorted out. Will you be available to help test?

floyd871, 2015-03-27:
Sure, go ahead.

floyd871, 2015-03-28:
Last updates to https://github.com/MarlinFirmware/Marlin/issues/1711 fixes mine too

oysteinkrog, 2015-03-28:
Fixed for me in head

closed 2015-03-28 by thinkyhead