MarlinFirmware/Marlin #1708: CoreXY: step_event_count

Salandora, 2015-03-27:
Hello,

Today I wanted to update my Firmware again, so I pulled the last Development version as always.
While reviewing the files I noticed the following in the planner.cpp

    #ifdef COREXY
        // corexy planning
        // these equations follow the form of the dA and dB equations on http://www.corexy.com/theory.html
        block->steps[A_AXIS] = labs(dx + dy);
        block->steps[B_AXIS] = labs(dx - dy);
      #else
        // default non-h-bot planning
        block->steps[X_AXIS] = labs(dx);
        block->steps[Y_AXIS] = labs(dy);
      #endif

...

    block->step_event_count = max(block->steps[X_AXIS], max(block->steps[Y_AXIS], max(block->steps[Z_AXIS], block->steps[E_AXIS])));

    // Bail if this is a zero-length block
    if (block->step_event_count <= dropsegments) return;

There is no  A and B_AXIS in the step_event_count which might end in a Segment drop if there is no Extruder movement or Z-Movement.

thinkyhead, 2015-03-27:
Those are equivalent to X and Y, just named differently for COREXY.

    enum AxisEnum {X_AXIS=0, Y_AXIS=1, A_AXIS=0, B_AXIS=1, Z_AXIS=2, E_AXIS=3, X_HEAD=4, Y_HEAD=5};

Salandora, 2015-03-27:
Ohh ok sorry didn't saw this.

closed 2015-03-27 by Salandora