MarlinFirmware/Marlin #1709: max_inactive_time corruption

alvinavr, 2015-03-27:
I am on version #1698 from yesterday.

The value of max_inactive_time is set to decimal 100 when a G28 or G29 command is issued.
This causes kill() to be called. A simple move of the X or Y carige does not trigger this.

I added some debug code to M85 to see what was happening. At power on the value is 0, correct. The M85 command correctly sets and reads(debug code) this value. I commented out the call to kill() and all works OK.

This has the look of a memory corruption bug. Any hints on how to find it?

I am testing on a RepRapPro Huxley. The only non configuration change I made is to copy and paste the algebraic thermistor code from the RepRapPro Marlin software. If I use the thermistor tables the bed temperature is wrong. 

Alvin

thinkyhead, 2015-03-27:
You're probably right about it being a buffer overrun. #1714 patches up temperature arrays just in case, so see if that makes a difference.

alvinavr, 2015-03-28:
I can confirm that the problem is fixed in #1707.

closed 2015-03-28 by alvinavr