MarlinFirmware/Marlin #1711: Target Bed Temp = 368° after startup on REPRAP_DISCOUNT_SMART_CONTROLLER

SebDE, 2015-03-27:
After flash this firmware with minimal configuration ( Board: RAMPS_13_EFB, LCD: #define REPRAP_DISCOUNT_SMART_CONTROLLER) the printer displays 368° bed temperature on lcd and starts to heat the bed. Hardware seems to be okay cause if I flash my old firmware (1.0.0) everything works fine.

emartinez167, 2015-03-27:
Temperature management really seems to be completely screwed. I turned my printer and didn't realize the extruder started heating by itself! If I hadn't checked the temp graph by accident, I would have now a burnt house! This needs serious attention!!!

Regards,

Ernesto

scotty1024, 2015-03-27:
I saw the same temperature with RAMPS_13_EFB if I had the thermistor set to -1 which had it using a non-existent thermocouple which apparently reads as ~368C.

oysteinkrog, 2015-03-27:
I got this problem as well, was very unexpected that it started heating just after flashing!

oysteinkrog, 2015-03-27:
Latest dev is certainly acting very strange, the target temp fluctuates randomly (when scrolling in menu?).

oysteinkrog, 2015-03-27:
Rebasing my config back to before 410f64782acb30083ae8abfb7f4719061bf1983d fixes it (though I now get the max bed temp error instead).

CheapGSean, 2015-03-27:
Yep, Same Problem here too. My bed starts heating up on startup of the ramps board, and my lcd claims that it is trying to heat to 365°. Odd.

thinkyhead, 2015-03-27:
Please check if #1714 fixed the issue.

AHHams, 2015-03-27:
Please also correct same in temperature.h

thinkyhead, 2015-03-27:
Done. So test the latest code and let me know if all is well.

thinkyhead, 2015-03-27:
Thank you bleeding-edge testers for your feedback! Temperature and leveling bugs have existed for a few days as a result of some comprehensive overhauls to the code.

Remember that the Development branch _will have bugs_, and that yes, your printer _can start fires_! Don't leave your printer unattended, especially with the latest code – for which __you are the tester__.

I look forward to feedback on the latest fixes, applied even before I've had my breakfast!

AHHams, 2015-03-28:
Temperature reading and controls work fine again for me. Thanks for the great work!

closed 2015-03-28 by thinkyhead