MarlinFirmware/Marlin #1719: Y axis is homing with Z when extruder assembly is unwired and MINTEMP is triggered.

dmmedia, 2015-03-28:
I have uplugged extruder cable, which contains the following wiring:
- extruder stepper;
- extruder heater;
- extruder fan;
- extruder thermistor;
- part fan.
After that I issued the command G28 Z0 by pressing Z home button in Repetier Host and Z axis started to home with Y axis.
I have QUICK_HOME enabled if that helps, but it should not interfere here.
When extruder is plugged in, then all axis are homed one at a time when I press appropriate axis home button.

alexborro, 2015-03-28:
Try pressing just the X homing button and check the behavior.
The try just the Y Homing button.

Cheers.

Alex.

dmmedia, 2015-03-28:
X homing button triggers X and Y homing.
Y homing button triggers only Y homing.

Also I have updated to the latest version and behaviour observed is the same.
But MINITEMP now triggers requirement of M999 to be sent to reset the printer which causes another issue in communication between Repetier Host and the printer. I will probably create new separate report about that after testing.

AnHardt, 2015-03-28:
If this cable is still unconnected - then your thermistor is unconnected and therefore 'Err: MINTEMP' and the requirement of M999 the expected behavior.
Please tell us more about your machine. Is it a coreXY?

dmmedia, 2015-03-28:
Printer is RigidBot, CoreXZ.

Home all behaviour is correct: X and Y first (QUICK_HOME), then first finish X, next Y and at last Z.

alexborro, 2015-03-28:
Where did you see Marlin has support for CoreXZ kinematics??

dmmedia, 2015-03-28:
I have commented out  define COREXY.

So everything else seems to work properly, except for homing behaviour.

I have performed more tests and it seems that extruder thermistor (and MINTEMP error) do not interfere with this issue.

I have plugged extruder in and still observe strange homing behaviour.

alexborro, 2015-03-28:
Sorry but a CoreXZ bot will not work on a CoreXY kinematics.

That is exactly the reason you are having this issue.

Try moving the axis X,  Y and Z each a time.

Cheers

Alex

thinkyhead, 2015-03-29:
CoreXZ isn't really supported in the current Marlin, is it? Not without some major tweaking.

dmmedia, 2015-03-29:
Hey, I have line `// #define COREXY` left commented, so CoreXY kinematics should not be the case.

I have tons of prints and many many hours of printer work. Everything was working properly until recently issue [Issue #1605] (https://github.com/MarlinFirmware/Marlin/issues/1605) occurred after one of commits. And that was minor issue.

Current issue occurred after I have pulled commits made over last seven days.

So if I properly understood, support for CoreXY advanced kinematics is a configurable Marlins feature and if it is switched off, then any conventional 3-axis printer is supported. And deltas also have own configurable support which can also be turned on or off.

dmmedia, 2015-03-29:
Sorry, my mistake of understanding what is CoreAnything. Thought it is simply which axes are tighten together, but it turned out to be complex mechanical system.
So my printer is not CoreXZ at all.
And issue still present.

thinkyhead, 2015-03-29:
Just curious, in case it's code common to both routines, what kind of behavior do you see with `QUICK_HOME` turned off?

thinkyhead, 2015-03-30:
#1736 fixes the typo, and makes a note about the home_offset being applied when setting positions. For example, currently the code will take:

    G28 X Y

…or…

    G28 X0 Y0

…the current position will be set to `X = 0` and `Y = 0`. But if you do this:

    G28 X0.0001 Y0.0001

The result will be `X = home_offset[X_AXIS] + 0.0001` and `Y = home_offset[Y_AXIS] + 0.0001`. We recently took these home_offset values out of `G92` and I think that if `G28` is supposed to be consistent with `G92` then we should remove them here also.

Another thing I noticed is that the `Z_RAISE_BEFORE_HOMING` values are being applied only _after_ the X and Y axes are already homed, and just before homing the Z axis, which would seem to defeat the purpose. But note that the `Z_RAISE_BEFORE_HOMING` value only applies if `ENABLE_AUTO_BED_LEVELING` is active.

So I wonder if that needs to be changed also, or if this is somehow intentional.

closed 2015-03-31 by thinkyhead