MarlinFirmware/Marlin #1725: Latest updates broke the homing capability.

emartinez167, 2015-03-29:
I have applied the latest changes on my Azteeg X3 Pro. First thing I noticed is that the MIN END STOPS logic appeared reversed so when I tried to Home X, the print head kept trying to go beyond 0. Subsequently, I changed the following in Configuration.h

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.

Now, what I am getting is that homing does not work at all. In fact, all the motor does is advance a bit and then stop. Is there something I should have been aware of? Should I revert to the original logic as the problem might be somewhere else?

emartinez167, 2015-03-29:
I have reverted to leaving the MIN END STOP logic as it was, because it will make the step motors move in the right direction, but still need to find out why the motors keep trying moving if the end stop has been triggered.

Natealus, 2015-03-29:
The changes I did swapped MIN and MAX mostly because I use a Delta and it wants to home to MAX. If you // that whole section they will be as normal. X min will be X min. Perhaps I can recommit that as default and my way as the option you have to manually uncomment.

Natealus, 2015-03-29:
and Heater 1 isn't -1 anymore. HE 1-4 are hot ends...5 is the controllerfan, 6 is the Extruder Auto Fan, and 7 is the Part cooling fan. I also added in servo functionality but that shouldn't do anything unless you define servos in configuration.h

dmmedia, 2015-03-29:
Check out endstop location in your config:

    // ENDSTOP SETTINGS:
    // Sets direction of endstops when homing; 1=MAX, -1=MIN
    #define X_HOME_DIR -1
    #define Y_HOME_DIR -1
    #define Z_HOME_DIR -1

Natealus, 2015-03-29:
Reason I've reversed the pins in the past is cause when I tried switching those to home to min...all sorts of weird crap was happening and it wasn't even stopping when it hit the endstops.

emartinez167, 2015-03-29:
The Azteeg X3 Pro official pinout states that:

X_MIN_PIN = 2
X_MAX_PIN = 3
Y_MIN_PIN = 14
Y_MAX_PIN = 15
Z_MIN_PIN = 18
Z_MAX_PIN = 19

Both Y and Z are reversed in the latest changes. I changed that to what Azteeg states, but behavior is still there.

Regards,

Ernesto

emartinez167, 2015-03-29:
Yes, but I assume that has broken it for the majority. If this is a Delta behavior, it should be managed using conditionals for Delta printers.

Regards,

Ernesto

Natealus, 2015-03-29:
Yeah if you pull that entire section out of your pins_azteeg_x3_pro.h it will be that original pin set as it's the same as the ramps_13 board.

emartinez167, 2015-03-29:
I disagree with your approach. Pinout for a board should be as stated by manufacturer. Differences caused by particular printer models should not cause issues for other printers, and should be handled by conditionals. Please revert this change.

Regards,

Ernesto

emartinez167, 2015-03-29:
Endstop setting are fine (-1) and should remain like that, as they are in alignment with physical layout of the printer. 

Regards,

Ernesto

Natealus, 2015-03-29:
Well most boards I've looked at either have all 6 but for some reason Panucatt decided to only put Min Endstops in this board. I didn't know there were too many other people with X3 Pros here. Couple of other people I know have a delta too. I can make my way the optional just have to have someone merge it in.

emartinez167, 2015-03-29:
It should be easy, just put the changes inside an #ifdef, that way only you will have them reversed.

Regards,

Ernesto

Natealus, 2015-03-29:
I'll have to test an #ifdef DELTA branch and see if it works.

emartinez167, 2015-03-29:
I see thee were other changes to the planner before your commit that might be the reason why the endstops are not working any more... Has to do with some arrays being created... And other parts of the code that rely on the older style not being changed accordingly. I am rolling back all changes up to that pont to see if that is the commit that broke it. More after I finish this.

Regards,

Ernesto

Natealus, 2015-03-29:
Yeah once my Viki 2 pull gets merged the X3 Pro pins will be fixed. That #ifdef DELTA worked just fine on my setup so that's how I structured it.

emartinez167, 2015-03-29:
I can confirm that PR #1723 from @thinkyhead broke homing. I rolled back the changes to before that and it all works fine. Nate, your pin swaps should still go into a Delta conditional.

Regards,

Ernesto

Natealus, 2015-03-29:
That is kind of odd. I'm on the latest commits too aside from pin stuff that still needs to be merged and I haven't had any problem with homing. It did change the G29 probing a bit I'm going to look at it more thoroughly tomorrow.

emartinez167, 2015-03-29:
That might be because the Delta setup is different that non-delta, hence some changes to the code could affect only non-deltas and spare you.

Regards,

Ernesto

Natealus, 2015-03-29:
It feels nice to be lucky I guess hehe. But yeah it's a shame I don't have a printer of each type to do some really effective testing. :)

emartinez167, 2015-03-29:
That's why it's good to have us testers with differerent hardware
configurations!

thinkyhead, 2015-03-30:
Pin swaps are now conditional for delta only, so that should no longer be causing trouble.

Meanwhile I can't see anything obvious in #1723 that affects homing… so far. Today's patches are focusing on `G28` and `G29` however, specifically for Delta. Leveling wasn't being applied to Delta, as a very important line of @jcrocholl's code was left out, so that should now be fixed.

As we keep focusing on these functions further, they should start to get a bit more solid. Contributors are experimenting with auto-leveling as well, to get best results and bring the code size down a bit. Much attention on this code region…

Natealus, 2015-03-30:
Hehe I was thinking it wasn't looking so stellar for having additional leveling points but it also has been originating near the Y tower or "front-right"? The other day after I changed the printable radius to 140 it seemed to catch the whole bed at that point. Still haven't been able to set my true 62mm Y Probe offset but it is a bit out of the best case scenario.

thinkyhead, 2015-03-30:
@Natealus How is 3-point leveling for you on the Delta? Is that something we even allow?

Natealus, 2015-03-30:
Not as yet from what I saw it seems like you either get auto grid leveling or you do manual mesh leveling which seems to defeat the purpose of having a z-probe but that's probably a further new feature to make that automatic. I think a 4-point tower to center would do a pretty good calculation...if someone wants a more specific one then they could do the grid and hit the whole bed.

thinkyhead, 2015-03-30:
@Natealus I was thinking, 6 points around the perimeter on a round delta, forming 2 triangles, and correlate the angles of those. On a Simpson, read the three points of its triangular bed.

Natealus, 2015-03-30:
Yeah that is a pretty good idea but I do think we also want a center probing too don't we? Cause that's usually how you fine tune the delta radius is getting the nozzle the same height at all 3 towers and then check the center and adjust the radius up or down if its tighter or looser than the towers.

thinkyhead, 2015-03-30:
@Natealus Ah yes, well in that case, you get 3 triangles with 3 perimeter points, or 6 with 6 perimeter points, more than enough…

emartinez167, 2015-03-30:
This has been fixed with the latest commits. It can be closed.

closed 2015-03-31 by thinkyhead