MarlinFirmware/Marlin #1737: Z_RAISE_BEFORE_HOMING only after homing XY?

thinkyhead, 2015-03-30:
As noted in #1736 it looks like `Z_RAISE_BEFORE_HOMING` occurs just when homing Z and only after homing X and Y. Am I misunderstanding the purpose of this setting, or should it be applied first, to raise the Z axis, before homing X and Y in `G28`?

AnHardt, 2015-03-30:
For 'save homing' is a requirement that x & y are homed.
If we have the sensor on a servo we need some space.
Lifting the nozzle before homing x,y is a goody to jump over the bed mounting screws/clips.
When homing z the switch has to be unswitched! (See https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Configuration.h#L447 )

Sniffle, 2015-03-30:
In my logical way of thinking it makes sense to me that if you definite a height to raise before homing and if the hotend is too close to the bed when homed for the servo(if using a servo) to extend but can't because the hotend isn't being raised before starting the homing process something is wrong.  

Ive ran into this a few times over the last week trying to double and triple verify the bed leveling bug.  At this point I've just gotten into the habit of raising the hotend 10mm before I restart g28/g29

thinkyhead, 2015-03-30:
So, are you confirming that `Z_RAISE_BEFORE_HOMING` is not being applied before homing X or Y? That certainly looks like what the code is doing, but I haven't tested it. If this is the case, then I will move the Z lift in front of the X and Y homing, and it will be applied only in the appropriate cases.

- Only if `SERVO_LEVELING` is on and Z homing direction is -1
- With `G28` (home all) or a `G28` that includes `Z`

It's also the case that this particular setting is only allowed for `ENABLE_AUTO_BED_LEVELING`. It would be nice (in spite of protestations against making this available) if it was also available when not using a probe. For machines with both Z max and min endstops, there is no caveat.

woodsmoke, 2015-03-30:
I'll confirm it. I see the Z raise only after G28XY finishes.

oysteinkrog, 2015-03-30:
This has been a big bug for me, I ended up modifying the LCD menu etc to only issue G28 X Y, otherwise I would end up with head crashes all the time since my Z servo endstop could not deploy.

thinkyhead, 2015-03-31:
#1748 has the Z axis raise before homing X and Y, but still homes Z last.

closed 2015-05-09 by thinkyhead