MarlinFirmware/Marlin #1739: Status poll of Graphical LCDs and Printing Performance

Natealus, 2015-03-30:
I was just curious how other people's experience are with the LCD screens and printer performance?

I've had my delta printer running at 20-80mm/s and 165 delta segments per second very smoothly. But with my Viki 2 Graphics LCD enabled, I have to drop the segments all the way to 38 to achieve smooth printing again.

What are your experiences?

daid, 2015-03-30:
I'm not surprised. The used LCD libraries are made for functionality, not for real-time behavior. This is why I opted for a home-grown LCD library for the Ultimaker2 instead of a "standard" Arduino LCD library. (The Ultimaker2 uses a 128x64 I2C LCD display)

clefranc, 2015-03-30:
@Natealus Same thing here with a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. I was at 120 segments a few weeks ago, I'm now way under 100 now.

Natealus, 2015-03-30:
So does it affect Deltas only or does it affect other printer types too?

thinkyhead, 2015-03-30:
@Natealus Ultimaker2 is a very nice box cartesian.

@clefranc Are you able to fit the code if you compile with `-O3` as the optimization level? I would be curious whether this makes any noticeable difference in performance.

Natealus, 2015-03-30:
@thinkyhead The Azteeg X3 Pro board I have is one of those Mega2560 256k boards I think it could spare more space for a possible upgrade of performance...where is this accomplished? 

So with an Ultimaker2 box cartesian like that does it use processor power with kinematics as well?

thinkyhead, 2015-03-30:
To change the optimization you need to find the `platform.txt` file that your Arduino app is using. http://www.instructables.com/id/Arduino-IDE-16x-compiler-optimisations-faster-code/?lang=pt&ALLSTEPS

thinkyhead, 2015-03-30:
@Natealus Cartesian is the most "straightforward" and requires no kinematic transform. Delta and SCARA require some trigonometry, and those math functions are expensive.

Natealus, 2015-03-30:
Thanks a bunch @thinkyhead I'm going to study that article. I've also merged in your code that hasn't merged in the main dev yet and all compile well. Going to give it a go. :)

clefranc, 2015-03-31:
@thinkyhead 
-Os : 120918 bytes
-O3 : 175908 bytes
There is a very (very) small improvement with -O3, not enough to call it a viable solution.

roboprint, 2015-03-31:
Same issue with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and Delta-printer.

Maybe it possible to implement something like "screen saver" to disable screen refresh after print start until keypress (rotary encoder rotate/click) ?

clefranc, 2015-03-31:
@roboprint
Nice idea.
Like a new command to disable the refresh MXXX S0|S1
Or something like a long press to disable the refresh (with a long beep), and another long press to enable (short beep).

roboprint, 2015-03-31:
@clefranc maybe just specify screen refresh rate in seconds? Mxxx S0 - disables, Mxxx Syyy = sets refresh rate to yyy seconds? Long keypess is also good can be used.

clefranc, 2015-03-31:
@roboprint Yes!
@thinkyhead What do you think about it?

AnHardt, 2015-03-31:
@roboprint 
Just try playing with: https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L305

thinkyhead, 2015-04-01:
Less screen refresh is always a good thing. It might also be possible to just draw less on the screen on each update. Instead of updating all the items on the display, just update one item per update-session. This will make the update routine itself block for less time.

AnHardt, 2015-04-01:
@thinkyhead 
I'll  work on the extruder-bed-fan graphic today. here we have a update of a lot of white space. Splitting into 4 separate graphics can improve that.

Natealus, 2015-04-01:
I must say going from default Os optimization to O3 I was able to turn my delta segments per sec up from 38 to 60...so I can say there is an improvement given the same code. :+1: But compared to my previous smooth prints at 175 the LCD sure does eat a wallop of processor lol.

boelle, 2015-04-01:
is this a bug or?

Natealus, 2015-04-01:
Well thinkyhead referenced an article up above about the platform.txt it's the compiling optimization routine that Arduino IDE uses. Two main concerns are sketch size and execution speed. Seeing as I have a 256k Mega 2560, I was able to sacrifice space to gain performance. My sketch is 202,112 bytes big but I still have about 50k of free space. Vs it being about 175K ish...(I don't exactly remember). The article describes a simple test he did to prove his theory.

boelle, 2015-04-01:
it just wanted to put a label on it, but it take i can put "not a bug" on this one

roboprint, 2015-04-01:
Yes, it is bug (full graphic display refresh causing freeze during refresh on delta printers). But this can be 'fixed' by implementing 'screen saver' or 'change screen refresh rate' features.

Natealus, 2015-04-01:
Yeah I think thinkyhead has been working really hard on temp and bed leveling issues the last few days but I think him and a few people intend to test a few ways to lighten or circumvent the delay of the graphics lcd system.

thinkyhead, 2015-04-02:
@boelle I think we might consider this a "feature request." Namely, that the graphical display not interfere with Delta printing (especially), as both require extra computing resources. There are workarounds, such as reducing delta segments per second or using a faster board, but I'm sure we can shave cycles from all the display updates.

Natealus, 2015-04-02:
Could help all around but yeah I think this issue is perhaps a ...mid priority in comparison to more important issues. You've been making fixes daily on bed leveling Scott that's awesome and I'm glad you're working hard on it. :)

AnHardt, 2015-04-02:
Saved some cycles today in the dogm-code.
Preview on https://github.com/AnHardt/Marlin/tree/StatusSpeedDogm.
To much features for a PR now - i guess.

Natealus, 2015-04-02:
Badass I look forward to giving it a try :)

AnHardt, 2015-04-02:
Heading for the 'Number to string conversion's again.
There are lots if modulo divisions. Every % consists of a division a multiplication and a subtraction on the Atmels. I know there are algorithms to avoid this and save about 50% of the cycles.
So this time i'm looking for pure speed - not space.

alexborro, 2015-04-02:
Just for your information, the Repetier firmware runs Graphic LCD on deltas
with no hiccups at all, on the same hardware.
So yes, there is a way to do that.

Cheers.

Alex.

Natealus, 2015-04-02:
Yeah I would still classify it as a performance-hindering bug...we've just been focusing on getting leveling and temps and homing working as intended. I had to reduce delta segments per second from 165 to 38...just to enable the LCD. So there is some inefficiencies or inappropriate interrupting happening that's causing it to delay so much more than Repetier's Graphical LCD system. Yea these 8-bit sub 30mhz aren't that powerful so every bit of efficiency helps lol.

AnHardt, 2015-04-03:
@alexborro 
Had a look into Repetier today.
The loop for writing the buffer to the display is quite the same as in Marlin.
The graphics they copy onto the info-screen are much smaller than in Marlin. This is nearly ready for a PR.
They use a modulo free number to string conversion. That's what i'm working on.

alexborro, 2015-04-03:
Thanks @AnHardt ,

I was talking yesterday with @thinkyhead on skype about this issue. I guess we are updating all the GLCD buffer all the time and that makes no sense.. We could update just the pixels that have changed.

And getting rid of modulo operations is really good as well.

Cheers.

Alex;

Natealus, 2015-04-04:
:+1:  :+1:

AnHardt, 2015-04-04:
A bit a improvement is done. See #1817.
I guess that's factor 4 or better.

alexborro, 2015-04-04:
Humm, nice.. I think the optimal idea would be calling the lcd refresh
function from an interruption executed at about 1hz. There is no reason for
a faster update. And the buttons/scroller function must be called faster to
make the display responsive... That way the LCD will be refreshed faster
when navigating through menus, but at a lower rate when at info screen...
That will be awesome for Delta bots.

thinkyhead, 2015-04-04:
@alexborro I'm still very interested in slicing up the updates into smaller segments. No matter how often you do the routine, if the routine still takes a long time to do the update, then there will be more chance of stuttering in movement. Ideally, it would be nice to able to update 1/10th of the screen at 10Hz. The main cost may be in setup for doing the update, so pre-calculation may help here too. I kind of like the idea of splitting up the main display update routine, so it counts up from 0 to however many things it normally draws. On each call of the draw routine, only draw a single item, then update the counter. If it clears the screen before drawing each frame, we'd have to get rid of that…

alexborro, 2015-04-04:
@thinkyhead Something I'm concerned about is the buttons reading routine. It should have high priority in order to have a responsive LCD.

No matter when we do the LCD refresh, we should pull the buttons quite fast (in a separated routine from refresh one).. And as soon as a button is pressed, update the LCD immediately.

Bear in mind I know almost nothing about the LCD and buttons pulling routines - I don't know how it currently works.

Cheers.

Alex.

AnHardt, 2015-04-05:
The 10Hz is already there in 'lcd_status_update_delay'-loop.
Should not be a big problem to path it down to lcd_implementation_status_screen().
Partial updates there are no problem. Give me some days.

There is also a possibility to split the update from the buffer to the display, but this makes things really uncomfortable. But that's only writing 1K of data written by software SPI - compared to the divisions this is close to nothing.

Button - encoder updates will have the same speed as before.

AnHardt, 2015-04-05:
If we want the 10 Hz spitted update #1817 makes no sense. Or, maybe, only for a week.

thinkyhead, 2015-04-05:
It definitely bears some experimentation. :zap:

AnHardt, 2015-04-05:
#1823 gave more insight to the problem.
Will make the 10 Hz update a bit more complicated. We'll see.

alexborro, 2015-04-05:
Nice job @AnHardt !!

My main concern on speed is not during menus navigation.. In this case it should be refresh as soon as a button is pressed. Why not refresh all the buffer at once instead of splitting in two sections ? This 2-part refresh causes an ugly aspect to the LCD during navigation..

We should stop fast refresh during print and LCD on info screen... that is 99.99% of the time and causes a huge performance degradation on Delta bots.

Cheers.

Alex.

AnHardt, 2015-04-14:
Got the 10Hz precalculation of the strings to work.
Preview: https://github.com/AnHardt/Marlin/tree/viki
A bit untidy for now - but works nice.
Also the work to do during the update is greatly reduced by using 'stripes'.

thinkyhead, 2015-04-14:
@AnHardt You are officially the display guru. :bow:

AnHardt, 2015-04-28:
Here i have a single-liner to improve the performance of the viki displays:
Change https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/dogm_lcd_implementation.h#L128

to:
    U8GLIB_NHD_C12864_2X u8g(DOGLCD_CS, DOGLCD_A0);

This requires u8glib v1.13 or newer.
The new device will only need 4 loops to draw the display, instead of 8 loops before.

I'd relay like to know if that improves the number of segments you can print.

Natealus, 2015-07-11:
I've been out of the loop for the last month and a half but I have some free time from work now so I'll probably be running more testing having to do with the segments with and without display to verify. Also will be trying to document the delta auto bed leveling. :)

thinkyhead, 2015-07-11:
@Natealus When you feel like testing homing / leveling, give this branch a run and post the output:
https://github.com/thinkyhead/Marlin/tree/debug_G29

Natealus, 2015-07-15:
Well congrats you guys did figure out the bug where in between the probing points, it is moving straight and not moving down in a bowl fashion. Kudos there. :) 

Unfortunately now it seems like it is not taking the probe switch into action when it is triggered to Stop moving down and not crush my probe rig. I've tried adjusting between and before probing heights, manual z height, z-probe extruder Z offset (M851) and while the initial height changes...they all still go down too far and flex the probe rig before pulling back up to move to the next point.

I have checked to see if the probe switch is still working to the firmware and it is. Tested multiple times. I also noticed that Z-probe and Z-min register at the same time if I enable min endstops...and that Z-probe still works if I disable them too but the problem still seems to be the same. Here is a paste bin of a few probes I let it do before I thought I should stop before it busts.

http://pastebin.com/qQB12y5C

Natealus, 2015-07-15:
Scratch that after fiddling with it some more in different positions I think it is a wiring issue. Not sure why it chooses now to start malfunctioning but I'm trying to fix it. No success so far I need a brake from tedious wiring crap.

thinkyhead, 2015-07-16:
@Natealus Eek, you had me worried there, that maybe it was doing the fast probe, then trying to do a slow probe without lifting up first... or something like that. Please keep me posted as you fix the wiring.

Natealus, 2015-07-18:
@thinkyhead Alrighty good news. :) After a much more sturdy wiring setup on my probe soldered and heat shrunk, it is reliably working again. It is appearing to work correctly. I've noticed my temps are going a little wonky and triggering heating errors so I'm re-tuning the PIDs.

I've been testing trying to get appropriate nozzle height (at 100 microns). I've been adjusting the M851 z-probe offset setting via console but I'm not certain it is actually moving. When that happened before I found that the change in the firmware and reuploading it was working while the eeprom M851 wasn't actually working. I'm going to test more thoroughly to verify that.

But it looks much more stable and I'm able to put it up to 11 point grid instead of the previous 8 due to compile sanity check errors that were recently fixed. The post-bed leveling movement looks alot more straight but I have to get the nozzle to bed distance right first to be more specific.

The recent servo changes caused some issues getting that working right again. I had to change my AzteegX3Pro_pins.h for the servo from servo0 to servo2 then it was addressable again. Then had to tell the servo section in configuration.h to use servo2 instead. I'm still having an old issue of G29 not extending the z-probe servo and I have to reset the printer before the nozzle hits the bed. But if I use G29 E...it works just fine. Except it extends and retracts the probe for every point. Is that necessary for accuracy or it excessive? I wouldn't think it would move too much probe point to point...maybe it could reset the angle every row perhaps?

Natealus, 2015-07-18:
Ok after further testing the M851 does seem to be working.

Natealus, 2015-07-18:
Also another pretty old issue...and I'm not sure if it has come up recently or if it's unique to my setup...but for bed leveling to work right I have to // out the #define Z_RAISE_AFTER_PROBING 50. Or else the probing numbers come out to 100s or 1000s...

->Communication State: AttemptingToConnect

<-start
->Communication State: Connected

<-echo: External Reset
->M105

<-Marlin 1.0.3 dev
<-echo: Last Updated: 2015-06-00 12:00 | Author: (Nate Stokes)
<-Compiled: Jul 18 2015
<-echo: Free Memory: 2628  PlannerBufferBytes: 2464
<-echo:V20 stored settings retrieved (378 bytes)
<-echo:Steps per unit:
<-echo:  M92 X80.00 Y80.00 Z80.00 E184.80
<-echo:Maximum feedrates (mm/s):
<-echo:  M203 X300.00 Y300.00 Z300.00 E25.00
<-echo:Maximum Acceleration (mm/s2):
<-echo:  M201 X5000 Y5000 Z5000 E10000
<-echo:Accelerations: P=printing, R=retract and T=travel
<-echo:  M204 P2300.00 R3000.00 T2500.00
<-echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
<-echo:  M205 S0.00 T0.00 B20000 X17.00 Z17.00 E5.00
<-echo:Home offset (mm):
<-echo:  M206 X0.00 Y0.00 Z0.00
<-echo:Endstop adjustment (mm):
<-echo:  M666 X0.00 Y0.00 Z0.00
<-echo:Delta settings: L=delta_diagonal_rod, R=delta_radius, S=delta_segments_per_second
<-echo:  M665 L269.00 R130.15 S160.00
<-echo:PID settings:
<-echo:  M301 P15.13 I0.92 D62.47 C1.00
<-echo:  M304 P417.48 I31.23 D1395.31
<-echo:Filament settings: Disabled
<-echo:Z-Probe Offset (mm):
<-echo:  M851 Z-13.33
<-ok T:0.0 /0.0 B:0.0 /0.0 T0:0.0 /0.0 T1:0.0 /0.0 T2:0.0 /0.0 T3:0.0 /0.0 @:0 B@:0
->M115

<-FIRMWARE_NAME:Marlin 1.0.3 dev SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:Abacus 3D Kraken EXTRUDER_COUNT:4 UUID:00000000-0000-0000-0000-000000000000
<-ok
->M105

<-ok T:24.1 /0.0 B:77.0 /0.0 T0:24.1 /0.0 T1:23.7 /0.0 T2:23.7 /0.0 T3:23.8 /0.0 @:0 B@:0
->G28

<-ok
->G29 E

<-G29 Auto Bed Leveling
->M114

->M105

->M105

<--949.89 100.05 150.16 0.17 -149.82 -299.80 -449.79 -599.78 -550.10 0.00 
<--449.91 -299.90 250.13 200.20 150.16 100.12 50.06 -0.09 349.74 0.00 
<-999.78 300.03 350.08 400.22 450.13 500.05 549.92 599.87 649.76 0.00 
<-1149.65 1099.81 1049.97 1000.07 950.07 900.03 850.00 799.87 749.81 699.78 
<-1099.63 1149.73 1199.83 1249.93 1299.98 1349.90 1399.81 1449.63 1499.57 1549.45 
<-2049.07 1999.38 1949.69 1899.78 1849.84 1799.87 1749.81 1699.73 1649.73 1599.57 
<-2798.93 2049.21 2099.47 2149.61 2199.68 2249.66 2299.68 2349.71 2399.48 2449.33 
<-2898.14 2848.63 2799.12 2749.41 2699.59 2649.63 2599.78 2549.71 2499.61 0.00 
<-3597.79 3448.63 2898.78 2949.04 2999.27 3049.51 3099.62 3149.59 2799.73 0.00 
<-0.00 0.00 0.00 3348.69 3299.23 3249.57 3199.67 0.00 0.00 0.00 
<-ok
<-echo:endstops hit:  Z:3626.51 Z_PROBE:3626.51
<-X:-49.00 Y:132.00 Z:3412.02 E:0.00 Count X: 3533.69 Y:3447.89 Z:3626.52
<-ok
<-ok T:25.3 /0.0 B:69.5 /0.0 T0:25.3 /0.0 T1:24.9 /0.0 T2:25.0 /0.0 T3:24.3 /0.0 @:0 B@:0
<-ok T:25.3 /0.0 B:69.5 /0.0 T0:25.3 /0.0 T1:24.9 /0.0 T2:25.0 /0.0 T3:24.3 /0.0 @:0 B@:0
->M105

closed 2016-05-04 by jbrazio