MarlinFirmware/Marlin #1741: Compile time error [current dev as of this post]

avluis, 2015-03-30:
Hey guys!
Just merged dev branch and I am getting a compile time error if I have more than one extruder defined (so the default - single extruder - configuration compiles fine).

Here is the error log:

Arduino: 1.6.1 (Mac OS X), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build options changed, rebuilding all
Using library Wire in folder: /Users/avluis/Documents/Arduino/libraries/Wire 
Using library LiquidCrystal_I2C in folder: /Users/avluis/Documents/Arduino/libraries/LiquidCrystal_I2C (legacy)
Using library LiquidTWI2 in folder: /Users/avluis/Documents/Arduino/libraries/LiquidTWI2 (legacy)
Using library U8glib in folder: /Users/avluis/Documents/Arduino/libraries/U8glib (legacy)
Using library LiquidCrystal in folder: /Users/avluis/Documents/Arduino/libraries/LiquidCrystal (legacy)
Using library SPI in folder: /Users/avluis/Documents/Arduino/libraries/SPI (legacy)

/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega -I/Users/avluis/Documents/Arduino/libraries/Wire -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal_I2C -I/Users/avluis/Documents/Arduino/libraries/LiquidTWI2 -I/Users/avluis/Documents/Arduino/libraries/U8glib -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal -I/Users/avluis/Documents/Arduino/libraries/SPI /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/BlinkM.cpp -o /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/BlinkM.cpp.o 
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega -I/Users/avluis/Documents/Arduino/libraries/Wire -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal_I2C -I/Users/avluis/Documents/Arduino/libraries/LiquidTWI2 -I/Users/avluis/Documents/Arduino/libraries/U8glib -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal -I/Users/avluis/Documents/Arduino/libraries/SPI /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/cardreader.cpp -o /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/cardreader.cpp.o 
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega -I/Users/avluis/Documents/Arduino/libraries/Wire -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal_I2C -I/Users/avluis/Documents/Arduino/libraries/LiquidTWI2 -I/Users/avluis/Documents/Arduino/libraries/U8glib -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal -I/Users/avluis/Documents/Arduino/libraries/SPI /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/ConfigurationStore.cpp -o /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/ConfigurationStore.cpp.o 
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega -I/Users/avluis/Documents/Arduino/libraries/Wire -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal_I2C -I/Users/avluis/Documents/Arduino/libraries/LiquidTWI2 -I/Users/avluis/Documents/Arduino/libraries/U8glib -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal -I/Users/avluis/Documents/Arduino/libraries/SPI /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/digipot_mcp4451.cpp -o /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/digipot_mcp4451.cpp.o 
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega -I/Users/avluis/Documents/Arduino/libraries/Wire -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal_I2C -I/Users/avluis/Documents/Arduino/libraries/LiquidTWI2 -I/Users/avluis/Documents/Arduino/libraries/U8glib -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal -I/Users/avluis/Documents/Arduino/libraries/SPI /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Marlin.cpp -o /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Marlin.cpp.o 
/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega -I/Users/avluis/Documents/Arduino/libraries/Wire -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal_I2C -I/Users/avluis/Documents/Arduino/libraries/LiquidTWI2 -I/Users/avluis/Documents/Arduino/libraries/U8glib -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal -I/Users/avluis/Documents/Arduino/libraries/SPI /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Marlin_main.cpp -o /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Marlin_main.cpp.o 
In file included from /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Configuration_adv.h:501:0,
                 from /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Configuration.h:752,
                 from /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Marlin.h:22,
                 from /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Marlin_main.cpp:30:
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Conditionals.h:342:57: error: braces around scalar initializer for type 'float'
     #define ARRAY_BY_EXTRUDERS(v1, v2, v3, v4) { v1, v2 }
                                                         ^
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build460817318659504757.tmp/Marlin_main.cpp:244:60: note: in expansion of macro 'ARRAY_BY_EXTRUDERS'
   float extruder_offset[EXTRUDERS][NUM_EXTRUDER_OFFSETS] = ARRAY_BY_EXTRUDERS(_EXY, _EXY, _EXY, _EXY);
                                                            ^
Error compiling.

thinkyhead, 2015-03-30:
Check for the existence of `Conditionals.h`, the line `#include "Conditionals.h"` at the top of `Configuration_adv.h`, and these lines at the end of `Configuration_adv.h`:

    #include "Conditionals.h"
    #include "SanityCheck.h"

avluis, 2015-03-30:
Doing that gives me:

Arduino: 1.6.1 (Mac OS X), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at processing.app.Sketch.getCurrentCode(Sketch.java:1480)
	at processing.app.EditorHeader.paintComponent(EditorHeader.java:194)
	at javax.swing.JComponent.paint(JComponent.java:1037)
	at javax.swing.JComponent.paintChildren(JComponent.java:870)
	at javax.swing.JComponent.paint(JComponent.java:1046)
	at javax.swing.JComponent.paintChildren(JComponent.java:870)
	at javax.swing.JSplitPane.paintChildren(JSplitPane.java:1030)
	at javax.swing.JComponent.paint(JComponent.java:1046)
	at javax.swing.JComponent.paintChildren(JComponent.java:870)
	at javax.swing.JComponent.paint(JComponent.java:1046)
	at javax.swing.JComponent.paintChildren(JComponent.java:870)
	at javax.swing.JComponent.paint(JComponent.java:1046)
	at javax.swing.JComponent.paintChildren(JComponent.java:870)
	at javax.swing.JComponent.paint(JComponent.java:1046)
	at javax.swing.JComponent.paintChildren(JComponent.java:870)
	at javax.swing.JComponent.paint(JComponent.java:1046)
	at javax.swing.JLayeredPane.paint(JLayeredPane.java:567)
	at javax.swing.JComponent.paintChildren(JComponent.java:870)
	at javax.swing.JComponent.paint(JComponent.java:1046)
	at javax.swing.JComponent._paintImmediately(JComponent.java:5106)
	at javax.swing.JComponent.paintImmediately(JComponent.java:4890)
	at javax.swing.RepaintManager$3.run(RepaintManager.java:814)
	at javax.swing.RepaintManager$3.run(RepaintManager.java:802)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.AccessControlContext$1.doIntersectionPrivilege(AccessControlContext.java:86)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:802)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:745)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:725)
	at javax.swing.RepaintManager.access$1000(RepaintManager.java:46)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1680)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:209)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:715)
	at java.awt.EventQueue.access$400(EventQueue.java:82)
	at java.awt.EventQueue$2.run(EventQueue.java:676)
	at java.awt.EventQueue$2.run(EventQueue.java:674)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.AccessControlContext$1.doIntersectionPrivilege(AccessControlContext.java:86)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:685)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:296)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:211)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:196)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:188)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:122)
Using library Wire in folder: /Users/avluis/Documents/Arduino/libraries/Wire 
Using library LiquidCrystal_I2C in folder: /Users/avluis/Documents/Arduino/libraries/LiquidCrystal_I2C (legacy)
Using library LiquidTWI2 in folder: /Users/avluis/Documents/Arduino/libraries/LiquidTWI2 (legacy)
Using library U8glib in folder: /Users/avluis/Documents/Arduino/libraries/U8glib (legacy)
Using library LiquidCrystal in folder: /Users/avluis/Documents/Arduino/libraries/LiquidCrystal (legacy)
Using library SPI in folder: /Users/avluis/Documents/Arduino/libraries/SPI (legacy)

/Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants/mega -I/Users/avluis/Documents/Arduino/libraries/Wire -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal_I2C -I/Users/avluis/Documents/Arduino/libraries/LiquidTWI2 -I/Users/avluis/Documents/Arduino/libraries/U8glib -I/Users/avluis/Documents/Arduino/libraries/LiquidCrystal -I/Users/avluis/Documents/Arduino/libraries/SPI /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build8393301854793890405.tmp/BlinkM.cpp -o /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build8393301854793890405.tmp/BlinkM.cpp.o 
In file included from /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build8393301854793890405.tmp/Configuration_adv.h:5:0,
                 from /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build8393301854793890405.tmp/Configuration.h:752,
                 from /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build8393301854793890405.tmp/Marlin.h:22,
                 from /var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build8393301854793890405.tmp/BlinkM.cpp:5:
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build8393301854793890405.tmp/SanityCheck.h:109:10: error: #error "The given RIGHT_PROBE_BED_POSITION can't be reached by the probe."
         #error "The given RIGHT_PROBE_BED_POSITION can't be reached by the probe."
          ^
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build8393301854793890405.tmp/SanityCheck.h:247:8: error: #error HEATER_1_PIN not defined for this board
       #error HEATER_1_PIN not defined for this board
        ^
/var/folders/by/j72slykd6nlfsn8h922xbm2r0000gn/T/build8393301854793890405.tmp/SanityCheck.h:251:6: error: #error HEATER_0_PIN not defined for this board
     #error HEATER_0_PIN not defined for this board
      ^
Error compiling.

Changing RIGHT_PROBE_BED_POSITION has no effect.

Sniffle, 2015-03-30:
have you tried compiling under arduino 1.0.6?

avluis, 2015-03-31:
One sec - will try that.

thinkyhead, 2015-03-31:
@avluis I suggest you get a clean copy of the latest Development branch. You should find that the latest Development code will compile with the default configurations. Then, compare your `Configuration.h` and `Configuration_adv.h` files to the most recent configuration files, just to make sure all the settings are in the right places. If you have a diff tool that lets you compare side-by-side (you can use the Github app to do this) that will make it easier. If you have an editor that does diffs, even better. (I use _BBEdit_ to compare and to apply diffs.)

avluis, 2015-03-31:
Got it, will try that and report how it goes.

avluis, 2015-03-31:
@thinkyhead 
Found the problem, if I set my extruder offsets then it won't compile:

#define EXTRUDER_OFFSET_X {0.0, 20.00} // (in mm) for each extruder, offset of the hotend on the X axis
#define EXTRUDER_OFFSET_Y {0.0, 5.00}  // (in mm) for each extruder, offset of the hotend on the Y axis

thinkyhead, 2015-03-31:
#1753 should fix the issue.

avluis, 2015-04-03:
I'm not having any compile errors after the merge with https://github.com/MarlinFirmware/Marlin/pull/1753.
I tried that out with Arduino 1.0.6 and 1.6.0 and 1.6.2.

thinkyhead, 2015-04-04:
The fixed-up code ended up prettier than my original adjustments anyway…

closed 2015-04-04 by thinkyhead