MarlinFirmware/Marlin #1742: Endstop open message misleading

nophead, 2015-03-30:
M119 used to reply with H or L, which made sense to me. Now it is TRIGGERED or open. How is open the opposite of triggered? Most switches are closed when not triggered. Should be "not triggered" I think.

thinkyhead, 2015-03-31:
Do any hosts depend on this? Should be easy to fix...

ZetaPhoenix, 2015-03-31:
As the author of the patch, I made the change from H/L to open/TRIGGERED as
deciphering whether H or L was referring to the logic pin status, or the
logic value passed to the software.

"open" was chosen in reference to using optical sensors where nothing was
in the way and was shorter and not capitalized to make the different stated
easy to see at a glance while testing the switches.

"clear" might be a acceptable replacement if that makes more sense and no
hosts are coded to the output.

-Jon

clefranc, 2015-03-31:
open, clear, idle, untriggered, deactivated
TRIGGERED, ACTIVATED, ???

AnHardt, 2015-03-31:
true - false?
Who cares? The test is so fast and simple.
Finding M119 is more a problem than using it.
H - L are ok for me and in doubt will break nothing.

nophead, 2015-03-31:
I don't think any hosts care but the change probably breaks many commissioning instructions on the web.

AnHardt, 2015-03-31:
Or how about some ASCII-art?

       +Z0 0+Y
         I/
      *--+--0+X
        /I
       0 *

For triggered -X, +Y, no Z
(This is a Joke)

thinkyhead, 2015-04-01:
Hmm…. Perhaps `M119` should just output the triggered endstops, or "none" if there are no endstops triggered.

    > M119
    Endstop status: x_min y_min z_min

    > G1 X20 Y20 Z20
    > M119
    Endstop status: (none)

Ziggy2013, 2015-04-01:
Why not just a simple 

> M119
Endstop triggered : Yes (or No)

Very simple to understand and the type of endstop and whether it is reversed or not does not matter.

AnHardt, 2015-04-01:
@Ziggy2013 
I use M119 only during debugging.
The information witch endstop triggered is essential.

Wurstnase, 2015-04-01:
@thinkyhead I think that not a good idea. Also not configured endstopps never shown and this could be good for debugging also.

Ziggy2013, 2015-04-01:
@AnHardt Actually that is what I meant to write 

M119
Endstop X (or Y or Z) min (or max) triggered : Yes (or No)

I mod a forum where lots of users have problems with endstop hardware/connections. M119 is very useful for debugging.

thinkyhead, 2015-04-01:
I don't like YES / NO as part of user interface. `H` and `L` aren't very friendly either. `TRIGGERED` / `open` is actually not that bad. Readable, explicit, and distinctive. I agree the output should list all endstops (or if endstops are named, only the ones named).

nophead, 2015-04-01:
TRIGGERED / open is plain wrong when you have a normally closed switch. How
is the user supposed to guess it refers to an un-blocked opto?

TRIGGERED / clear is much better but I think it should also display H and L
for the physical state of the pin. It is then backwards compatible with
older instructions and also gives more information. I.e both the logical
and physical states making it obvious when the wrong polarity has been
configured or if the pin is actually in the wrong state.

thinkyhead, 2015-04-01:
@nophead I'm personally not confused by the use of the word "open" in this context, even though I also use NC switches. `M119` reports the _endstop state_ and should not report the _switch state_. Still, I'm open to other suggestions that don't include the words "open" and "closed."

Perhaps `M119` could use some options, so that one can request the output that is most useful for the current application. If you wanted to know whether the switch is physically open or closed (NC=not triggered, NO=triggered), you could specifically ask for that data. If you just wanted to know whether the endstop is currently being interpreted as "triggered" you could ask for that. A most-detailed report could list all the switches, include their type (NO or NC), switch state (NC=OPEN/closed, NO=open/CLOSED), and endstop state (TRIPPED / un-tripped)…

dantman, 2015-06-04:
M119's results are orthogonal to whether the endstop is a NO switch, NC switch, optical, or even hall. What's really being reported is (for each endstop) if Marlin thinks that an axis is at that endstop based on how it's configured.

So how about aiming for terminology that fits that. Like "hit" or "reached".

"hit" appears a lot in endstop related code, so how about:
```
x-min: hit
y-min: not hit
z-min: hit
...

Wackerbarth, 2015-06-05:
I agree. We should be able to handle this with macros.

thinkyhead, 2015-06-10:
Hosts may expect these strings in response to `M119` so we have to pass this by host developers before making any such change.

nophead, 2015-06-10:
Well its already changed from H and L to TRIGGERED or open without consulting anybody. That is why I opened this issue.

thinkyhead, 2015-06-12:
Alas, this change came long before my time… Such a change would not be made today without consulting the community. Perhaps it is worth looking back to see what kind of discussion happened around the time of the commit…

    8ac5b29e (ZetaPhoenix 2012-09-16 14:51:24 -0700 1231)         SERIAL_PROTOCOLLN(((READ(X_MIN_PIN)^X_ENDSTOPS_INVERTING)?MSG_ENDSTOP_HIT:MSG_ENDSTOP_OPEN));

closed 2016-04-25 by thinkyhead