MarlinFirmware/Marlin #1744: G29 V4 P5

Wurstnase, 2015-03-30:
Hi all,

I don't have a RAMPS installed at the moment. But on my Due when I try G29 V3 P5 or V4 the printer stops when reaching the 6nd point. Just a Marlin4Due-issue? G29 V2 P5 works and also G29 V4 P4.

Or any idea?

thinkyhead, 2015-03-31:
What kind of result do you see with `G29 V3 P5 D` ?

Wurstnase, 2015-03-31:
Same result.
The printer stops and I can't sent any new command. Also the LCD-menu looks like when printing.

thinkyhead, 2015-03-31:
@Wurstnase The `qr_solve` function is one of those rare routines (maybe the only routine) that uses `malloc` to allocate SRAM. What kind of board are you using, and how much Free RAM does the firmware report when you start it?

Wurstnase, 2015-03-31:
I will check the free RAM this evening. 
I'm using a Arduino Due:
SRAM 96 KB (two banks: 64KB and 32KB)

Roxy-3DPrintBoard, 2015-03-31:
> @Wurstnase The qr_solve function is one of those rare routines (maybe the only routine) that uses malloc to allocate SRAM. What kind of board are you using, and how much Free RAM does the firmware report when you start it?

If the root problem is a shortage of RAM for malloc() to grab and use...   I'm trying to get the iterative LSF uploaded today!  It doesn't use malloc() and hardly grabs any memory on the stack.

Wurstnase, 2015-03-31:
I don't think, that memory is the problem here.
```19:46:37.590 : echo: Free Memory: 87476  PlannerBufferBytes: 2688```

Wurstnase, 2015-03-31:
Just update to 1.6.2 (1.6.3 for Due) and that part is gone. Thanks anyway!

closed 2015-03-31 by Wurstnase