MarlinFirmware/Marlin #1751: Change in behavior for [XYZ]_[MIN/MAX]_POS from release 1.0.2 to dev branch

croadfeldt, 2015-03-31:
in Release 1.0.2, setting X_MIN_POS to -30 would offset 0 to the correct location for my bed. I have a Wilson TS Reprap which has the X home off bed, Y is close as well. Setting the various X and Y MIN_POS to negative values put 0,0 right on the corner of the bed, which made slicing easy and other configs easy since I didn't have to account for the offset.

It looks as though the current dev branch does not do this right now.

Based on the documentation in the config header, I'm not sure I was using the correct config item to accomplish the bed offset, but it worked so I used it.

My questions:
1. Is the X_MIN_POS, for example, correctly used for a bed offset?
2. If so, I will update the config documentation so it's more clear that it can be used that way.
3. If not, is there another mechanism to use so 0,0 can be offset from X,Y and even Z home?
4. Is that the correct use of 0,0? Is it suppose to be at the corner for cartesian setups?

thinkyhead, 2015-03-31:
@croadfeldt The settings `X_MIN_POS` and `X_MAX_POS` tell the firmware the farthest points that the robot can physically move in the cartesian space. The robot won't move past a min X endstop, so if you have a min endstop your `X_MIN_POS` should be 0. But if the machine has a max X endstop, then the X carriage could conceivable move below zero. This can be useful for a machine with a bed probe to the right of the nozzle.

Where your (0, 0) is depends on your endstop settings. If endstops are at left and in the back, then your (0, 0) will be where your endstops are. If you have any max endstops that are not at (0, 0) you can set the manual endstop positions to be at some other position, and the firmware will then know where (0, 0) is after homing (`G28`).

There is also a concept of home_offset which the firmware can add to the current position (`M206: Set Additional Homing Offset (X Y Z)`) but you should leave that at (0, 0, 0) unless experimenting.

croadfeldt, 2015-03-31:
@thinkyhead Thanks, appreciate it. If you don't mind I will poke around the docs and see if that can be cleaned up a bit to make it more clear then, if it's needed.

closed 2015-03-31 by croadfeldt