MarlinFirmware/Marlin #1759: Compiler error when no sdsupport

Wurstnase, 2015-03-31:
Hi,

I got an error when SDSUPPORT is not defined.

```
\stepper.cpp: In function 'void TC3_Handler()':
\stepper.cpp:327:44: error: 'SD_FINISHED_STEPPERRELEASE' was not declared in this scope

     if ((cleaning_buffer_counter == 1) && (SD_FINISHED_STEPPERRELEASE)) enquecommands_P(PSTR(SD_FINISHED_RELEASECOMMAND));

                                            ^

In file included from \Marlin.h:14:0,

                 from \stepper.cpp:24:

\stepper.cpp:327:94: error: 'SD_FINISHED_RELEASECOMMAND' was not declared in this scope

     if ((cleaning_buffer_counter == 1) && (SD_FINISHED_STEPPERRELEASE)) enquecommands_P(PSTR(SD_FINISHED_RELEASECOMMAND));

                                                                                              ^

C:\Program Files (x86)\Arduino\hardware\arduino\sam\cores\arduino/avr/pgmspace.h:8:20: note: in definition of macro 'PSTR'

 #define PSTR(str) (str)

                    ^

Error compiling.
```

AnHardt, 2015-03-31:
@Wurstnase 
Can't reproduce this.
What version are you on? 
We have a different line-number currently. https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/stepper.cpp#L392

Wurstnase, 2015-03-31:
Sure, this is my Due-Fork, but most parts should be the same.
Did you comment out your display?

AnHardt, 2015-03-31:
@Wurstnase 
You have a PR: https://github.com/Wurstnase/Marlin4Due/pull/19
Please close this.

closed 2015-03-31 by Wurstnase

AnHardt, 2015-03-31:
Ah. Scot fixed that this morning.https://github.com/MarlinFirmware/Marlin/commit/3348b3bf0c3870a013422371281fb74650f83ff8