MarlinFirmware/Marlin #1760: G29 must require homing of all axes

gaussgss, 2015-03-31:
When Z axis isn't homed and bed auto-leveling requested, G29 produces wrong results.

thinkyhead, 2015-04-01:
@gaussgss Currently the `G29` command doesn't care if Z is at a known position, just that X and Y have been homed. It probably should prevent using `G29` unless Z has also been homed (especially on deltas).

Ziggy2013, 2015-04-01:
It's not clear to me why Z must be homed before a G29.  The G29 itself needs to establish where the Z home is a first step before probing. And the point the G29 uses to establish Z Home could be different depending on the type of probing (eg 3 point, 9 point etc)

thinkyhead, 2015-04-01:
@Ziggy2013 Well, that 's what `G29` is failing to do – establishing a known Z position before probing. If it doesn't get it from a `G28` homing command, then it must establish it when first starting to probe the bed, and I'm not sure it does that.

Wurstnase, 2015-04-01:
I don't think that's a good idea to implement the home z in the g29. On Repetier this makes also some trouble. For the transformation you need a fix point from where you can calculate it. If you don't home z and make a g29, the probe will not know where it is.

You can't measure the length of your living room from everywhere. You will go to one wall/edge first.

nophead, 2015-04-01:
I home to the top and use my probe to find where the bed is, not as a
bottom end stop. I think Marlin is broken because it treats the probe as a
bottom end stop, so whenever it hits it resets the Z home point. At some
point this needs to be fixed so it uses the height information from the
plane equation, which is more accurate.

If you don't a bottom endstop then the probe needs to act as one in G28 but
not G29. I feel yet another horrible global flag being born.

Ziggy2013, 2015-04-01:
IMHO the G29 should first establish a start "reference" Z = 0 point using the probe.  Where that reference point is on the X/Y plane needs to be set based on the type of probing to be done.  eg a 3 point probe may require a different start Z reference point to a 9 point probe.

If G28 is also going to set that Z "reference" point as well, then the same X/Y reference point as used by the G29 probing needs to be used.  

Need to also keep in mind that not all auto leveling implementations use a separate Z probe.  For example, some implementations use the extruder tip itself to find the Z reference point.  Others use a dockable Z probe sled. In these implementations should G28 always try to find the Z reference point using the extruder or the sled?  

.

thinkyhead, 2015-04-02:
We have a PR pending that establishes a separate `Z_PROBE_PIN` so that the probe can be treated differently from an endstop. I suspected that the logic was being screwed up by the resetting of the Z position every time the probe switch is triggered. I will continue to document and do a full write-up of the program flow during bed leveling, and this should help to reveal where changes need to be made so that probes and endstops can get along together. Note that in addition to probes that use the nozzle as the probe, there are some that use an NC switch with a floating X gantry, and these could conceivably re-use the Z endstop by applying the right (opposite) logic. (e.g., if the Z axis goes "below" the z endstop it breaks the circuit. Only works with an NO switch as the Z min endstop.)

ProjectEnder, 2015-04-02:
I know with the Delta you need to G28 sending G29. i would have to double check but i dont think if DELTA is defined that the received G29 runs a G28 before. this might be a good thing to add for the DELTA mode as for those you always need to home the unit first.

TechMasterJoe, 2015-04-03:
I honestly think that g29 needs to implement a integrated ver of safe home z before starting where it touches off the bed center and stores that info as a reference just for auto bed leveling. As a plain z home can be done anywhere with the side effect of over or under comp with auto bed leveling it's like tossing in a random value for i in a PID never made sense to me honestly think we need a fixed optical endstop that's on the z axis independent of the bed as a hard reference that we set to a sheet of paper off the center of the bed I know this sound like more work but hey with a hard set home the topical bed map can even tell you direction and how many turns each bed screw is off by making bed leveling a breeze

thinkyhead, 2015-04-04:
I think that if you issue a `G29` and the axes haven't been homed, it should just call `gcode_G28` directly instead of issuing a message that you need to home first. I think that would be a reasonable _expected behavior_ for this use case.

TechMasterJoe, 2015-04-04:
agreed

Ziggy2013, 2015-04-06:
Please keep in mind that the M206 and G92 gcode functions also need to be handled if G28's are proposed to be done implicitly with a G29.

ProjectEnder, 2015-04-10:
I was just looking over the code, to see if there would be a conflict with the M206 and G92 by Adding a G28 to the G29 command. As far as i can tell (still trying to study the code enough to know where everything is)  M206 and G92 do not call, there should be no problem adding a command to call G28 before running running G29.

thinkyhead, 2015-04-11:
I think the issue is, if you do a `G29` expecting values that you intentionally set with a `G92` (or if offsets were changed using `M206`) then a `G28` could mess that up. Sometimes `G92` can do the job of a `G28` (on such machines as the QuadRap which have no end-stops).

Ziggy2013, 2015-04-11:
More generally the issue is the G29 needs known X = 0 and Y =0 reference
points to be able to probe the bed and calculate the bed plane using the
measured Z probing points. The G29 also needs to measure/calculate an
assumed Z = 0 reference point.  The known X = 0, Y = 0, Z = 0 reference
points then need to be the same during the print otherwise the calculated
plane will not be usable when adjusting the Z to match the bed plane.

G28 homing or G92 / M206 offset adjustments should all be ok PRIOR to the
G29 being completed but not after.

boelle, 2015-08-03:
This issue was moved to MarlinFirmware/MarlinDev#20

closed 2015-08-03 by boelle