MarlinFirmware/Marlin #1761: Wrong Z homing position.

fotis400, 2015-03-31:
I just finished some modifications to my RepRap, and also pulled latest Marlin.

No matter what are the settings in the config files, when my Z axis homes the printer reports Z 15mm.

I really can't get it what causes the 15mm offset. It should report 200mm (Z_MAX_POS), and because of this the software endpoints do not work. (I can send G01 Z20 which will cause crashing of the Z axis).

It seems like a bug to me...

Here is my Configuration.h
http://pastebin.com/E2g5gVn8

Wurstnase, 2015-03-31:
Deactivate your EEPROM 
*or*
read your config-file with ```M502``` and save it again with ```M500``` to EEPROM.

fotis400, 2015-03-31:
I had already did a M502. I also just tried to deactivated it.

The problem persists.

fotis400, 2015-03-31:
I also tried to set up manual home positions.

Again they work for all axes, apart from Z which remains at 15mm

thinkyhead, 2015-04-01:
@fotis400 Your setup with Max X and Max Z endstops is unusual, so probably why we haven't seen this so far. I have a pending PR #1764 which fixes some issues with bed leveling, so try that or wait till it's merged. The only place that I see a "15mm" figure is in your settings for `Z_RAISE_BEFORE_PROBING` and `Z_RAISE_AFTER_PROBING` but I don't see how those could be involved.

fotis400, 2015-04-01:
I just tested it, and in fact the offset is always equal to `Z_RAISE_AFTER_PROBING`. For some reason this part is messed up. (I am going to MAX endstop, so the servo probe is irrelevant).

I also pulled the latest commit (which should include the PR stated above), but now the problem is different.
Instead of reporting 200mm after G28, it reports 215mm! It seems also wrong to me.

fotis400, 2015-04-01:
Very strange behavior.... I send a G28.

The Z homes. Then it lowers some mm for no reason. Then the X homes. Again the Z lowers for no reason some more mm. And finally the Y homes.
Sending a M114 after that, still reports Z 214mm

boelle, 2015-04-01:
i looked at the config and could not see what your Z lenght are

fotis400, 2015-04-01:
line 381
`#define Z_MAX_POS 200`

fotis400, 2015-04-01:
@thinkyhead what is unusual with my Max X and Max Z endstops?

thinkyhead, 2015-04-02:
The problem, I believe, is that there is assumption in the code that if you are using a Z probe, then you must also be using the probe to home Z. That will need to be corrected.

Natealus, 2015-04-02:
If that is the right path...I might have a small thing to add...every time I home my machine x and y go up and right before z goes to complete its first homing bump it extends the Z-probe. Bumps the second time then retracts it as if it thinks its necessary in this setup. I'm aware with others that it is but maybe thats part of the problem is that this function is being asked to do 2 jobs? Servo Endstoping and Z-probing...two somewhat different functions aren't they? In order to get it to work correctly I had to enable 
#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command
and
#define SERVO_ENDSTOPS {-1, -1, 2} // Servo index for X, Y, Z. Disable with -1
#define SERVO_ENDSTOP_ANGLES {0,0, 0,0, 0,75} // X,Y,Z Axis Extend and Retract angles

I'm not sure if this is helpful or irrelevant but hopefully its of some use to you. :)

Natealus, 2015-04-02:
By the way I didn't realize but I documented my findings on This issue in issue #1769

boelle, 2015-04-04:
Then Mendel90 build could also be at stake here... they use a Z max only.... they gome by going up and when at the switch they know how far down the bed is. 

Its a VERY stable way since it will not push a Z min switch down

davbardi, 2015-04-28:
I'm attempting to auto bed level with a servo mounted z probe on a home made COREXY printer (which still needs some tuning) and I'm getting the issue where the printer head is starting too high. I've noticed the more out of level the bed the higher the head error above the bed (I have glass plates and can create make the bed very unlevel by putting a pencil under one side).  Auto levelling appears to be fundamentally working (apart from the level error) as I can clearly see the X-axis change as the head moves (very obvious with the pencil!). Another point I noticed is if PROBE_SERVO_DEACTIVATION_DELAY 300  is defined then (quite unexpectedly) the error gets much larger.  The hotend is also lifting a lot more between each bed probe during the G29.  Here the verbose responses from G29 V4 for the "pencil-lifted" bed.
My probe offset is -2.16mm

This is with PROBE_SERVO_DEACTIVATION_DELAY  not defined (giving a head height approx. 5.2mm too high):-

20:15:46.856 : +-----------+
20:15:46.856 : |...Back....|
20:15:46.856 : |Left..Right|
20:15:46.856 : |...Front...|
20:15:46.856 : +-----------+
20:15:46.856 : +2.90938 +3.25938
20:15:46.872 : -3.17812 -2.99062
20:15:46.872 : planeNormal x: -0.002337 y: -0.041125 z: 1.000000
20:15:46.872 : 
20:15:46.872 : Bed Level Correction Matrix:
20:15:46.872 : +0.999997 +0.000000 +0.002337
20:15:46.872 : -0.000096 +0.999155 +0.041090
20:15:46.872 : -0.002335 -0.041090 +0.999153

This is with PROBE_SERVO_DEACTIVATION_DELAY  defined (giving a head height approx. 11.4mm too high):-
20:10:46.525 : +-----------+
20:10:46.525 : |...Back....|
20:10:46.525 : |Left..Right|
20:10:46.525 : |...Front...|
20:10:46.525 : +-----------+
20:10:46.541 : +2.91250 +3.06250
20:10:46.541 : -3.05000 -2.92500
20:10:46.541 : planeNormal x: -0.001196 y: -0.039833 z: 1.000000
20:10:46.541 : 
20:10:46.541 : Bed Level Correction Matrix:
20:10:46.541 : +0.999999 +0.000000 +0.001196
20:10:46.541 : -0.000048 +0.999208 +0.039802
20:10:46.541 : -0.001195 -0.039802 +0.999207

I'm no coder, but hopefully these observations help towards finding the error.

thinkyhead, 2015-04-30:
@davbardi Thanks for the report. That is some strange behavior, which I hope will be easy to track down.  On my first reading of the code, it appears that if you have `PROBE_SERVO_DEACTIVATION_DELAY` set to 0 then it doesn't deploy a probe at all. However, I also see that in some cases it might call deploy or stow the z probe even though that value is 0….

I wonder if you could help out by using this version of Marlin which prints a very detailed log of homing and leveling: https://github.com/thinkyhead/Marlin/tree/debug_leveling

I'm constantly adding more logging to that branch and keeping it up to date with changes, so what I hope to catch in your case is whether or not it is doing more than one Z raise on any step, or whether it's doing more than one deploy or stow action – basically to track down the line where it is getting confused.

davbardi, 2015-04-30:
@thinkyhead.  I had repeated this over at issue 1769 (which you are working on) as well.  Following through your recent discussions, I've added in the  sync_plan_position();  as mentioned by @lrpirlet (and believe you added). This does stop this strange behaviour.  I still get the printer coming up too high and this offset is larger if the bed unevenness is larger (but at least having  PROBE_SERVO_DEACTIVATION_DELAY set to 300 or disabled has no effect on the problem).  I'll still upload your branch and run some further reports.

davbardi, 2015-04-30:
@thinkyhead. I'm setting up using your https://github.com/thinkyhead/Marlin/tree/debug_leveling.
I've did have a problem compiling. In Marlin_main.cpp I had to remark out 4665 to 4672 (seems to be a repeat of the previous block of code.  I also changed line 1512  to "if (probe_action == ProbeStay) {"   - I'm guessing it should be "probe_action" not "retract_action" (retract_action isn't defined).  Let me know if this assumption is wrong.

Once this was compiling, I ran four tests.  Two (one with PROBE_SERVO_DEACTIVATION_DELAY set to 300, and one commented out) with the pencil uinder plate edge to force about 6mm bed out of level, and then repeated with the pencil removed (ie bed is nearly level).  In each case I still get the head above bed on start (about 5.5mm too high with pencil and 0.5 mm without). 
Having the PROBE_SERVO_DEACTIVATION_DELAY set to 300 made no perceptible difference to the above-bed error. However having PROBE_SERVO_DEACTIVATION_DELAY set to 300 does effect the G28 command - it cleanly raises the probe about 8-10mm before swinging the probe arm off vertical.  When PROBE_SERVO_DEACTIVATION_DELAY set to 300 is commented out it does not appear to raise the probe before swinging it off vertical (slightly drags it across the bed). 
I also see variation in the G29 process - on some runs the lift between probes is much higher than others.  Also sometimes it probes the four corners in a mirrored "C" pattern, sometimes in a mirrored "Z".  I’m not yet clear on the pattern in this.
Here are the results for 
G28
G29 V4
for all four tests;

// #define PROBE_SERVO_DEACTIVATION_DELAY 300
Bed has pencil along back of plate - giving about making it about 6mm higher at back
 
12:40:17.155 : gcode_G28 >>>
12:40:17.155 : >>> home_axis(0
12:40:17.539 : > TRIGGER ENDSTOP > current_position: (0.00, 0.00, 10.00)
12:40:17.539 : > AFTER sync_plan_position > SET current_position: (0.00, 0.00, 10.00)
12:40:17.543 : axis_is_at_home 0 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 10.00)
12:40:17.543 : <<< home_axis(0
12:40:17.547 : > homeX: (0.00, 0.00, 10.00)
12:40:17.547 : >>> home_axis(1
12:40:18.550 : > TRIGGER ENDSTOP > current_position: (0.00, 0.00, 10.00)
12:40:18.550 : > AFTER sync_plan_position > SET current_position: (0.00, 0.00, 10.00)
12:40:18.550 : axis_is_at_home 1 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 10.00)
12:40:18.550 : <<< home_axis(1
12:40:18.550 : > homeY: (0.00, 0.00, 10.00)
12:40:18.550 : > Z_SAFE_HOMING >>>
12:40:18.566 : Raise Z (before homing) by 4.00
12:40:18.566 : > home_all_axis > current_position: (0.00, 0.00, 0.00)
12:40:18.566 : > home_all_axis > destination: (33.00, 124.00, 4.00)
12:40:19.581 : >>> home_axis(2
12:40:27.173 : > TRIGGER ENDSTOP > current_position: (33.00, 124.00, 0.00)
12:40:27.173 : > AFTER sync_plan_position > SET current_position: (33.00, 124.00, 0.00)
12:40:27.173 : axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (33.00, 124.00, 2.16)
12:40:27.173 : > SERVO_ENDSTOPS > Stow with write()
12:40:27.173 : <<< home_axis(2
12:40:27.173 : <<< Z_SAFE_HOMING
12:40:27.188 : > (home_all_axis || homeZ) > final: (33.00, 124.00, 2.16)
12:40:27.188 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
12:40:27.188 : <<< gcode_G28
12:40:27.188 : ok
12:41:03.229 : gcode_G29 >>>
12:41:03.230 : G29 Auto Bed Leveling
12:41:03.230 : enable_endstops(true)
12:41:03.230 : z_before = (before) 15.00
12:41:03.230 : probe_pt >>>
12:41:03.230 : > ProbeAction:1
12:41:03.236 : > current_position: (33.00, 124.00, 2.16)
12:41:03.236 : Z Raise to z_before 15.00
12:41:03.241 : > do_blocking_move_to: (33.00, 124.00, 15.00)
12:41:03.241 : do_blocking_move_to: (33.00, 124.00, 15.00)
12:41:06.469 : > do_blocking_move_to XY: (8.00, 29.00, 15.00)
12:41:06.470 : do_blocking_move_to: (8.00, 29.00, 15.00)
12:41:07.244 : > ProbeDeploy
12:41:07.244 : deploy_z_probe > current_position: (8.00, 29.00, 15.00)
12:41:13.880 : run_z_probe > current_position: (8.00, 29.00, -1.59)
12:41:13.880 : <<< probe_pt
12:41:13.886 : z_before = (between) 5.00
12:41:13.886 : probe_pt >>>
12:41:13.886 : > ProbeAction:0
12:41:13.887 : > current_position: (8.00, 29.00, -1.59)
12:41:13.893 : Z Raise to z_before 3.41
12:41:13.893 : > do_blocking_move_to: (8.00, 29.00, 3.41)
12:41:13.894 : do_blocking_move_to: (8.00, 29.00, 3.41)
12:41:15.153 : > do_blocking_move_to XY: (123.00, 29.00, 3.41)
12:41:15.154 : do_blocking_move_to: (123.00, 29.00, 3.41)
12:41:19.756 : run_z_probe > current_position: (123.00, 29.00, -1.46)
12:41:19.756 : Raise Z (between) by 5.00
12:41:19.756 : > ProbeStay > do_blocking_move_to: (123.00, 29.00, 3.54)
[... 388 lines cut ...]
13:08:05.452 : <<< probe_pt
13:08:05.452 : > probing complete > current_position: (123.00, 179.00, 16.70)
13:08:05.452 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
13:08:05.468 : Eqn coefficients: a: -0.00168478 b: -0.00295833 d: 2.51364970
13:08:05.468 : Mean of sampled points: 2.00937509
13:08:05.468 : 
13:08:05.468 : Bed Height Topography:
13:08:05.468 : +-----------+
13:08:05.468 : |...Back....|
13:08:05.468 : |Left..Right|
13:08:05.468 : |...Front...|
13:08:05.468 : +-----------+
13:08:05.468 : -0.13438 -0.30938
13:08:05.468 : +0.32813 +0.11562
13:08:05.468 : planeNormal x: 0.001685 y: 0.002958 z: 1.000000
13:08:05.468 : set_bed_level_equation_lsq > current_position: (122.97, 178.95, 17.44)
13:08:05.468 : 
13:08:05.484 : Bed Level Correction Matrix:
13:08:05.484 : +0.999999 +0.000000 -0.001685
13:08:05.484 : -0.000005 +0.999996 -0.002958
13:08:05.484 : +0.001685 +0.002958 +0.999994
13:08:05.484 : <<< gcode_G29
13:08:05.484 : ok

Hopefully this is of some help...

thinkyhead, 2015-04-30:
Thanks for the code fixes, @davbardi, I should not code without sleep. I will go over the logs and see what it all means. Meanwhile, you should always leave `PROBE_SERVO_DEACTIVATION_DELAY` enabled, otherwise the code thinks there's no servo probe.

lrpirlet, 2015-04-30:
​@davbardi <https://github.com/davbardi>, I had the same feeling until I
had a look at my display... at the end of G28, the z value is NOT zero...
So you need to issue G1 Z0 before printing...
The G29 will start and end up at the same Z value...
This is where I put a thick cardboard on my bed so that I could stop the
printer if it tries to punch the cardboard...

thinkyhead, 2015-04-30:
@lrpirlet @davbardi Ok, I'm scouring the code with a fine-tooth comb to see where this thing is losing its mind. One thing I notice is that there are too many conditions related to servos, endstops, and probing, which makes it hard to make sense of their meaning sometimes. Here are the configuration settings and what they affect in `Marlin_main.cpp`:
```
SERVO_ENDSTOPS: Array specifying which axis endstops are servos

#define SERVO_LEVELING (defined(ENABLE_AUTO_BED_LEVELING) && PROBE_SERVO_DEACTIVATION_DELAY > 0)
SERVO_LEVELING = ENABLE_AUTO_BED_LEVELING && PROBE_SERVO_DEACTIVATION_DELAY

ENABLE_AUTO_BED_LEVELING && !Z_PROBE_SLED && (SERVO_ENDSTOPS || Z_PROBE_ALLEN_KEY)
gcode_M401, gcode_M402: Enable deploy and stow commands

ENABLE_AUTO_BED_LEVELING && !Z_PROBE_SLED && SERVO_ENDSTOPS
gcode_M401, gcode_M402: Raise the axis before action

Z_PROBE_SLED
setup: Turn off SERVO0_PIN
dock_sled: Enable
gcode_G29: Dock and un-dock the sled
gcode_G31, gcode_G32: Enable

ENABLE_AUTO_BED_LEVELING && !Z_PROBE_SLED
gcode_G30: Enable

ENABLE_AUTO_BED_LEVELING && Z_HOME_DIR < 0
axis_is_at_home(Z_AXIS): Add zprobe_zoffset to current_position[Z_AXIS]

SERVO_ENDSTOPS
servo_init: Set servos to the raised position
homeaxis: Do servo.write at the end
deploy_z_probe: Engage with servo.write
stow_z_probe: Raise Z axis, stow with servo.write

SERVO_ENDSTOPS && !Z_PROBE_SLED
homeaxis: Engage with deploy_z_probe (SERVO_LEVELING + Z) or servo.write at start

SERVO_ENDSTOPS && !Z_PROBE_SLED && SERVO_LEVELING
homeaxis(Z_AXIS): Call deploy_z_probe (instead of servo.write) for Z

SERVO_LEVELING
servo_init: Call servo.detach for the Z axis
homeaxis(Z_AXIS): Call deploy_z_probe at start (instead of servo.write)
gcode_M280: Call servo.attach(0)/detach (in addition to servo.write)

SERVO_LEVELING && !Z_PROBE_SLED
homeaxis(Z_AXIS): Call stow_z_probe at the end
axis_is_at_home(Z_AXIS): Set the current Z position to -zprobe_zoffset

SERVO_ENDSTOPS && SERVO_LEVELING
deploy_z_probe: Call servo.attach(0)/detach (in addition to servo.write)
stow_z_probe: Call servo.attach(0)/detach (in addition to servo.write)

Z_PROBE_ALLEN_KEY && !Z_PROBE_SLED
gcode_G29: Call deploy/stow_z_probe at start/end

Z_PROBE_ALLEN_KEY && !SERVO_ENDSTOPS
stow_z_probe: Stow the allen-key probe

!Z_PROBE_SLED && !Z_PROBE_ALLEN_KEY
probe_pt: Allow deploy/stow_z_probe as specified
```

thinkyhead, 2015-05-01:
I could use some consensus on the meaning of the `Z_RAISE_BEFORE` and `Z_RAISE_AFTER` values. Should they specify the absolute Z position or should they specify a relative position? For example, I notice in @davbardi's first `G29` log, the Z moves up to 15 from 2.16. But perhaps it should raise the axis by 15mm up to 17.16 instead. For the `*_BETWEEN_*` value it makes sense to use a relative value. The `BEFORE` and `AFTER` values are intended to provide clearance for the servo to rotate, and `Z_RAISE_BEFORE_HOMING` is also documented as being for probe clearance, but just for homing, not probing.

davbardi, 2015-05-01:
From my observations the height above the bed the hotend starts at is directly related to how un-level the bed is (with approximately 6mm pencil test it was about 5.5mm, with a bed close to level it was about 0.5mm).   The more uneven the bed the worse the problem.  I suspect if the bed is perfectly level then the problem would not be noticed....
I may be wrong, but this suggests that the problem is somehow related to how the calculated bed level error is reapplied back to Z.  Although application of pre-set lift distances (Raise before and afters etc) may still be causing a problem, the error (if any) caused by these fixed settings should be the same no matter what the bed un-levelness is (now I'm making up words..).

Roxy-3DPrintBoard, 2015-05-01:
> I could use some consensus on the meaning of the Z_RAISE_BEFORE_PROBING and Z_RAISE_AFTER_PROBING values. 

When those values were originally put into the code the referred to the 'current position' of the  nozzle.      Raise before didn't care where the nozzle was, it was going to raise it that much before starting.   That handled what ever condition the printer was in to make sure the probe had room to engage.

In all of my coding I treated Raise after probing as if the probe was pressed against the bed and triggered.  This is how much was necessary to raise the Z before retracting or moving the nozzle.  That worked and did not cause problems.

thinkyhead, 2015-05-01:
@davbardi There's a set of lines in both of the `set_bed_level_equation_*` functions that apply leveling compensation to `current_position`. This includes the Z axis.

    current_position[Z_AXIS] = corrected_position.z;

This line was getting negated in earlier code by the added line:

    current_position[Z_AXIS] = -zprobe_zoffset;

That line was removed due to the consensus of some Marlin developers, but it may have been the right thing at one time. I suggest adding that line at the end of both `set_bed_level_equation_3pts` and set_bed_level_equation_lsq` and see what that does. But also try leaving that line out, and simply commenting out the line `current_position[Z_AXIS] = corrected_position.z` to see the difference.

davbardi, 2015-05-01:
@thinkyhead,    remarking out the  current_position[Z_AXIS] = corrected_position.z  seems be a lot better (possibly fixed) .  I tested with the approx. 6mm bed unlevel and it appears to be starting correct.  Tried with bed approx. level and even tried it with 20mm unlevel (had to increase the raise before and afters).  The probe did seem to hit the bed (magazine)slightly more as the bed unlevel was higher.  It could be that I've got a magazine on the bed to protect it, so it is a bit inaccurate.  I'll run some more tests.
Here are the results with the two changes you suggested.
Running gcode
M851
G28
G29 V4
then a test print

------------------------------------------------------------------------------------------
---------------------------Test 1---------------------------------------------------------
Bed very unlevel (pencil under back)
current_position[Z_AXIS] = -zprobe_zoffset; added to
set_bed_level_equation_lsq
set_bed_level_equation_3pts

        current_position[X_AXIS] = corrected_position.x;
        current_position[Y_AXIS] = corrected_position.y;
        current_position[Z_AXIS] = corrected_position.z; 
        current_position[Z_AXIS] = -zprobe_zoffset;

Results:
1. The hotend  now starts about 20.5mm too high (was 5.5mm before this change).

08:11:12.622 : echo:Z Offset :
08:11:12.626 : -2.16
08:11:12.626 : ok
08:11:18.095 : gcode_G28 >>>
08:11:18.099 : >>> home_axis(0
08:11:20.885 : > TRIGGER ENDSTOP > current_position: (0.00, 0.00, 0.00)
08:11:20.885 : > AFTER sync_plan_position > SET current_position: (0.00, 0.00, 0.00)
08:11:20.889 : axis_is_at_home 0 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 0.00)
08:11:20.889 : <<< home_axis(0
08:11:20.892 : > homeX: (0.00, 0.00, 0.00)
08:11:20.892 : >>> home_axis(1
08:11:24.805 : > TRIGGER ENDSTOP > current_position: (0.00, 0.00, 0.00)
08:11:24.805 : > AFTER sync_plan_position > SET current_position: (0.00, 0.00, 0.00)
08:11:24.809 : axis_is_at_home 1 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 0.00)
08:11:24.809 : <<< home_axis(1
08:11:24.813 : > homeY: (0.00, 0.00, 0.00)
08:11:24.813 : > Z_SAFE_HOMING >>>
08:11:24.813 : Raise Z (before homing) by 4.00
08:11:24.817 : > home_all_axis > current_position: (0.00, 0.00, 0.00)
08:11:24.817 : > home_all_axis > destination: (33.00, 124.00, 4.00)
08:11:25.829 : >>> home_axis(2
08:11:25.832 : deploy_z_probe > current_position: (33.00, 124.00, 0.00)
08:11:33.874 : > TRIGGER ENDSTOP > current_position: (33.00, 124.00, 0.00)
08:11:33.874 : > AFTER sync_plan_position > SET current_position: (33.00, 124.00, 2.16)
08:11:33.878 : axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (33.00, 124.00, 2.16)
08:11:33.882 : > SERVO_ENDSTOPS > Stow with write()
08:11:33.882 : > SERVO_LEVELING > stow_z_probe
08:11:33.886 : stow_z_probe > current_position: (33.00, 124.00, 2.16)
08:11:33.886 : Raise Z (after) by 15.00
08:11:33.890 : > SERVO_ENDSTOPS > do_blocking_move_to: (33.00, 124.00, 17.16)
08:11:33.890 : do_blocking_move_to: (33.00, 124.00, 17.16)
08:11:37.937 : <<< home_axis(2
08:11:37.937 : <<< Z_SAFE_HOMING
08:11:37.941 : > (home_all_axis || homeZ) > final: (33.00, 124.00, 17.16)
08:11:37.941 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
08:11:37.941 : <<< gcode_G28
08:11:37.941 : ok
08:12:07.017 : gcode_G29 >>>
08:12:07.017 : G29 Auto Bed Leveling
08:12:07.021 : enable_endstops(true)
08:12:07.021 : z_before = (before) 15.00
08:12:07.021 : probe_pt >>>
08:12:07.021 : > ProbeAction:1
08:12:07.025 : > current_position: (33.00, 124.00, 17.16)
08:12:07.025 : Z Raise to z_before 15.00
08:12:07.025 : > do_blocking_move_to: (33.00, 124.00, 15.00)
08:12:07.029 : do_blocking_move_to: (33.00, 124.00, 15.00)
[... 182 lines cut ...]
08:41:54.915 : > SERVO_ENDSTOPS > do_blocking_move_to: (123.00, 179.00, 19.21)
08:41:54.918 : do_blocking_move_to: (123.00, 179.00, 19.21)
08:41:58.966 : <<< probe_pt
08:41:58.966 : > probing complete > current_position: (123.00, 179.00, 19.21)
08:41:58.970 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
08:41:58.978 : Eqn coefficients: a: -0.00201087 b: 0.03779166 d: -1.87064266
08:41:58.978 : Mean of sampled points: 1.45312500
08:41:58.978 : 
08:41:58.983 : Bed Height Topography:
08:41:58.983 : +-----------+
08:41:58.983 : |...Back....|
08:41:58.983 : |Left..Right|
08:41:58.983 : |...Front...|
08:41:58.983 : +-----------+
08:41:58.986 : +2.90938 +2.75938
08:41:58.986 : -2.67812 -2.99062
08:41:58.991 : planeNormal x: 0.002011 y: -0.037792 z: 1.000000
08:41:58.995 : set_bed_level_equation_lsq > current_position: (122.96, 179.61, 19.21)
08:41:58.995 : 
08:41:58.995 : Bed Level Correction Matrix:
08:41:58.999 : +0.999998 +0.000000 -0.002011
08:41:58.999 : +0.000076 +0.999287 +0.037765
08:41:59.003 : +0.002009 -0.037765 +0.999285
08:41:59.003 : <<< gcode_G29
08:41:59.003 : ok

davbardi, 2015-05-02:
@thinkyhead,  I've tried testing on the glass plate now and yes still getting an incorrect offset. On a nearly level plate it is printing quite well.  But when I lift the back side with a pencil (approx. 6mm) the hotend is dragging on the glass ( worked out this was about 1.5mm too low).

thinkyhead, 2015-05-02:
@davbardi Thank you for your detailed testing and reports! I'm reading through the logs now. Would you say that `G28` is now working reliably?

thinkyhead, 2015-05-02:
Hmm, when homing, I see that the `home_offset[Z_AXIS]` value is added. However, when the bed is probed, the `home_offset[Z_AXIS]` is not added to the Z position when it hits the bed. That might not actually be too serious, because the probing seems to be acquiring sane values (0.1 and 0.16), but it's something to watch out for.

And, I see that when probing starts, it tries to move Z to 15, but since the `current_position` of Z has had the `home_offset[Z_AXIS]` applied, it actually moves to a point 11.73mm above the bed. So, that should be adjusted.

thinkyhead, 2015-05-02:
By the way, a `home_offset[Z_AXIS]` value of 3.27 will of course cause printing to start 3.27mm below the bed, so I assume it's just there for testing.

davbardi, 2015-05-02:
@thinkyhead,, yes G28 seems to working well.

The 
      echo:Z Offset :
      -3.27
      ok
comes from the M851 command run before G28 - just to show the current program probe offset for testing...

thinkyhead, 2015-05-02:
@davbardi I see now, thanks. I conflated `zprobe_zoffset` with `home_offset[Z_AXIS]`. So I'm still wondering when it's appropriate to add `zprobe_zoffset`. Currently it is being added in `axis_is_at_home(Z_AXIS)`, which should be good in all cases. But when probing that function isn't called. Maybe it should be (It would do no harm, just set the correct home coordinate for the Z axis). I will add some experimental stuff and, actually, put `AUTO_BED_LEVELING` onto my i3 (without the servo part) and do faux leveling so I can test here.

thinkyhead, 2015-05-04:
@davbardi I have gone back to some older code and created a hybrid to see how it compares. It incorporates all the changes to leveling and homing that "modernize" the code and fix bugs, but it leaves out all the changes since March 16 that would affect behavior. So, it has the same bugs that would have existed on March 16, but none of the new bugs introduced since that date. It doesn't include any logging, but if you would like to test it and describe its behavior, that would be very useful.

https://github.com/thinkyhead/Marlin/tree/before_delta_merge

davbardi, 2015-05-04:
@thinkyhead, I've tried "before_delta_merge".  
Ouch. Steppers made not-a-good grinding noise after zeroing on x and y.  Z zeroing did not deploy the probe before zeroing (I killed it before it hit the bed). My probe servo made some not so good noises and stopped going back far enough to the clear position (may have broken something. When I put back the previous firmware I had move the arm round to a higher setting range to get it to work...).  Not worried as the machine is just for testing (and have plenty of spare servos..).  I've a CoreXY that is a modified Smartfriendz SmartCore, just to get something up cheaply...
Short story - unable to test with this firmware.  Think I'd be spending more time fixing other issues instead of the bed-levelling :) .

lrpirlet, 2015-05-06:
@thinkyhead, I had one hour so tested **before_delta_merge**
That firmware just forces the axis on the endstop...
I made a diff with 1.0.3 DEV dated 26-apr...
In the function ```homeaxis(AxisEnum axis)```
I find at line 1562
``` enable_endstops(false); // Disable endstops while moving away ```
without corresponding ```enable_endstops(true); ``` before moving toward the endstop

I also see that ```sync_plan_position();``` is NOT issued after second Z going to endstop... I predict that if I had time to comment and recompile, the G28 would NOT have set Z = 0 with the right value...

thinkyhead, 2015-05-06:
@davbardi @lrpirlet Yes, do be careful with that code! It includes bugs from that point, including not including the code for enable_endstops where it should. The commit https://github.com/thinkyhead/Marlin/commit/99977ef36e2f659ff2d35884fc7983616c81a410 has all the changes that need to be made to bring the code to its _current state_ – both the good and the bad. I will post an updated branch that includes those kinds of bug-fixes but nothing that should change behavior.

thinkyhead, 2015-05-06:
@davbardi @lrpirlet Ok, I applied all the simple bug-fixes to that branch, so you should no longer have endstop problems with it: https://github.com/thinkyhead/Marlin/tree/before_delta_merge

lrpirlet, 2015-05-07:
@thinkyhead Thanks... I downloaded this latest version. I used my configuration.h and configuration_adv.h to compile and  upload the firmware...
G28 works fine except maybe for the probe retracting without first going away from the bed ( the cardboard bed is flexible enough to allow it) so I would not like to do in on my borosilicate bed.
G1 Z0 brings it a the correct height ( thickness of a sheet of paper)
G29 completes fine and seems to give a much better result. I put a pencil under my cardboard bed (@davbardi nice idea :-) ). I lowered the Hot-end tip to Z=0, then moved it in a square pattern... The hot-End tip did follow the cardboard and did not even mark teh cardboard...

I have an excellent feeling about G29 now... Thanks.

Please find the repetier log... It may bring some 
```C++
09:38:34.803 : start
09:38:34.803 : echo: External Reset
09:38:34.803 : Marlin beforeDelta 07.05
09:38:34.803 : echo: Last Updato:Hardcoded Default Settings thor: LRP Tuned 7.05.15
09:38:34.803 : Compiled: May  7 2015
09:38:34.803 : echo: Free Memory: 2805  PlannerBufferBytes: 1232
09:38:34.803 : echo:Hardcoded Default Settings Loaded
09:38:34.803 : echo:Steps per unit:
09:38:34.803 : echo:  M92 X80.00 Y80.00 Z4000.00 E406.00
09:38:34.803 : echo:Maximum feedrates (mm/s):
09:38:34.803 : echo:  M203 X200.00 Y200.00 Z1.00 E45.00
09:38:34.803 : echo:Maximum Acceleration (mm2):
09:38:34.803 : echo:  M201 X1000 Y1000 Z50 E5000
09:38:34.803 : echo:Accelerations: P=printing, R=retract and T=travel
09:38:34.803 : echo:  M204 P1500.00 R2000.00 T1500.00
09:38:34.803 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z j?Bj??s),  E=maximum E jerk (mm/s)
09:38:34.803 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
09:38:34.803 : echo:Home offset (mm):
09:38:34.803 : echo:  M206 X0.00 Y0.00 Z0.00
09:38:34.803 : echo:Material heatup parameters:
09:38:34.803 : echo:  M145 M0 H180 B70 F255
09:38:34.803 : echo:  M145 M1 H240 B100 F255
09:38:34.803 : echo:PID settings:
09:38:34.803 : echo:  M301 P22.41 I1.56 D80.41 C1.00
09:38:34.803 : echo:Filament settings: Disabled
09:38:34.803 : echo:Z-Probe Offset (mm):
09:38:34.803 : echo:  M851 Z-2.33
09:38:35.364 : N1 M110 *2
09:38:35.832 : N1 M110 *2
09:38:38.749 : echo:SD init fail
09:38:38.749 : echo:Unknown command: ""
09:38:38.749 : ok
09:38:38.749 : N2 M115 *4
09:38:38.827 : ok
09:38:38.843 : FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Pi3 3D LabteK EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
09:38:38.843 : ok
09:38:38.843 : N4 M111 S6 *67
09:38:38.874 : N5 T0 *31
09:38:38.874 : ok
09:38:38.874 : N6 M20 *55
09:38:38.874 : echo:Active Extruder: 0
09:38:38.874 : ok
09:38:38.874 : Begin file list
09:38:38.874 : N7 M80 *60
09:38:38.874 : End file list
09:38:38.874 : ok
09:38:38.874 : ok
09:38:38.874 : N8 M220 S100 *73
09:38:38.874 : N9 M221 S100 *73
09:38:38.874 : ok
09:38:38.874 : ok
09:38:38.874 : N10 M111 S6 *118
09:38:38.890 : N11 T0 *42
09:38:38.890 : ok
09:38:38.890 : echo:Active Extruder: 0
09:38:38.890 : ok
09:38:50.605 : N18 G28 *10
09:39:21.681 : ok
09:42:26.935 : N109 M114 *15
09:42:26.935 : X:114.00 Y:104.00 Z:2.33 E:0.00 Count X: 114.00 Y:104.00 Z:2.33
09:42:26.935 : ok
[... 51 lines cut ...]
09:49:53.353 : Bed Level Correction Matrix:
09:49:53.353 : +0.999998 +0.000000 +0.001752
09:49:53.353 : -0.000054 +0.999523 +0.030872
09:49:53.353 : -0.001752 -0.030872 +0.999522
09:49:53.353 : ok
09:50:20.591 : N178 M114 *9
09:50:20.591 : X:186.01 Y:177.05 Z:2.02 E:0.00 Count X: 186.00 Y:176.90 Z:7.81
09:50:20.591 : ok
09:51:40.182 : N218 G0 X0 Y0 F1000 *100
09:51:40.182 : ok
09:52:28.355 : N242 G1 X100 Y100 *13
09:52:28.355 : ok
09:52:43.612 : N250 G1 Z0 *69
09:52:43.612 : ok
09:53:22.393 : N270 G1 X110 Y110 *12
09:53:22.456 : ok
09:53:53.125 : N286 G1 X150 Y150 *5
09:53:53.125 : ok
09:54:14.513 : N297 G1 X150 Y50 *52
09:54:14.513 : ok
09:54:46.197 : N313 G1 X50 Y50 *8
09:54:46.197 : ok
09:55:10.002 : N326 G1 X50 Y150 *63
09:55:10.002 : ok
```

nophead, 2015-05-07:
G1 Z0 should just touch the bed, not be the thickness of paper away, which can be anything between about 0.1mm and 0.2mm. A big random offset compared to the average layer height.

davbardi, 2015-05-07:
@thinkyhead , @lrpirlet.  I've just tried out the latest https://github.com/thinkyhead/Marlin/tree/before_delta_merge, - much smoother.  Have not had time to heavily test , but it does appears to be working well.  It seems to be moving the head the same clearance when I tested with bed near level, and again with a pencil under the back.   I agree there is the probe retraction problem, but I suspect that isn't a major thing to fix.   I'll leave it to you experts to work out why the post delta merge has change things.  I'm not much of a coder, and my head nearly exploded just trying to follow the logic for bed-levelling :)

thinkyhead, 2015-05-08:
@davbardi Thanks for giving that a test! It's so close and yet so far. The probe not moving away from the bed has surely been fixed by now. I probably just need to cherry-pick the right lines in the diff. And that will leave very few differences between __before_delta_merge__ and the current homing/leveling code. I will apply a couple more of the less scary-looking patches which bring it closer to the current code and we'll see if I can get the after-probing raise to happen…

thinkyhead, 2015-07-10:
Ok, this branch can be used to get debug output based on leveling code as of July 9…
https://github.com/thinkyhead/Marlin/tree/debug_G29

boelle, 2015-08-03:
moved to: https://github.com/MarlinFirmware/MarlinDev/issues/29

closed 2015-08-03 by boelle