MarlinFirmware/Marlin #1763: Tune > Nozzle does not work.

epatel, 2015-04-01:
I tried the current commit just now
commit be593b600b8b9b427656dd27be7fc277aec0a8a5

Couldn't change the Nozzle temperature while printing, it showed briefly then returned to the menu.

I am running on a RAMPS 1.4 + REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

AnHardt, 2015-04-01:
Will be fixed by #1758

thinkyhead, 2015-04-01:
@epatel Audio feedback was missing, fixed by #1758, but otherwise it should work.

AnHardt, 2015-04-01:
@thinkyhead 
Have you tried it? 
What is the side defect of the beep? 
A delay!

epatel, 2015-04-01:
Sorry, not tested yet. I am in sweden and just got to work. Will test later at home.

thinkyhead, 2015-04-01:
@AnHardt I have tried beeping menus, as I was working on improving the sound last night, but I actually haven't tried editing values with the code I flashed yesterday (which has no menu beep). Are you saying that because there was no beep (hence no delay as a side-effect) that the value was not updating? Or are you trying to point out something else?

AnHardt, 2015-04-01:
@thinkyhead 
I say as the titele of #1758 say 'Fix: can't enter MENU_MULTIPLIER_ITEM' what is the analysed version of #1763 'Tune > Nozzle does not work.' 
You are just not able to enter the menuitem without the delay.
I have not analysed this completely but compared to your last commit at that place.
I guess there has to be at least one interrupt to clear the button state - or someting like that.

epatel, 2015-04-01:
Just tested with latest commit. Works now! :+1:

closed 2015-04-01 by epatel

thinkyhead, 2015-04-02:
@AnHardt Wow, that is interesting. I would think the absence of a delay _might_ cause the value to be updated more than once, not skipped. The menu system is one area needing a little bit of logical cleanup anyway.