MarlinFirmware/Marlin #1769: strange auto level behaviour...

MoonshineSG, 2015-04-01:
Using the latest codes (as of posting), I have some strange behaviour:

1. When issuing code M401 the probe deploys without raising Z. M402 (retract) first moves the Z up with 15 then retracts the probe. 

Shouldn't be the other way, or raise/ensure Z is high enough for both M401 and M402 ? 

2. After performing a auto levelling (G29 P3 V3 T) the Z is left at an odd difference of around 7.2mm and it basically prints "in the air" ...

3. During G29 the probe raises more that pre-setup 5 (Z_RAISE_BETWEEN_PROBINGS) between the different points during probing... 

Can anyone explain/help ?  Before doing the latest updates, I was able to print at fairly ok quality, but with very troublesome manual levelling of the bed, so I setup a servo arm to do auto levelling..  

Configuration.h: http://pastebin.com/is1bQAcn

Wurstnase, 2015-04-01:
2.) I've seen the same. While probing the Z-axis is moving twice. Only the at the first position it's only moving once.

MoonshineSG, 2015-04-01:
Yes, smth like that .. it moves up in 2 steps....

thinkyhead, 2015-04-01:
#1764 addresses some issues in the general area of probing and leveling. Try the latest code and see if it improves anything. I've been adding some more documentation to these functions, hoping to make the flow clearer. Hopefully this will all be more solid soon.

MoonshineSG, 2015-04-01:
Deployed 1764 and the results are below:

1. Still the same (most likely not changed)

2.  After G29 the Z0 is still too high (about 5mm in my case)

3. still happening. From second autolevel point the raising of the probe happens on a "2 step" higher than the  Z_RAISE_BETWEEN_PROBINGS

roboprint, 2015-04-01:
Same results for 2 and 3, using delta printer

mhespenh, 2015-04-01:
2) Offset appears to be around 4mm on my Prusa i3, able to correct it after G29 by adding -4mm to M851 value.

3) Confirmed same as MoonshineSG- Z-axis raises 'Z_RAISE_BETWEEN_PROBINGS'mm twice with slight delay between each between every probe-point.

thinkyhead, 2015-04-02:
Thanks for the reports. I'll have another long look at the code, continue to document, and hopefully it will become more clear where it's messing up. Some of this began with moving the Z_RAISE into the `retract_probe` routine. If the probe is already retracted, it should bail out of `retract_probe` right away, and I suspect it may not be doing this.

If someone wants to take the initiative to add lots of SERIAL_ECHO commands to all the involved functions, and get a full snapshot of the flow, that would be excellent. I _could maybe_ do probe testing without a probe (as I don't have one) by pretending that the Z min endstop is a probe, but I think it will be easier and better for users with actual probes to play around with this.

Natealus, 2015-04-02:
So I found something interesting that verifies this double Z movement between probings. It turns out it is related to the Z_RAISE_AFTER_PROBING setting because it moves the 50mm of that first...then the 20 I put in for BETWEEN_PROBINGS. In fact I was able to verify that Z_RAISE_BEFORE_PROBING actually was having no effect because I took it all the way to 70mm and it still was deploying my servo probe too close to the bed. When I raised BETWEEN_PROBINGS from 10mm to 20mm it regained that Z height it needed to not contact the probe.

But as well I have Z_RAISE_AFTER_PROBING commented out for a few days because it would hijack the Manual Z I had of 367.55 to the value of AFTER_PROBING of 50mm. And thus would probe in mid air and get nothing. On the most recent merges I decided to enable it as I wanted a through test and it turns out that now it Adds that 50 to the 367.55 for a total of 417.55. So I've adjusted the manual Z to compensate to continue testing and thats when I discovered this "Double Raise" on the Z height and that its directly related to Z_RAISE_AFTER_PROBING. 

As I tested in several different configuration variables, I've verified that Z_RAISE_BEFORE_PROBING seems to have no effect on the G29 auto probing process.

TechMasterJoe, 2015-04-03:
compairing code i find this
current build
"float measured_z,
                z_before = Z_RAISE_BETWEEN_PROBINGS + (probePointCounter ? current_position[Z_AXIS] : 0);"

build i downloaded on 3-15-15
"float measured_z,
                z_before = probePointCounter == 0 ? Z_RAISE_BEFORE_PROBING : current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS;"

my head says they do the same 
but giving the math fake data to display using a uno i get mixed numbers
granted the best way is to swap them and see but its 5am and to many sleepy people to wake up

Natealus, 2015-04-03:
Time for the morning routine to shine hmm? :) 

But yeah it looks like the RAISE_BEFORE got knocked out of the current build in that section but could that section be called for setting the initial Z position for each probing and maybe a split off Z_RAISE_BEFORE_PROBING call isn't getting pointed to?

thinkyhead, 2015-04-04:
#1798

MoonshineSG, 2015-04-04:
Tested with a3e129e091.

none of the above 1, 2 or 3 seams fixed. And it is super strange... it starts with G28.... 

I have defined in source Z_PROBE_OFFSET_FROM_EXTRUDER -5.5, and I overwrite this value using M851 Z-2.70 and save to EEPROM

After a G28 when I do G1 Z0, the nozzle is left at a certain distance from the bed. I am allowed to go to minus values up to  the value I set in Configuration.h (in this case -5.5, but i tested with other values...). The -5.5 in this case is more than the distance between the bed and the nozzle so it would push the bed (and probably crack the glass if I try)
The actual distance between the nozzle and the bed when doing a G1 Z0 after a G28 is 4.5 and I can't figure out where that came from...

thinkyhead, 2015-04-04:
@MoonshineSG Sometimes it's the difference between some setting, as in `10.0 - 4.5 = 5.5`.

MoonshineSG, 2015-04-04:
But that 5.5 should have been overwritten by the 2.7... Or I am misunderstanding what the M851 does...

MoonshineSG, 2015-04-04:
And my settings are:
#define Z_RAISE_BEFORE_HOMING  15   
#define Z_RAISE_BEFORE_PROBING 15   
#define Z_RAISE_BETWEEN_PROBINGS 5  
#define Z_RAISE_AFTER_PROBING 15  

no 10 here..

thinkyhead, 2015-04-04:
Process of illumination. :bulb:

TechMasterJoe, 2015-04-04:
OK i did a few test and added some mindless debugging
heres what i see
when it calls Probe_pt the first thing it calls is
```do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], z_before);
// it's getting "z_before"
//from
" float measured_z,
z_before = Z_RAISE_BETWEEN_PROBINGS + (probePointCounter ? current_position[Z_AXIS] : 0);"
then
do_blocking_move_to(x - X_PROBE_OFFSET_FROM_EXTRUDER, y - Y_PROBE_OFFSET_FROM_EXTRUDER, current_position[Z_AXIS]);
// and then
 #if Z_RAISE_BETWEEN_PROBINGS > 0
      if (retract_action == ProbeStay) {
        do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS);
        st_synchronize();
      }
    #endif
```

thinkyhead, 2015-04-04:
Ha, I think you've got something there. I will look more closely after much needed sleep! :zzz:

TechMasterJoe, 2015-04-04:
this is what i have done to probe_pt
in a way that supports both delta and us other guys lol
but it needs a way to skip the first point for delta's 
much like the z_before skips point 0
it did fix the Z offset problems i was having as well
```
static float probe_pt(float x, float y, float z_before, ProbeAction retract_action=ProbeEngageAndRetract, int verbose_level=3) {
     #ifdef DELTA
      #if Z_RAISE_BETWEEN_PROBINGS > 0
        if (retract_action == ProbeStay) { do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS); st_synchronize(); }
     #else
        do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], z_before);
     #endef
    // move to right place  
    do_blocking_move_to(x - X_PROBE_OFFSET_FROM_EXTRUDER, y - Y_PROBE_OFFSET_FROM_EXTRUDER, current_position[Z_AXIS]);

    #if !defined(Z_PROBE_SLED) && !defined(Z_PROBE_ALLEN_KEY)
      if (retract_action & ProbeEngage) deploy_z_probe();
    #endif

    run_z_probe();
    float measured_z = current_position[Z_AXIS];
    #endif

    #if !defined(Z_PROBE_SLED) && !defined(Z_PROBE_ALLEN_KEY)
      if (retract_action & ProbeRetract) stow_z_probe();
    #endif

    if (verbose_level > 2) {
      SERIAL_PROTOCOLPGM(MSG_BED);
      SERIAL_PROTOCOLPGM(" X: ");
      SERIAL_PROTOCOL_F(x, 3);
      SERIAL_PROTOCOLPGM(" Y: ");
      SERIAL_PROTOCOL_F(y, 3);
      SERIAL_PROTOCOLPGM(" Z: ");
      SERIAL_PROTOCOL_F(measured_z, 3);
      SERIAL_EOL;
    }
    return measured_z;
  }
```

TechMasterJoe, 2015-04-04:
I know I'm eating a ton of space but hey like it materhackers

doing it this way might get around the need to pass a probe count for delta's
```
static float probe_pt(float x, float y, float z_before, ProbeAction retract_action=ProbeEngageAndRetract, int verbose_level=3) {
     #ifndef DELTA
        do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], z_before);
     #endif
    // move to right place  
    do_blocking_move_to(x - X_PROBE_OFFSET_FROM_EXTRUDER, y - Y_PROBE_OFFSET_FROM_EXTRUDER, current_position[Z_AXIS]);

    #if !defined(Z_PROBE_SLED) && !defined(Z_PROBE_ALLEN_KEY)
      if (retract_action & ProbeEngage) deploy_z_probe();
    #endif

    run_z_probe();
    float measured_z = current_position[Z_AXIS];
    
     #ifdef DELTA
        #if Z_RAISE_BETWEEN_PROBINGS > 0
        if (retract_action == ProbeStay) { do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS); st_synchronize(); }
        #endif
     #endif
 

    #if !defined(Z_PROBE_SLED) && !defined(Z_PROBE_ALLEN_KEY)
      if (retract_action & ProbeRetract) stow_z_probe();
    #endif

    if (verbose_level > 2) {
      SERIAL_PROTOCOLPGM(MSG_BED);
      SERIAL_PROTOCOLPGM(" X: ");
      SERIAL_PROTOCOL_F(x, 3);
      SERIAL_PROTOCOLPGM(" Y: ");
      SERIAL_PROTOCOL_F(y, 3);
      SERIAL_PROTOCOLPGM(" Z: ");
      SERIAL_PROTOCOL_F(measured_z, 3);
      SERIAL_EOL;
    }
    return measured_z;
  }
```

thinkyhead, 2015-04-04:
Well, that function has long had `z_before` as an argument, which is, I believe. supposed to indicate the Z coordinate where the axis should go before starting the individual probe. The activation of the probe wasn't part of this function until recently, so there was more fine-grained control. I think it makes sense to go to each individual call to `probe_pt` and make a note about where the Z is supposed to be, before and after that individual call to `probe_pt`, and even insert the code that should be getting called for that part. Then it may become clear that another argument is needed, or that the tested conditions are lacking.

Just from a review of the code you show above, it seems like there is something missing. I'm thinking…

    #ifdef DELTA
      #if Z_RAISE_BETWEEN_PROBINGS > 0
        if (retract_action == ProbeStay) {
          current_position[Z_AXIS] += Z_RAISE_BETWEEN_PROBINGS;
          do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]);
          st_synchronize();
        }
      #endif
    #else
      do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], z_before);
    #endif

TechMasterJoe, 2015-04-04:
i swear I'm not saying this to be a dick no really
scroll right
it's the same just all on the same line. all I do is agree with you I'm in PST it's 4:41AM I'm half asleep as it is so i might still be wrong lets just say auto spell correct is the only thing saving me now, i think i made 15 typos just typing this. 
the only reason i put the work in is i do have a probe and encoders on the printer next to me so if it moves i can tell how far +-.002mm far better then it can move given it's a 12" makerfarm i3v

thinkyhead, 2015-04-04:
Not the same. The code is updating `current_position[Z_AXIS]` before calling `do_blocking_move`, whereas the old code passes `current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS` as the last argument.

In any case, I followed the codepath, and I see that it's not necessary. The value gets updated at the end of `prepare_move_raw`.

TechMasterJoe, 2015-04-04:
ah ya i see it now. back to that sleep thing
if i get annoying just tell me off lol
well i'm really going to sleep now so check back in a few hours

TechMasterJoe, 2015-04-05:
after looking even more and to the sum of 80 echo's
it looks like Z_RAISE_AFTER_PROBING is get added once
but moves twice set it to 10 mm and Z probe thinks it's 10mm+ probe offset in this case 12.15 is on the LCD for me
but it's really 22.15 this adds the extra 7.85 I'm seeing on the dot
the odd part is it don't even know that it's an extra 10mm up and i can find out for the life of me where that 2X is happening

```
19:48:07.094 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.80 E5.00
19:48:07.094 : echo:Home offset (mm):
19:48:07.094 : echo:  M206 X0.00 Y0.00 Z0.00
19:48:07.094 : echo:PID settings:
19:48:07.094 : echo:  M301 P22.20 I1.08 D100.00
19:48:07.094 : echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
19:48:07.094 : echo:  M207 S1.00 F1500.00 Z0.00
19:48:07.094 : echo:Recover: S=Extra length (mm) F:Speed (mm/m)
19:48:07.104 : echo:  M208 S0.00 F480.00
19:48:07.104 : echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
19:48:07.104 : echo:  M209 S0
19:48:07.104 : echo:Filament settings: Disabled
19:48:07.104 : echo:Z-Probe Offset (mm):
19:48:07.114 : echo:  M851 Z-2.15
19:48:07.114 : ok
19:48:12.604 : N17 G28 *5
19:48:12.604 : #1480 Z:0.000
19:48:12.604 : #1490 Z:0.000
19:48:12.724 : #1499 Z: 0.000
19:48:12.934 : #983 current_position0.000
19:48:12.934 : #983 base_home_pos0.000
19:48:12.934 : #983 home_offset0.000
19:48:12.934 : #1555 Z: 0.000
19:48:12.934 : #1480 Z:0.000
19:48:12.944 : #1490 Z:0.000
19:48:13.054 : #1499 Z: 0.000
19:48:13.324 : #983 current_position0.000
19:48:13.324 : #983 base_home_pos0.000
19:48:13.324 : #983 home_offset0.000
19:48:13.324 : #1555 Z: 0.000
19:48:15.774 : #1948 Z:0.000
19:48:15.774 : #1206 deploy_Z_probe
19:48:16.074 : #1480 Z:0.000
19:48:22.334 : #1490 Z:0.000
19:48:24.584 : #1499 Z: 0.000
19:48:28.004 : #983 current_position0.000
19:48:28.004 : #983 base_home_pos0.000
19:48:28.004 : #983 home_offset0.000
19:48:28.004 : #1555 Z: 0.000
19:48:31.904 : #1274 Z_RAISE_AFTER_PROBING set to 0
19:48:31.904 : #1285 Retract_Z_probe
19:48:32.204 : #2011 Z: 2.150
19:48:32.204 : ok
```

thinkyhead, 2015-04-05:
Thanks @TechMasterJoe – every bit of information helps. I will be going over this whole leveling fiasco more thoroughly again tonight.

TechMasterJoe, 2015-04-05:
just synced to current build as of 3mins ago testing and adding echo's add needed

TechMasterJoe, 2015-04-05:
STUPID FACT
over the last 2 months marlins active Size has shrunk 23Kb for me
RAMBO + reprap graphic LCD + ABL was over 176,3XX bytes now under 144,xxx bytes
so thank you very much thinkyhead

TechMasterJoe, 2015-04-05:
current build Z_RAISE_AFTER_PROBING
dead trying a few things but it don't rise at all
pro is the offsets are right now

thinkyhead, 2015-04-05:
@TechMasterJoe Arduino 1.6.2 compiles Marlin something like 5K smaller also, so thank you open source movement.

thinkyhead, 2015-04-05:
Notes on z raising, but I still haven't gone through the code and followed the full flow. I have several leads to follow up on in my notes.

ear0wax, 2015-04-05:
I can confirm that Raise between probings wasn't working with last night build I would up rolling back firmwares.

maar1201gh, 2015-04-05:
Changing line code 1349 on marlin_main.cpp removes the "2 step" higher 
FROM: 
do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS);
TO:
do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]);

BUT, STILL EXTRUDING ON THE AIR

thinkyhead, 2015-04-07:
Of someone is able to add `SERIAL_ECHO` commands throughout the leveling code to print out the values of `current_position[Z_AXIS]`, `destination[Z_AXIS]`, the stepper's view of the current position (insert a `SERIAL_ECHO` into `st_synchronize`), the state of the endstops, and all that kind of thing, then it would give a nice clear picture of what is happening and when. If this sounds daunting, I can do the job – I will add a bunch of debug output to the code and make a #define for it which is off by default – you can enable it and then we'll have some good data to work with.

MoonshineSG, 2015-04-07:
If you add the ECHO I can run a couple of tests and send you the results. I would add them myself, but I am not sure where...

TechMasterJoe, 2015-04-07:
under what section
toss me a file name and a line / call to look for and i will start adding them
i can test it any time i have a dev copy untouched with config setup complete if you want to just add them and send the file to me email my user name @ gmail. then i can just swap the file and log the the data, i have a 32ch 100mhz logic analyzer and can wire up to all the outputs as well if you want to get real data i can sync and time stamp the files to match

thinkyhead, 2015-04-07:
Little by little fixing I'm fixing up the probe leveling. The probe should now deploy with `G29` in cases where the X probe starts from the right. At the moment I am adding SERIAL_ECHO throughout the leveling process. I will post that soon, and after the data has been collected I will remove the extra code.

thinkyhead, 2015-04-08:
Ok, I have posted the latest code, with debugging added. To help debug the probing behavior, get the code at this branch:

https://github.com/thinkyhead/Marlin/tree/more_configurator

Find the option at the top of `Marlin_main.cpp` named `DEBUG_LEVELING` and enable it. Then run probing, if possible, and you will see output to the serial console. Paste the output from the serial console on Issue #1844.

MoonshineSG, 2015-04-08:
src/Marlin_main.cpp:1245:77: error: macro "SERIAL_ECHOLNPGM" passed 2 arguments, but takes just 1
src/Marlin_main.cpp: In function ‘void print_xyz(const char*, float, float, float)’:
src/Marlin_main.cpp:1028:7: error: initializer fails to determine size of ‘__c’
src/Marlin_main.cpp: In function ‘void deploy_z_probe()’:
src/Marlin_main.cpp:1245:7: error: ‘SERIAL_ECHOLNPGM’ was not declared in this scope
src/Marlin_main.cpp: In function ‘float probe_pt(float, float, float, ProbeAction, int)’:
src/Marlin_main.cpp:1395:7: error: call of overloaded ‘serial_echopair_P(const char*, ProbeAction&)’ is ambiguous
src/Marlin_main.cpp:1395:7: note: candidates are:
src/Marlin_main.cpp:382:6: note: void serial_echopair_P(const char*, float)
src/Marlin_main.cpp:383:6: note: void serial_echopair_P(const char*, double)
src/Marlin_main.cpp:384:6: note: void serial_echopair_P(const char*, long unsigned int)
src/Marlin_main.cpp:1429:11: error: call of overloaded ‘serial_echopair_P(const char*, int)’ is ambiguous
src/Marlin_main.cpp:1429:11: note: candidates are:
src/Marlin_main.cpp:382:6: note: void serial_echopair_P(const char*, float)
src/Marlin_main.cpp:383:6: note: void serial_echopair_P(const char*, double)
src/Marlin_main.cpp:384:6: note: void serial_echopair_P(const char*, long unsigned int)
src/Marlin_main.cpp: In function ‘void homeaxis(AxisEnum)’:
src/Marlin_main.cpp:1629:7: error: call of overloaded ‘serial_echopair_P(const char*, AxisEnum&)’ is ambiguous
src/Marlin_main.cpp:1629:7: note: candidates are:
src/Marlin_main.cpp:382:6: note: void serial_echopair_P(const char*, float)
src/Marlin_main.cpp:383:6: note: void serial_echopair_P(const char*, double)
src/Marlin_main.cpp:384:6: note: void serial_echopair_P(const char*, long unsigned int)
.build/mega2560/Makefile:1076: recipe for target '.build/mega2560/src/Marlin_main.o' failed

thinkyhead, 2015-04-08:
Should be better now.
https://github.com/thinkyhead/Marlin/archive/more_configurator.zip

MoonshineSG, 2015-04-08:
The results are below, but the probe went way off to the right and I had to use my finger to trigger the Z-stop, otherwise the nozzle would have crashed into the glass...  (my probe is right-back of the nozzle)

Send: M503 S0
Recv: echo:  M92 X80.50 Y80.50 Z406.00 E83.17
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E35.00
Recv: echo:  M201 X5000 Y5000 Z90 E10000
Recv: echo:  M204 P950.00 R950.00 T950.00
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:
Recv:   M301 P84.66 I12.30 D145.65
Recv:   M304 P259.05 I48.80 D343.79
Recv: echo:echo:  M851 Z-2.70

Send: G29 P2 V3 T
Recv: gcode_G29 >>>
Recv: G29 Auto Bed Leveling
Recv: enable_endstops(true)
Recv: probe_pt >>>
Recv: > ProbeAction:1
Recv: echo:> current_position: (52.00, 83.01, 17.70)
Recv: echo:> do_blocking_move_to: (52.00, 83.01, 15.00)
Recv: echo:do_blocking_move_to: (52.00, 83.01, 15.00)
Recv: echo:> do_blocking_move_to: (2.00, 3.00, 15.00)
Recv: echo:do_blocking_move_to: (2.00, 3.00, 15.00)
Recv: > ProbeDeploy
Recv: echo:deploy_z_probe > current_position: (2.00, 3.00, 15.00)
Recv: echo:run_z_probe > current_position: (2.00, 3.00, -1.40)
Recv: Bed X: 45.000 Y: 15.000 Z: -1.399
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: echo:> current_position: (2.00, 3.00, -1.40)
Recv: echo:> do_blocking_move_to: (2.00, 3.00, 3.60)
Recv: echo:do_blocking_move_to: (2.00, 3.00, 3.60)
Recv: echo:> do_blocking_move_to: (137.00, 3.00, 3.60)
Recv: echo:do_blocking_move_to: (137.00, 3.00, 3.60)
Recv: echo:run_z_probe > current_position: (137.00, 3.00, -12.00)
Recv: Raise Z by 5
Recv: echo:> ProbeStay > do_blocking_move_to: (137.00, 3.00, -7.00)
Recv: echo:do_blocking_move_to: (137.00, 3.00, -7.00)
Recv: Bed X: 180.000 Y: 15.000 Z: -12.000
Recv: probe_pt >>>
Recv: > ProbeAction:0
Recv: echo:> current_position: (137.00, 3.00, -7.00)
Recv: echo:> do_blocking_move_to: (137.00, 3.00, -2.00)
Recv: echo:do_blocking_move_to: (137.00, 3.00, -2.00)
Recv: echo:> do_blocking_move_to: (2.00, 168.00, -2.00)
Recv: echo:do_blocking_move_to: (2.00, 168.00, -2.00)
Recv: echo:run_z_probe > current_position: (2.00, 168.00, -1.90)
Recv: Raise Z by 5
Recv: echo:> ProbeStay > do_blocking_move_to: (2.00, 168.00, 3.10)
Recv: echo:do_blocking_move_to: (2.00, 168.00, 3.10)
Recv: Bed X: 45.000 Y: 180.000 Z: -1.901
Recv: probe_pt >>>
Recv: > ProbeAction:2
Recv: echo:> current_position: (2.00, 168.00, 3.10)
Recv: echo:> do_blocking_move_to: (2.00, 168.00, 8.10)
Recv: echo:do_blocking_move_to: (2.00, 168.00, 8.10)
Recv: echo:> do_blocking_move_to: (137.00, 168.00, 8.10)
Recv: echo:do_blocking_move_to: (137.00, 168.00, 8.10)
Recv: echo:run_z_probe > current_position: (137.00, 168.00, -3.52)
Recv: > ProbeStow
Recv: echo:stow_z_probe > current_position: (137.00, 168.00, -3.52)
Recv: echo:do_blocking_move_to: (137.00, 168.00, 11.48)
Recv: Bed X: 180.000 Y: 180.000 Z: -3.520
Recv: echo:> probing complete > current_position: (137.00, 168.00, 11.48)
Recv: ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
Recv: Eqn coefficients: a: -0.04525634 b: 0.02417526 d: -1.97079811
Recv: Mean of sampled points: -4.70504903
Recv: 
Recv: Bed Height Topography:
Recv: +-----------+
Recv: |...Back....|
Recv: |Left..Right|
Recv: |...Front...|
Recv: +-----------+
Recv:  +2.80357 +1.18534
Recv:  +3.30603 -7.29495
Recv: 
Recv: planeNormal x: 0.045256 y: -0.024175 z: 1.000000
Recv: echo:set_bed_level_equation_lsq > current_position: (136.35, 168.38, 13.60)
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +0.998978 +0.000000 -0.045210
Recv: +0.001092 +0.999709 +0.024119
Recv: +0.045197 -0.024143 +0.998686
Recv: Z Probe End Script: G1 X0 Y0 F20000
Recv: echo:enqueueing "G1 X0 Y0 F20000"
Recv: <<< gcode_G29

[results from octopi Version: 1.2.0-dev-668-g3ebd54d (devel branch) - if this makes a difference... ]

lrpirlet, 2015-04-08:
OK, I have been testing the code I got last week-end...
I have a Prusa I3.
It looks like I have a problem  with G28, G29 (2 different issues maybe ) and M48...
I never get my Hot-end to home Z correctly... At the end of G28 and G29 (5*5 probe) after a G1 Z0, the head is 4mm away from the bed. the M48 find a negative position that added to the probe-offset sums up to 4 mm
Then if I g29 with P2 (multiple of 2)... it gets crazy...
Please find here some , hopefully, relevant information..

I will now use the modified code and try again....
Sorry, depending on time available, I may have some results tomorow or the day after... I will not post anything if you fix it before ... 

BTW, thanks for your efforts.

Sorry, I made a mistake and did NOT keep the development configuration.h when I compiled an old  working firmware in the printer...

Louis Richard

+++++++++++++ Configuration.H extracts +++++++++++++
// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

#define min_software_endstops false // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 190
#define Y_MAX_POS 190
#define Z_MAX_POS 168

  // Offsets to the probe relative to the extruder tip (Hotend - Probe)
  // X and Y offsets must be integers
  #define X_PROBE_OFFSET_FROM_EXTRUDER -19 //-19.20 Probe on: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -9  //-9.40 Probe on: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -2.33  //-2.33 -below (always!)

  #define Z_RAISE_BEFORE_HOMING 6       // (in mm) Raise Z before homing (G28) for Probe Clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case

  #define XY_TRAVEL_SPEED 6000         // X and Y axis travel speed between probes, in mm/min

  #define Z_RAISE_BEFORE_PROBING 8   //How much the extruder will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 2  //How much the extruder will be raised when traveling from between next probing points
  #define Z_RAISE_AFTER_PROBING 8    //How much the extruder will be raised after the last probing point.

++++++++repetier logs and comments +++++++++++

19:38:09.074 : start
19:38:09.074 : echo: External Reset
19:38:09.074 : Marlin1.0.2
19:38:09.074 : echo: Last Updated: Apr  6 2015 21:59:14 | Author: LRP
19:38:09.074 : Compiled: Apr  6 2015
19:38:09.074 : echo: Free Memory: 2842  PlannerBufferBytes: 1232
19:38:09.074 : echo:Hardcoded Default Settings Loaded
19:38:09.074 : echo:Steps per unit:
19:38:09.074 : echo:  M92 X80.00 Y80.00 Z4000.00 E406.00
19:38:09.074 : echo:Maximum feedrates (mm/s):
19:38:09.074 : echo:  M203 X200.00 Y200.00 Z1.00 E45.00
19:38:09.074 : echo:Maximum Acceleration (mm/s2):
19:38:09.074 : echo:  M201 X1000 Y1000 Z50 E5000
19:38:09.074 : echo:Accelerations: P=printing, R=retract and T=travel
19:38:09.074 : echo:  M204 P1500.00 R2000.00 T1500.00
19:38:09.090 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), Xho:Home oXY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
19:38:09.090 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
19:38:09.090 : echo:Home offset (mm):
19:38:09.090 : echo:  M206 X0.00 Y0.00 Z0.00
[... 117 lines cut ...]

20:21:43.764 : ok
20:21:59.801 : N950 G28 X0 Y0 *62
20:22:01.018 : ok
20:22:08.568 : N955 G29 P2 *121

Here comes the fun... Probe first row ok... does not do zig-zag and goes to opposite corner from origin (in my case fron left) probe, then raise the probe and goes to 4th corner... RAISE the probe and try to probe ... I do have a Z-switch wired in paralell to the servo probe... so let go...

20:22:08.568 : G29 Auto Bed Leveling
20:23:39.703 : Eqn coefficients: a: 0.00663226 b: -0.00626000 d: -2.38215875
20:23:39.719 : planeNormal x: -0.006632 y: 0.006260 z: 1.000000
20:23:39.719 : 
20:23:39.719 : Bed Level Correction Matrix:
20:23:39.719 : +0.999978 +0.000000 +0.006632
20:23:39.719 : +0.000042 +0.999980 -0.006260
20:23:39.719 : -0.006632 +0.006260 +0.999958
20:23:39.719 : ok
20:32:48.357 : N1229 M114 *63
20:32:48.357 : X:33.99 Y:179.01 Z:-0.98 E:0.00 Count X: 34.00 Y:179.00 Z:-1.88
20:32:48.357 : ok

YOUP it thinks it is UNDER the bed :-) just for Fun, I measure the height and get 2.42 mmm ... 1.88 + 2.30 = 4.18 I guess that the corrected plan added about 0.18 mm (But pls consider the measure error)

IT does not correspnd to anything except maybe in the Configuration_adv.h where I can see Z_HOME_BUMP_MM 2 that gets multiplied by 2 in the HOMEAXIS routine ... Sorry I am NOT a programmer... I can read some C and sort of understand but in this case in am... completely lost.
:-(

lrpirlet, 2015-04-08:
OK, I found a backup of Configuration.h so was easy to edit the Marlin-more_configurator one...
 Got no time to write long comments...
Head is stil 4 mm above bed.. no more problem with probe raizing when Probing 2*2...
Here the repetier log..
I must correct, I do get twice a raise in G29 after first going up...
Very funny, my head seems to travel INSIDE the bed, my printer report :-)

+++++++++repetier log++++++++
22:16:18.368 : start
22:16:18.368 : echo: External Reset
22:16:18.368 : Marlin1.0.3 dev
22:16:18.368 : echo: Last Updated: Apr  8 2015 22:14:25 | Author: LRP tuned
22:16:18.368 : Compiled: Apr  8 2015
22:16:18.368 : echo: Free Memory: 2425  PlannerBufferBytes: 1232
22:16:18.384 : echo:Hardcoded Default Settings L???)echo:Steps per unit:
22:16:18.384 : echo:  M92 X80.00 Y80.00 Z4000.00 E406.00
22:16:18.384 : echo:Maximum feedrates (mm/s):
22:16:18.384 : echo:  M203 X200.00 Y200.00 Z1.00 E45.00
22:16:18.384 : echo:Maximum Acceleration (mm/s2):
22:16:18.384 : echo:  M201 X1000 Y1000 Z50 E5000
22:16:18.384 : echo:Accelerations: P=printing, R=retract and T=travel
22:16:18.399 : echo:time (m500.00 R2000.00 T1500.00
22:16:18.399 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
22:16:18.399 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
22:16:18.399 : echo:Home offset (mm):
22:16:18.399 : echo:  M206 X0.00 Y0.00 Z0.00
22:16:18.399 : echo:PID settings:
22:16:18.415 : echo:  M301 P22.41 I1.56 D80.41
22:16:18.415 : echo:Filament settings: Disabled
22:16:18.415 : echo:Z-Probe Offset (mm):
22:16:18.415 : echo:  M851 Z-2.33
22:16:18.852 : N1 M110 *2
22:16:18.852 : N1 M110 *2
22:16:18.852 : N2 M115 *4
22:16:19.429 : N4 M111 S6 *67
22:16:19.429 : N5 T0 *31
22:16:19.429 : N6 M20 *55
22:16:19.429 : N7 M80 *60
22:16:22.346 : echo:SD init fail
22:16:22.346 : echo:Unknown command: ""
22:16:22.346 : ok
22:16:22.424 : ok
22:16:22.440 : FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Pi3 3D LabteK EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
22:16:22.440 : ok
22:16:22.440 : N9 M220 S100 *72
22:16:22.440 : N10 M221 S100 *113
22:16:22.440 : N11 M111 S6 *119
22:16:22.440 : N12 T0 *41
22:16:22.440 : ok
22:16:22.440 : echo:Active Extruder: 0
22:16:22.440 : ok
22:16:22.440 : Begin file list
22:16:22.440 : End file list
22:16:22.440 : ok
22:16:22.440 : ok
22:16:22.455 : ok
22:16:22.455 : ok
22:16:22.455 : ok
22:16:22.455 : echo:Active Extruder: 0
22:16:22.455 : ok
22:16:40.114 : N22 G28 X0 Y0 *2
22:16:40.114 : gcode_G28 >>>
22:16:42.127 : home_axis with axis=0
22:16:42.127 : echo:home_axis > current_position: (0.00, 0.00, 0.00)
22:16:42.127 : echo:> homeX: (0.00, 0.00, 0.00)
22:16:49.927 : home_axis with axis=1
22:16:49.927 : echo:home_axis > current_position: (0.00, 0.00, 0.00)
22:16:49.927 : echo:> homeY: (0.00, 0.00, 0.00)
22:16:49.927 : echo:> Set X: (0.00, 0.00, 0.00)
22:16:49.927 : echo:> Set Y: (0.00, 0.00, 0.00)
[... 585 lines cut ...]
22:38:45.762 : echo:> ProbeStay > do_blocking_move_to: (189.00, 179.00, 1.36)
22:38:45.762 : echo:do_blocking_move_to: (189.00, 179.00, 1.36)
22:38:48.788 : probe_pt >>>
22:38:48.788 : > ProbeAction:2
22:38:48.788 : echo:> current_position: (189.00, 179.00, 1.36)
22:38:48.788 : echo:> do_blocking_move_to: (189.00, 179.00, 4.36)
22:38:48.788 : echo:do_blocking_move_to: (189.00, 179.00, 4.36)
22:38:51.815 : echo:> do_blocking_move_to: (34.00, 179.00, 4.36)
22:38:51.815 : echo:do_blocking_move_to: (34.00, 179.00, 4.36)
22:39:07.384 : echo:run_z_probe > current_position: (34.00, 179.00, -1.82)
22:39:07.384 : > ProbeStow
22:39:07.384 : echo:stow_z_probe > current_position: (34.00, 179.00, -1.82)
22:39:07.384 : echo:do_blocking_move_to: (34.00, 179.00, 6.19)
22:39:15.730 : echo:> probing complete > current_position: (34.00, 179.00, 6.19)
22:39:15.730 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
22:39:15.730 : Eqn coefficients: a: 0.00110806 b: 0.00007667 d: -1.84227933
22:39:15.730 : planeNormal x: -0.001108 y: -0.000077 z: 1.000000
22:39:15.745 : echo:set_bed_level_equation_lsq > current_position: (34.01, 179.00, 6.13)
22:39:15.745 : 
22:39:15.745 : Bed Level Correction Matrix:
22:39:15.745 : +0.999999 +0.000000 +0.001108
22:39:15.745 : -0.000000 +1.000000 +0.000077
22:39:15.745 : -0.001108 -0.000077 +0.999999
22:39:15.745 : <<< gcode_G29
22:39:15.745 : ok

Wurstnase, 2015-04-08:
Here is my debug: https://gist.github.com/Wurstnase/da62ee63ae2a02ca3aee

thinkyhead, 2015-04-09:
Thanks all, a lot to read through! I hope to make sense of it all soon...

lrpirlet, 2015-04-09:
OK, I verified my idea... The ONLY change I made from yesterday testing was to change in configurqtion_adv.h file  "#define Z_HOME_BUMP_MM" from 2 to 3... So I compiled with "#define Z_HOME_BUMP_MM 3"..
The distance from the bed is now 6 mm... In fact, I measured 5.98mm mde of a stack of paper, I guess it is closed enough...

As I understand it, the issue is somewhere in the HOMEAXIS routine... In fact, this parameter is only  **multiplied by 2** here in marlin_main.cpp line 1168

      // move back down slowly to find bed
      set_homing_bump_feedrate(Z_AXIS);

   **zPosition -= home_bump_mm(Z_AXIS) * 2;**
      line_to_z(zPosition);
      st_synchronize();
      endstops_hit_on_purpose(); // clear endstop hit flags

      current_position[Z_AXIS] = st_get_position_mm(Z_AXIS);

Now, as I said, I am not quite a progrqmmer, and I just do not fully understand this routine... I guess we are closer.

thinkyhead, 2015-04-10:
That is a very important setting – `Z_HOME_BUMP_MM`. I started out with it too low on my machine, and quickly figured out to set it higher - in my case 2mm was enough - but 2.5 or 3 may be better for probes..

thinkyhead, 2015-04-10:
I've added new logging of all the Z raise values during homing and probing, and made the output slightly cleaner. Clear the browser cache and re-download: https://github.com/thinkyhead/Marlin/archive/more_configurator.zip

thinkyhead, 2015-04-10:
I'm thinking, similar to the way the manual bed leveling works in a step-by-step fashion, there should be a GCode to probe a point on the bed. The point probed is added to an array of a fixed length. When enough points have been probed, you send another GCode that applies the probed points to a bed leveling matrix. Or, applies the most recent probe to the Z homing position. Breaking it up like this could simplify processes like this a lot, and make these simple tricks exposable to the outside in a simple fashion.

lrpirlet, 2015-04-10:
@thinkyhead, Very nice idea, this new set of m code could be used as a Library to build for example a new G29 with lots and lots of possibilities (and parameters)...
 A new G29 that would "make" the bed parallel to the X-Y plan (bed leveling as per common naming)... 
we could then supperpose Mesh leveling to help addesion for a not quite so constant distance between a bed point and a X-Y surface that should be a plan but that is not... the mesh could then be pushed in eeprom once for ever because would be applied to a bed parallel to the X-Y axis.
We could even have a additional m code to generate the huge MESH map..

More, all that would be a lot easier to maintain... one code, one function.

Maybe a new feature? Hopefully we will not need that to solve this bug :-(

I have downloaded the new zip. in a couple of hours, I'll compile, test and report :-)

Thanks to take care, anyway

lrpirlet, 2015-04-10:
ooops 
It looks like I have a compile error in the added instructions...

In file included from Marlin_main.cpp:30:0:
Marlin_main.cpp: In function 'void stow_z_probe()':
Marlin_main.cpp:1333:59: error: 'Z_RAISE_AFTER_PROBINGS' was not declared in this scope
             SERIAL_ECHOPAIR("Raise Z (after) by ", (float)Z_RAISE_AFTER_PROBINGS);
                                                           ^
Marlin.h:91:71: note: in definition of macro 'SERIAL_ECHOPAIR'
 #define SERIAL_ECHOPAIR(name,value) do{ serial_echopair_P(PSTR(name),(value)); }while(0)
                                                                       ^
Marlin_main.cpp:1335:144: error: 'Z_RAISE_AFTER_PROBINGS' was not declared in this scope
             print_xyz("> SERVO_ENDSTOPS > do_blocking_move_to", current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS] + Z_RAISE_AFTER_PROBINGS);
                                                                                                                                                ^
Error compiling.

So, I modified the code to read...
          #ifdef DEBUG_LEVELING
// (float)Z_RAISE_AFTER_PROBINGS gives compile error... so commented next line and modified 1336-1338
//          SERIAL_ECHOPAIR("Raise Z (after) by ", (float)Z_RAISE_AFTER_PROBINGS);
            SERIAL_EOL;
//          print_xyz("> SERVO_ENDSTOPS > do_blocking_move_to", current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS] + Z_RAISE_AFTER_PROBINGS);
            print_xyz("> SERVO_ENDSTOPS > do_blocking_move_to !WRONG VALUE! ", current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]);
            print_xyz(">LRP LRP pls add 8, value for Z_RAISE_AFTER_PROBINGS", current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]);
          #endif

Wurstnase, 2015-04-10:
add a
``` #define Z_RAISE_AFTER_PROBINGS 10 ``` 
or any other length in your config.h

lrpirlet, 2015-04-10:
Thanks... My configuration.h reads: 
   #define Z_RAISE_BETWEEN_PROBINGS 3  

I will try 3.5 because I see **(float)**Z_RAISE_AFTER_PROBINGS

Pure (non) educated guess here :-)

lrpirlet, 2015-04-10:
OK, It does not work.

SO I removed one line and changed the offending line to read:
          #ifdef DEBUG_LEVELING
//            SERIAL_ECHOPAIR("Raise Z (after) by ", (float)Z_RAISE_AFTER_PROBINGS);
//            SERIAL_EOL;
            print_xyz("> SERVO_ENDSTOPS > do_blocking_move_to !LRP WRONG VALUE! see next line", current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]);
            print_xyz("I print  current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]", current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]);
            print_xyz(" + Z_RAISE_AFTER_PROBINGS value is 8.. so add 8 to current position... Thanks :-)", current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]);
          #endif

It now compiles...BUT SOME MANUAL ADDITION WILL BE NEEDED... lol

result: G28 followed by m114 then G1 Z0 ... the  final distance between bed and head is 4mm
I hope that helps...

++++++++++++++repetier log++++++++++
22:37:42.938 : start
22:37:42.954 : echo: External Reset
22:37:42.954 : Marlin1.0.3 dev
22:37:42.954 : echo: Last Updated: Default Setting35:59 | Author: LRP tuned
22:37:42.954 : Compiled: Apr 10 2015
22:37:42.954 : echo: Free Memory: 2123  PlannerBufferBytes: 1232
22:37:42.954 : echo:Hardcoded Default Settings Loaded
22:37:42.954 : echo:Steps per unit:
22:37:42.954 : echo:  M92 X80.00 Y80.00 Z4000.00 E406.00
22:37:42.954 : echo:Maximum feedrates (mm/s):
22:37:42.970 : echo:  M203 X200.00 Y200.00 Z1.00 E45.00
22:37:42.970 : echo:Maximum Acceleration (mm/J?Recho:  M201 X1000 Y1000 Z50 E5000
22:37:42.970 : echo:Accelerations: P=printing, R=retract and T=travel
22:37:42.970 : echo:  M204 P1500.00 R2000.00 T1500.00
22:37:42.985 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z j??(mm/s),  E=maximum E jerk (mm/s)
22:37:42.985 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
22:37:42.985 : echo:Home offset (mm):
22:37:42.985 : echo:  M206 X0.00 Y0.00 Z0.00
22:37:42.985 : echo:PID settings:
22:37:42.985 : echo:  M301 P22.41 I1.56 D80.41
22:37:42.985 : echo:Filament settings: Disabled
22:37:42.985 : echo:Z-Probe Offset (mm):
22:37:42.985 : echo:  M851 Z-2.33
22:37:43.453 : N1 M110 *2
22:37:43.453 : N1 M110 *2
22:37:43.453 : N2 M115 *4
22:37:44.030 : N4 M111 S6 *67
22:37:44.046 : N5 T0 *31
22:37:44.046 : N6 M20 *55
22:37:44.046 : N7 M80 *60
22:37:46.916 : echo:SD init fail
22:37:46.932 : echo:Unknown command: ""
22:37:46.932 : ok
22:37:47.010 : ok
22:37:47.010 : FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Pi3 3D LabteK EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
22:37:47.010 : N9 M220 S100 *72
22:37:47.010 : N10 M221 S100 *113
22:37:47.010 : N11 M111 S6 *119
22:37:47.010 : N12 T0 *41
22:37:47.026 : ok
22:37:47.026 : ok
22:37:47.026 : echo:Active Extruder: 0
22:37:47.026 : ok
22:37:47.026 : Begin file list
22:37:47.026 : End file list
22:37:47.026 : ok
22:37:47.026 : ok
22:37:47.026 : ok
22:37:47.026 : ok
22:37:47.026 : ok
22:37:47.026 : echo:Active Extruder: 0
22:37:47.026 : ok
22:37:57.946 : N18 G28 *10
22:37:57.946 : gcode_G28 >>>
22:38:03.140 : home_axis with axis=0
22:38:03.140 : home_axis > current_position: (0.00, 0.00, 0.00)
22:38:03.140 : > homeX: (0.00, 0.00, 0.00)
22:38:06.697 : N24 M114 *49
22:38:07.945 : home_axis with axis=1
22:38:07.945 : home_axis > current_position: (0.00, 0.00, 0.00)
22:38:07.945 : > homeY: (0.00, 0.00, 0.00)
22:38:07.945 : > Z_SAFE_HOMING >>>
22:38:07.945 : Raise Z (before homing) by 6.00
22:38:07.945 : > home_all_axis > current_position: (0.00, 0.00, 0.00)
22:38:07.945 : > home_all_axis > destination: (114.00, 104.00, 6.00)
22:38:13.998 : deploy_z_probe > current_position: (114.00, 104.00, 0.00)
22:38:29.754 : home_axis with axis=2
22:38:29.754 : home_axis > current_position: (114.00, 104.00, 0.00)
22:38:29.754 : home_axis > SERVO_ENDSTOPS > Stow with write()
22:38:29.754 : home_axis(Z_AXIS) > SERVO_LEVELING > stow_z_probe
22:38:29.754 : stow_z_probe > current_position: (114.00, 104.00, 0.00)
22:38:29.770 : > SERVO_ENDSTOPS > do_blocking_move_to !LRP WRONG VALUE! see next line: (114.00, 104.00, 0.00)
22:38:29.770 : I print  current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]: (114.00, 104.00, 0.00)
22:38:29.770 : + Z_RAISE_AFTER_PROBINGS value is 8.. so add 8 to current position... Thanks :-): (114.00, 104.00, 0.00)
22:38:29.770 : do_blocking_move_to: (114.00, 104.00, 8.00)
22:38:42.125 : > home_all_axis > final: (114.00, 104.00, 8.00)
22:38:42.125 : <<< Z_SAFE_HOMING
22:38:42.125 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
22:38:42.125 : <<< gcode_G28
22:38:42.125 : ok
22:38:42.125 : N28 G1 Z0 *120
22:38:42.140 : X:114.00 Y:104.00 Z:10.33 E:0.00 Count X: 114.00 Y:104.00 Z:10.33
22:38:42.140 : ok
22:38:42.156 : ok

lrpirlet, 2015-04-10:
And G29 gives well something funny (Under the bed) while the head is 11.76 mm away from the bed after a G1 Z0...( that maesure is FAR from precise, I feel like I have in between 11 and 12 mm)

++++++++++++repetier log +++++++++++++
22:43:49.129 : N19 G28 X0 Y0 *10
22:43:49.129 : gcode_G28 >>>
22:43:50.112 : home_axis with axis=0
22:43:50.112 : home_axis > current_position: (0.00, 0.00, 0.00)
22:43:50.112 : > homeX: (0.00, 0.00, 0.00)
22:43:51.064 : home_axis with axis=1
22:43:51.079 : home_axis > current_position: (0.00, 0.00, 0.00)
22:43:51.079 : > homeY: (0.00, 0.00, 0.00)
22:43:51.079 : > Set X: (0.00, 0.00, 0.00)
22:43:51.079 : > Set Y: (0.00, 0.00, 0.00)
22:43:51.079 : > Z_SAFE_HOMING >>>
22:43:51.079 : <<< Z_SAFE_HOMING
22:43:51.079 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
22:43:51.079 : <<< gcode_G28
22:43:51.079 : ok
22:44:07.802 : N29 G29 P2 T V4 *125
22:44:07.802 : gcode_G29 >>>
22:44:07.802 : G29 Auto Bed Leveling
22:44:07.802 : enable_endstops(true)
22:44:07.802 : Raise Z (before) by 8.00
22:44:07.802 : probe_pt >>>
22:44:07.802 : > ProbeAction:1
22:44:07.802 : > current_position: (0.00, 0.00, 0.00)
22:44:07.802 : > do_blocking_move_to: (0.00, 0.00, 8.00)
22:44:07.802 : do_blocking_move_to: (0.00, 0.00, 8.00)
22:44:15.852 : > do_blocking_move_to: (34.00, 29.00, 8.00)
22:44:15.852 : do_blocking_move_to: (34.00, 29.00, 8.00)
22:44:16.351 : > ProbeDeploy
22:44:16.367 : deploy_z_probe > current_position: (34.00, 29.00, 8.00)
22:44:42.325 : run_z_probe > current_position: (34.00, 29.00, -9.84)
22:44:42.325 : Bed X: 15.000 Y: 20.000 Z: -9.842
22:44:42.325 : <<< probe_pt
22:44:42.341 : Raise Z (between) by 3.00
22:44:42.341 : probe_pt >>>
22:44:42.341 : > ProbeAction:0
22:44:42.341 : > current_position: (34.00, 29.00, -9.84)
22:44:42.341 : > do_blocking_move_to: (34.00, 29.00, -6.84)
22:44:42.341 : do_blocking_move_to: (34.00, 29.00, -6.84)
22:44:45.367 : > do_blocking_move_to: (189.00, 29.00, -6.84)
22:44:45.367 : do_blocking_move_to: (189.00, 29.00, -6.84)
22:44:57.613 : run_z_probe > current_position: (189.00, 29.00, -9.70)
22:44:57.613 : Raise Z (between) by 3.00
22:44:57.613 : > ProbeStay > do_blocking_move_to: (189.00, 29.00, -6.70)
22:44:57.613 : do_blocking_move_to: (189.00, 29.00, -6.70)
22:45:00.640 : Bed X: 170.000 Y: 20.000 Z: -9.698
22:45:00.640 : <<< probe_pt
22:45:00.640 : Raise Z (between) by 3.00
22:45:00.640 : probe_pt >>>
22:45:00.640 : > ProbeAction:0
22:45:00.655 : > current_position: (189.00, 29.00, -6.70)
22:45:00.655 : > do_blocking_move_to: (189.00, 29.00, -3.70)
22:45:00.655 : do_blocking_move_to: (189.00, 29.00, -3.70)
22:45:03.682 : > do_blocking_move_to: (34.00, 179.00, -3.70)
22:45:03.682 : do_blocking_move_to: (34.00, 179.00, -3.70)
22:45:19.781 : run_z_probe > current_position: (34.00, 179.00, -9.83)
22:45:19.781 : Raise Z (between) by 3.00
22:45:19.781 : > ProbeStay > do_blocking_move_to: (34.00, 179.00, -6.83)
22:45:19.781 : do_blocking_move_to: (34.00, 179.00, -6.83)
22:45:22.807 : Bed X: 15.000 Y: 170.000 Z: -9.829
22:45:22.807 : <<< probe_pt
22:45:22.807 : Raise Z (between) by 3.00
22:45:22.807 : probe_pt >>>
22:45:22.807 : > ProbeAction:2
22:45:22.807 : > current_position: (34.00, 179.00, -6.83)
22:45:22.807 : > do_blocking_move_to: (34.00, 179.00, -3.83)
22:45:22.823 : do_blocking_move_to: (34.00, 179.00, -3.83)
22:45:25.834 : > do_blocking_move_to: (189.00, 179.00, -3.83)
22:45:25.834 : do_blocking_move_to: (189.00, 179.00, -3.83)
22:45:41.075 : run_z_probe > current_position: (189.00, 179.00, -9.68)
22:45:41.075 : > ProbeStow
22:45:41.075 : stow_z_probe > current_position: (189.00, 179.00, -9.68)
22:45:41.091 : > SERVO_ENDSTOPS > do_blocking_move_to !LRP WRONG VALUE! see next line: (189.00, 179.00, -9.68)
22:45:41.091 : I print  current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]: (189.00, 179.00, -9.68)
22:45:41.091 : + Z_RAISE_AFTER_PROBINGS value is 8.. so add 8 to current position... Thanks :-): (189.00, 179.00, -9.68)
22:45:41.091 : do_blocking_move_to: (189.00, 179.00, -1.68)
22:45:49.437 : Bed X: 170.000 Y: 170.000 Z: -9.676
22:45:49.437 : <<< probe_pt
22:45:49.437 : > probing complete > current_position: (189.00, 179.00, -1.68)
22:45:49.437 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
22:45:49.452 : Eqn coefficients: a: 0.00096126 b: 0.00011833 d: -9.86153221
22:45:49.452 : Mean of sampled points: -9.76137542
22:45:49.452 : 
22:45:49.452 : Bed Height Topography:
22:45:49.452 : +-----------+
22:45:49.452 : |...Back....|
22:45:49.452 : |Left..Right|
22:45:49.452 : |...Front...|
22:45:49.452 : +-----------+
22:45:49.452 : -0.06787 +0.08563
22:45:49.452 : -0.08112 +0.06338
22:45:49.452 : planeNormal x: -0.000961 y: -0.000118 z: 1.000000
22:45:49.468 : set_bed_level_equation_lsq > current_position: (189.00, 179.00, -1.88)
22:45:49.468 : 
22:45:49.483 : Bed Level Correction Matrix:
22:45:49.483 : +1.000000 +0.000000 +0.000961
22:45:49.483 : -0.000000 +1.000000 +0.000118
22:45:49.483 : -0.000961 -0.000118 +1.000000
22:45:49.483 : <<< gcode_G29
22:45:49.483 : ok
22:47:39.185 : N91 M114 *63
22:47:39.185 : X:189.00 Y:179.00 Z:-1.90 E:0.00 Count X: 189.00 Y:179.00 Z:-1.70
22:47:39.185 : ok
22:48:48.059 : N125 G1 Z0 *68
22:48:48.059 : ok

thinkyhead, 2015-04-10:
I fixed the misspelling in the branch.

maverikou, 2015-04-11:
What I don't understand is why probing fails after a random amount of points?

Ran my delta 5 times on latest code, probing failed in different places each time and then the planner did not know exactly where it was - probe stowing completely missed the target.

Wurstnase, 2015-04-11:
Could this something have to do with interrupt protected blocks?

lrpirlet, 2015-04-11:
@thinkyhead, Thanks, maybe I should invest in some glasses...
 
 I ran another test and learn some more... I do become more and more persuaded that the problem is in G28, NOT in G29. I strongly suspect the Z Safe Homing feature

++++++++++ repetier log AND comment imbeded ++++++++
I made A G28...
The head was about 4 mm above the bed (a cardboard that can be punched and give me time to react to kill the printer)
I lowered the head close to the bed and made sure that the Z switch wired in parallel to the probe was triggered at 1 mm hight
I made sure to issue then G92 Z0 

I then issued G28 X0 Y0

21:38:32.567 : N1333 G28 X0 Y0 *0
21:38:32.567 : gcode_G28 >>>
21:38:33.175 : home_axis with axis=0
21:38:33.175 : home_axis > current_position: (0.00, 0.00, 1.80)
21:38:33.175 : > homeX: (0.00, 0.00, 1.80)
21:38:33.799 : home_axis with axis=1
21:38:33.799 : home_axis > current_position: (0.00, 0.00, 1.80)
21:38:33.799 : > homeY: (0.00, 0.00, 1.80)
21:38:33.799 : > Set X: (0.00, 0.00, 1.80)
21:38:33.799 : > Set Y: (0.00, 0.00, 1.80)
21:38:33.799 : > Z_SAFE_HOMING >>>
21:38:33.799 : <<< Z_SAFE_HOMING
21:38:33.799 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
21:38:33.799 : <<< gcode_G28
21:38:33.799 : ok

The head is still 1 mm away from the bed, I issue G29... At the end of G29, the head is still at 1 mm
So I conclude that G28 ends up 4 mm to high.

I think I will disable Z_SAFE_Homing tomorow and report... ( time to cover the bed with a cardboard and try with a finger on the kill button)

21:38:56.248 : N1346 G29 P2 T V4 *118
21:38:56.248 : gcode_G29 >>>
21:38:56.248 : G29 Auto Bed Leveling
21:38:56.263 : enable_endstops(true)
21:38:56.263 : Raise Z (before) by 8.00
21:38:56.263 : probe_pt >>>
21:38:56.263 : > ProbeAction:1
21:38:56.263 : > current_position: (0.00, 0.00, 1.80)
21:38:56.263 : > do_blocking_move_to: (0.00, 0.00, 8.00)
21:38:56.263 : do_blocking_move_to: (0.00, 0.00, 8.00)
21:39:02.488 : > do_blocking_move_to: (34.00, 29.00, 8.00)
21:39:02.488 : do_blocking_move_to: (34.00, 29.00, 8.00)
21:39:03.002 : > ProbeDeploy
21:39:03.002 : deploy_z_probe > current_position: (34.00, 29.00, 8.00)
21:39:16.434 : run_z_probe > current_position: (34.00, 29.00, 2.67)
21:39:16.434 : Bed X: 15.000 Y: 20.000 Z: 2.670
21:39:16.434 : <<< probe_pt
21:39:16.434 : Raise Z (between) by 3.00
21:39:16.434 : probe_pt >>>
21:39:16.434 : > ProbeAction:0
21:39:16.434 : > current_position: (34.00, 29.00, 2.67)
21:39:16.434 : > do_blocking_move_to: (34.00, 29.00, 5.67)
21:39:16.434 : do_blocking_move_to: (34.00, 29.00, 5.67)
21:39:19.461 : > do_blocking_move_to: (189.00, 29.00, 5.67)
21:39:19.461 : do_blocking_move_to: (189.00, 29.00, 5.67)
21:39:31.941 : run_z_probe > current_position: (189.00, 29.00, 2.66)
21:39:31.941 : Raise Z (between) by 3.00
21:39:31.941 : > ProbeStay > do_blocking_move_to: (189.00, 29.00, 5.66)
21:39:31.941 : do_blocking_move_to: (189.00, 29.00, 5.66)
21:39:35.014 : Bed X: 170.000 Y: 20.000 Z: 2.661
21:39:35.014 : <<< probe_pt
21:39:35.014 : Raise Z (between) by 3.00
21:39:35.014 : probe_pt >>>
21:39:35.014 : > ProbeAction:0
21:39:35.029 : > current_position: (189.00, 29.00, 5.66)
21:39:35.029 : > do_blocking_move_to: (189.00, 29.00, 8.66)
21:39:35.029 : do_blocking_move_to: (189.00, 29.00, 8.66)
21:39:38.040 : > do_blocking_move_to: (34.00, 179.00, 8.66)
21:39:38.056 : do_blocking_move_to: (34.00, 179.00, 8.66)
21:39:54.030 : run_z_probe > current_position: (34.00, 179.00, 2.67)
21:39:54.030 : Raise Z (between) by 3.00
21:39:54.046 : > ProbeStay > do_blocking_move_to: (34.00, 179.00, 5.67)
21:39:54.046 : do_blocking_move_to: (34.00, 179.00, 5.67)
21:39:57.057 : Bed X: 15.000 Y: 170.000 Z: 2.665
21:39:57.072 : <<< probe_pt
21:39:57.072 : Raise Z (between) by 3.00
21:39:57.072 : probe_pt >>>
21:39:57.072 : > ProbeAction:2
21:39:57.072 : > current_position: (34.00, 179.00, 5.67)
21:39:57.072 : > do_blocking_move_to: (34.00, 179.00, 8.67)
21:39:57.072 : do_blocking_move_to: (34.00, 179.00, 8.67)
21:40:00.099 : > do_blocking_move_to: (189.00, 179.00, 8.67)
21:40:00.099 : do_blocking_move_to: (189.00, 179.00, 8.67)
21:40:15.683 : run_z_probe > current_position: (189.00, 179.00, 2.49)
21:40:15.683 : > ProbeStow
21:40:15.683 : stow_z_probe > current_position: (189.00, 179.00, 2.49)
21:40:15.683 : Raise Z (after) by 8.00
21:40:15.683 : > SERVO_ENDSTOPS > do_blocking_move_to: (189.00, 179.00, 10.49)
21:40:15.683 : do_blocking_move_to: (189.00, 179.00, 10.49)
21:40:24.029 : Bed X: 170.000 Y: 170.000 Z: 2.493
21:40:24.029 : <<< probe_pt
21:40:24.029 : > probing complete > current_position: (189.00, 179.00, 10.49)
21:40:24.029 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
21:40:24.029 : Eqn coefficients: a: -0.00058226 b: -0.00057333 d: 2.73070001
21:40:24.045 : Mean of sampled points: 2.62237501
21:40:24.045 : 
21:40:24.045 : Bed Height Topography:
21:40:24.045 : +-----------+
21:40:24.045 : |...Back....|
21:40:24.045 : |Left..Right|
21:40:24.045 : |...Front...|
21:40:24.045 : +-----------+
21:40:24.045 : +0.04288 -0.12888
21:40:24.045 : +0.04737 +0.03863
21:40:24.045 : planeNormal x: 0.000582 y: 0.000573 z: 1.000000
21:40:24.060 : set_bed_level_equation_lsq > current_position: (188.99, 178.99, 10.71)
21:40:24.060 : 
21:40:24.060 : Bed Level Correction Matrix:
21:40:24.060 : +1.000000 +0.000000 -0.000582
21:40:24.060 : -0.000000 +1.000000 -0.000573
21:40:24.060 : +0.000582 +0.000573 +1.000000
21:40:24.060 : <<< gcode_G29
21:40:24.060 : ok
21:40:50.908 : N1367 M114 *52
21:40:50.986 : X:188.99 Y:178.99 Z:10.72 E:0.00 Count X: 189.00 Y:179.00 Z:10.51
21:40:50.986 : ok
21:41:42.502 : N1393 M114 *63
21:41:42.502 : X:188.99 Y:178.99 Z:10.72 E:0.00 Count X: 189.00 Y:179.00 Z:10.51
21:41:42.502 : ok
21:41:55.106 : N1400 G1 Z0 *119
21:41:55.106 : ok

The head is about 1 mm away from the bed

thinkyhead, 2015-04-12:
https://github.com/thinkyhead/Marlin/archive/debug_leveling.zip

I've updated the debug branch, adding logging at the point where `zprobe_zoffset` is added to the Z axis position. I'm wondering if `zprobe_zoffset` should be instead applied at a more central point – either in `homeaxis` or `axis_is_at_home` instead of _only_ in `G28` as it is now.

I noticed that on robots with a Z max endstop, the Z bed probe might have still been deploying as home starts. I will have to double-check on that. Clearly, the probe should not be deployed when homing to a Z max endstop.

[90 more replies]

closed 2015-08-03 by boelle